dusty-nv/jetson-containers #796: Building *Speaches* stuck on due to uv install on system python environment

JonnyTran, 2025-01-24:
Hi,

I'm trying to build [Speaches](https://github.com/speaches-ai/speaches) for STT and TTS integration with OpenWebUI and Ollama. Since it uses `faster-whisper` backend, I figured I would be able to extend the base image [faster-whisper](https://github.com/dusty-nv/jetson-containers/tree/master/packages/speech/faster-whisper) to build speaches from their [Dockerfile](https://github.com/speaches-ai/speaches/blob/master/Dockerfile).

This has proved very difficult due to their use of `uv` to install python virtual environments. Ssince we have to use the pre-built python binding stored at `/usr/local/lib/python3.11/dist-packages` to make use of CUDA libs, it is difficult to `uv pip install` or `pip install` directly to the python system environment without setting the proper PATH, making extending jetson-containers as a base images to other framework difficult.

Is there a way we can switch away from using the system python installation in [packages/build/python](https://github.com/dusty-nv/jetson-containers/tree/master/packages/build/python) to something `~/.venv/` by default? This would make dealing with python environments much easier.

Model: NVIDIA AGX Orin DevKit - Jetpack 6.2 L4T 36.4.3

RS-Credentive, 2025-02-13:
Thanks for your reply. I'll open a separate issue on the ComfyUI thing - I don't want to pollute this discussion.

johnnynunez, 2025-02-13:
> Thanks for your reply. I'll open a separate issue on the ComfyUI thing - I don't want to pollute this discussion.

Dusty is on vacation during some days, I will ask what think about uv

RS-Credentive, 2025-02-13:
I will say that it's fast and fabulous and very aggressive about enforcing venvs, and does some smart things with links to reduce the disk space used for multiple copies of different packages. it could be quite helpful to have it as core part of jetson-containers, but it could be very complicated to integrate and I don't know if it would cause more problems for other AI packages. I don't want to pretend it's just a drop in replacement.

JonnyTran, 2025-02-13:
Hi folks! I understand the difficulties with using venv.

My [proposed PR](https://github.com/dusty-nv/jetson-containers/pull/801/) only tries to move away from the `/usr/local/lib/python3.11/` system Python environment and setting `/opt/conda/` as the new system Python environment. It doesn't enforce using venv at all (by setting the `CONDA_PREFIX=/opt/conda` environment variable), so all `uv pip install` commands will automatically install to this location. Additionally, all existing `pip3 install` commands will also target this environment, so it's compatible with other Jetson containers without requiring any changes.