dusty-nv/jetson-containers #811: PyPI index (http://jetson.webredirect.org) is down

nicolofranceschi, 2025-02-01:
Looking in indexes: http://jetson.webredirect.org/jp6/cu122, https://pypi.ngc.nvidia.com                      
1.107 WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPConnection object at 0xffffa2c67af0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')'

johnnynunez, 2025-02-01:
> Looking in indexes: http://jetson.webredirect.org/jp6/cu122, https://pypi.ngc.nvidia.com 1.107 WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after [...]

https://pypi.jetson-ai-lab.dev/ new page

nicolofranceschi, 2025-02-03:
how can i update that ? im building a docker container FROM dustynv/l4t-pytorch:r36.2.0 there is a new version with the repo link update ?

FROM dustynv/l4t-pytorch:r36.2.0

WORKDIR /code

RUN apt update && apt install -y sudo && sudo apt install default-jdk -y

RUN apt-get install -y libmagic1
RUN apt-get install -y libreoffice
RUN apt-get install -y poppler-utils
RUN export HF_HUB_ENABLE_HF_TRANSFER=1

RUN pip install --upgrade pip

COPY ./requirements.txt /code/requirements.txt

RUN pip install -r requirements.txt

RUN pip install http://jetson.webredirect.org/jp6/cu126/+f/5cf/9ed17e35cb752/torch-2.5.0-cp310-cp310-linux_aarch64.whl#sha256=5cf9ed17e35cb7523812aeda9e7d6353c437048c5a6df1dc6617650333049092
RUN pip install http://jetson.webredirect.org/jp6/cu126/+f/5f9/67f920de3953f/torchvision-0.20.0-cp310-cp310-linux_aarch64.whl#sha256=5f967f920de3953f2a39d95154b1feffd5ccc06b4589e51540dc070021a9adb9

RUN wget https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2204/arm64/cuda-keyring_1.1-1_all.deb
RUN sudo dpkg -i cuda-keyring_1.1-1_all.deb
RUN sudo apt-get update
RUN sudo apt-get -y install cudnn
RUN sudo apt-get -y install cudnn-cuda-12

RUN pip install psycopg2-binary

COPY ./app/dw.py /code/app/dw.py

RUN python3 app/dw.py

COPY ./app /code/app

COPY ./ /code

RUN sudo apt install supervisor

ENV QDRANTLINK=http://qdrant:6333
ENV POSTGRES_URI=postgresql://admin:admin@postgres:5432/db
ENV REDIS_URI=redis://cache:6379/0

COPY supervisord.conf /etc/supervisor/conf.d/supervisord.conf

EXPOSE 7680

# Start supervisord
CMD ["/usr/bin/supervisord", "-c", "/etc/supervisor/conf.d/supervisord.conf"]

johnnynunez, 2025-02-03:
> how can i update that ? im building a docker container FROM dustynv/l4t-pytorch:r36.2.0 there is a new version with the repo link update ?
>
> FROM dustynv/l4t-pytorch:r36.2.0
>
> WORKDIR /code
>
> RUN apt [...]

dustynv/l4t-pytorch:r36.4.0

dusty-nv, 2025-02-03:
@nicolofranceschi if you needed to stick with the same base image, in your dockerfile or environment you could set `PIP_INDEX_URL` again with the updated URL.  Those originally get set here:

* https://github.com/dusty-nv/jetson-containers/blob/master/packages/cuda/cuda/Dockerfile.pip
* https://github.com/dusty-nv/jetson-containers/blob/master/packages/cuda/cuda/config.py

nicolofranceschi, 2025-02-17:
sorry guys, today is the same, i think is really down now,

0.388 Looking in indexes: https://pypi.jetson-ai-lab.dev/, https://pypi.ngc.nvidia.com                                                                                                                      
0.660 WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffb5a458d0>: Failed to establish a new connection: [Errno 111] Connection refused')': /accelerate/ 

DOCKERFILE

FROM dustynv/l4t-pytorch:r36.4.0

WORKDIR /code

RUN apt-get update && apt-get install -y libmagic1 libreoffice poppler-utils && rm -rf /var/lib/apt/lists/*

RUN export HF_HUB_ENABLE_HF_TRANSFER=1
ENV PIP_INDEX_URL=https://pypi.jetson-ai-lab.dev/

[image]

closed 2025-02-17 by nicolofranceschi

kairin, 2025-02-17:
no problem running other commands, pip won't change its config to run properly...

what url should i change pip's config to manually?

```
root@kkkORIN:/# ls
bin  boot  data  dev  etc  home  lib  media  mnt  opt  ppp  proc  root  run  sbin  srv  sys  tmp  usr  var
root@kkkORIN:/# pip install flash-attn
Looking in indexes: https://pypi.jetson-ai-lab.dev/jp6/cu128, https://pypi.ngc.nvidia.com
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa43b3da0>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu128/flash-attn/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa4171d90>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu128/flash-attn/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa4172330>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu128/flash-attn/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa4172540>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu128/flash-attn/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa4172720>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu128/flash-attn/
ERROR: Could not find a version that satisfies the requirement flash-attn (from versions: none)
ERROR: No matching distribution found for flash-attn
root@kkkORIN:/# pip config set global index-url jetson.webdirect.org/jp6/cu122
ERROR: Got unexpected number of arguments, expected 2. (example: "pip config set [name] [value]")
root@kkkORIN:/# pip config set index-url jetson.webdirect.org/jp6/cu122
ERROR: Key does not contain dot separated section and key. Perhaps you wanted to use 'global.index-url' instead?
root@kkkORIN:/# pip config set global.index-url jetson.webdirect.org/jp6/cu122
Writing to /usr/pip.conf
root@kkkORIN:/# pip install flash-attn
Looking in indexes: https://pypi.jetson-ai-lab.dev/jp6/cu128, https://pypi.ngc.nvidia.com
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff7fed7ec0>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu128/flash-attn/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff7f03e180>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu128/flash-attn/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff7f03e510>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu128/flash-attn/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff7f03e6f0>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu128/flash-attn/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff7f03e7b0>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu128/flash-attn/
ERROR: Could not find a version that satisfies the requirement flash-attn (from versions: none)
ERROR: No matching distribution found for flash-attn
root@kkkORIN:/# pip config set global.index-url https://pypi.jetson-ai-lab.dev/jp6/cu128
Writing to /usr/pip.conf
root@kkkORIN:/# pip install flash-attention
Looking in indexes: https://pypi.jetson-ai-lab.dev/jp6/cu128, https://pypi.ngc.nvidia.com
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffb2a467e0>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu128/flash-attention/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffb2319fd0>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu128/flash-attention/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffb231a540>: Failed to establish a new connection: [Errno 111] Connection [... cut ...]
Setting up libhttp-daemon-perl (6.13-1ubuntu0.1) ...
Setting up audacious (4.1-2) ...
Setting up unar (1.10.1-2build11) ...
Setting up liblwp-protocol-https-perl (6.10-1) ...
Setting up libwww-perl (6.61-1) ...
Setting up libxml-parser-perl:arm64 (2.46-3build1) ...
Setting up libxml-twig-perl (1:3.52-1) ...
Setting up libnet-dbus-perl (1.2.0-1build3) ...
Processing triggers for mailcap (3.70+nmu1ubuntu1) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for libc-bin (2.35-0ubuntu3.9) ...
root@kkkORIN:/# 
```

dusty-nv, 2025-02-18:
@kairin @nicolofranceschi https://pypi.jetson-ai-lab.dev is back up now, sorry about that, we try to prevent further outage in the future 👍

nicolofranceschi, 2025-04-29:
hi, it look the same again
 
 => => # WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor
 => => # .urllib3.connection.HTTPSConnection object at 0xffffb5ff27a0>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu126/pyt
 => => # hon-crontab/                                                                                                                                        
 => => # WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor
 => => # .urllib3.connection.HTTPSConnection object at 0xffffb5ff2950>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu126/pyt
 => => # hon-crontab/

dusty-nv, 2025-04-29:
Sorry yes there is a power outage just now due to storms in the area.  I will let you know when service is restored, thank you.

Ironically i am in the process of setting up battery backup.

nicolofranceschi, 2025-07-11:
hi, i think is down again https://pypi.jetson-ai-lab.dev

nicolofranceschi, 2025-07-11:
Can I in some way help with that ? There is a uptime page or something ? is kinda crucial to me that registry, so if I can help you guys in some way