dusty-nv/jetson-containers #817: piper-tts fails to build during phonemize stage

OriNachum, 2025-02-04:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### jetson-containers Component

Packages

### Bug

Trying to build piper-tts package on newly setup Orin Nano Super 8GB fails during phonemize stage

[image]

 

### Environment

Jetpack 6.2

### Are you willing to submit a PR?

- [x] Yes I'd like to help by submitting a PR!

OriNachum, 2025-02-04:
Opened PR 815:
https://github.com/dusty-nv/jetson-containers/pull/815

And it is pending review

ms1design, 2025-02-21:
Resolved in 4feb6c8fa6708e1bd3df4197d122bccd312f9086