dusty-nv/jetson-containers #839: No support for NVILA?

George-Polya, 2025-02-10:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Question

I used VLM of Jetson Platform Service. In here, I sawyour mlc-llm patch.

I tried to use Efficient-Large-Model/NVILA-8B. 

Then, I saw the error below.

```
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.10/dist-packages/mlc_llm/build.py", line 47, in <module>
    main()
  File "/usr/local/lib/python3.10/dist-packages/mlc_llm/build.py", line 41, in main
    parsed_args = core._parse_args(parsed_args)  # pylint: disable=protected-access
  File "/usr/local/lib/python3.10/dist-packages/mlc_llm/core.py", line 444, in _parse_args
    parsed = _setup_model_path(parsed)
  File "/usr/local/lib/python3.10/dist-packages/mlc_llm/core.py", line 494, in _setup_model_path
    validate_config(args.model_path)
  File "/usr/local/lib/python3.10/dist-packages/mlc_llm/core.py", line 538, in validate_config
    config["model_type"] in utils.supported_model_types
AssertionError: Model type qwen2 not supported.
```

Maybe, I think that's because mlc patches. How can I use NVILA?

dusty-nv, 2025-02-10:
Hi @George-Polya , I rebuilt it last week in AWQ:  https://github.com/dusty-nv/jetson-containers/commit/75c1ab16a3430a57571d37d2d1c50a9c4ababe7f    `dustynv/awq:r36.4.0`

the NVILA repo supports it in HF and AWQ TinyChat.  I have the action to profile it and add TinyChat support to OpenAI server.  When it is more ready I will retag `dustynv/awq` as `dustynv/vila` since those are circular dependencies now.

George-Polya, 2025-02-10:
Thank you for your response.

George-Polya, 2025-03-10:
Can I use NVILA in nano_llm container?

dusty-nv, 2025-03-10:
I have not gotten back to AWQ TinyChat yet, but still plan to because desire high-quality / high-performance quantization for R1, QwQ, and NVILA.  I suspect those and the chat.completion server implementation for them will end up in NanoLLM.

George-Polya, 2025-03-10:
@dusty-nv Thank you for your reply. 
I also try it using mlc-llm==0.19.0, tvm=0.19.0

I apologize for rushing you

johnnynunez, 2025-07-28:
> > Hi [@George-Polya](https://github.com/George-Polya) , I rebuilt it last week in AWQ: [75c1ab1](https://github.com/dusty-nv/jetson-containers/commit/75c1ab16a3430a57571d37d2d1c50a9c4ababe7f) `dustynv/awq:r36.4.0`
> > the NVILA repo supports it in HF and AWQ TinyChat. I have the action to profile it and add TinyChat support to OpenAI server. When it is more ready I will retag `dustynv/awq` as `dustynv/vila` since those are circular dependencies now.
> 
> Hi, I couldn't find /opt/awq in dustynv/awq:r36.4.0, could you please check this out?

That repo is deprecated, use vllm autoawq