dusty-nv/jetson-containers #855: CUDA 12.6 torch libraries are unaligned

OriNachum, 2025-02-15:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### jetson-containers Component

Packages

### Bug

Loading a CUDA 12.6 container and installing in it torch-2.6.0rc and torchvision-2.5.0 manually from site then importing them in python ends up in version mismatch.

Old package download link didn’t work

### Environment

Jetson Orin Nano Super vanilla with Jetson-container- inside the container.

### Additional

This used to work in past as I documented the process.
Moving to CUDA 128 bypassed the issue as it also fits.

### Are you willing to submit a PR?

- [x] Yes I'd like to help by submitting a PR!

johnnynunez, 2025-02-17:
> ### Search before asking
> * [x]  I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.
>
> ### jetson-containers [...]

hello! In cuda 12.6 exists pytorch 2.5 that it's by default. Pytorch 2.6.0.rc0 was created for other nvidia teams to test flex attention.

closed 2025-02-17 by johnnynunez

johnnynunez, 2025-02-17:
> The problem is that I need to install the packages manually, in managed in uv (For Kokoro-FastAPI)

You can put url and version torch=2.5.1