dusty-nv/jetson-containers #863: How to use function call in MLC

tebie6, 2025-02-20:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Question

Hi, @dusty-nv  ,
I hope you're doing well. I wanted to ask if you've ever used the function call feature in MLC. I tried using it in the MLC image, but tool_calls keeps returning None. I'm not sure if you've encountered this issue before. I've searched quite a bit but haven't been able to find a solution. So, I thought I'd reach out to see if you have any experience with this or any insights to share. I’d really appreciate any help. Thanks a lot!

ms1design, 2025-02-21:
Hi @tebie6, the function calling in MLC-LLM is still not 100% compliant with OpenAI Chat Completions schema. This about it as a inline function calling within the `content` field only if not using any `grammars`.

Read more here:
- https://github.com/mlc-ai/mlc-llm/pull/1734
- https://github.com/mlc-ai/mlc-llm/pull/2744
- https://github.com/mlc-ai/mlc-llm/pull/1734
- https://github.com/mlc-ai/mlc-llm/pull/1378

tebie6, 2025-02-23:
> Hi [@tebie6](https://github.com/tebie6), the function calling in MLC-LLM is still not 100% compliant with OpenAI Chat Completions schema. This about it as a inline function calling within the `content` field only if not using any `grammars`.嗨，在MLC-中调用的功能 - LLM仍然不符合OpenAI聊天完成模式的100％。仅在不使用任何`grammars`时，它是在`content`字段中呼叫的内联函数。
> 
> Read more here:  在此处阅读更多：
> 
> * [[Serving] Add OpenAI compatible function calling to Serving  mlc-ai/mlc-llm#1734](https://github.com/mlc-ai/mlc-llm/pull/1734)
> * [[Tool] Prelim support for function calling with Llama3.1 and Hermes2 mlc-ai/mlc-llm#2744](https://github.com/mlc-ai/mlc-llm/pull/2744)
> * [[Serving] Add OpenAI compatible function calling to Serving  mlc-ai/mlc-llm#1734](https://github.com/mlc-ai/mlc-llm/pull/1734)
> * [[REST] OpenAI Rest API compatible function calling mlc-ai/mlc-llm#1378](https://github.com/mlc-ai/mlc-llm/pull/1378)

tebie6, 2025-02-23:
> Hi [@tebie6](https://github.com/tebie6), the function calling in MLC-LLM is still not 100% compliant with OpenAI Chat Completions schema. This about it as a inline function calling within the `content` field only if not using any `grammars`.
> 
> Read more here:
> 
> * [[Serving] Add OpenAI compatible function calling to Serving  mlc-ai/mlc-llm#1734](https://github.com/mlc-ai/mlc-llm/pull/1734)
> * [[Tool] Prelim support for function calling with Llama3.1 and Hermes2 mlc-ai/mlc-llm#2744](https://github.com/mlc-ai/mlc-llm/pull/2744)
> * [[Serving] Add OpenAI compatible function calling to Serving  mlc-ai/mlc-llm#1734](https://github.com/mlc-ai/mlc-llm/pull/1734)
> * [[REST] OpenAI Rest API compatible function calling mlc-ai/mlc-llm#1378](https://github.com/mlc-ai/mlc-llm/pull/1378)

@ms1design Thank you very much for your response! I truly appreciate your time and effort in answering this question. Your insights have been incredibly helpful to me. Thanks again, and I wish you all the best!

closed 2025-02-23 by tebie6