dusty-nv/jetson-containers #883: Why does jetpack >6 use ubuntu:22.04 as base image instead of nvcr.io/nvidia/l4t-jetpack:r36.X.X

tonynajjar, 2025-02-27:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Question

This is a follow up on https://github.com/dusty-nv/jetson-containers/issues/763#issuecomment-2678328508

johnnynunez, 2025-02-27:
I think that you refer 24.04 image?
it is because Jetpack 7 comes with ubuntu 24.04, so we will support it asap.

tonynajjar, 2025-02-27:
hmm no, take a look at the code here: https://github.com/dusty-nv/jetson-containers/blob/9889aedb049622f8bbaab3d24ddb6da922ea07af/jetson_containers/l4t_version.py#L231

For Jetpack 5, `nvcr.io/nvidia/l4t-jetpack:r35.4.1` is used but for Jetpack 6, a bare `ubuntu:22.04` image is used although `nvcr.io/nvidia/l4t-jetpack:r36.4.0` exists and I'm wondering why. I couldn't find the Dockerfile of `nvcr.io/nvidia/l4t-jetpack:r36.4.0` to understand the difference but I assume there are some pretty important changes maybe?

johnnynunez, 2025-02-28:
> > I think that you refer 24.04 image?
> 
> hmm no, take a look at the code here:
> 
> [jetson-containers/jetson_containers/l4t_version.py](https://github.com/dusty-nv/jetson-containers/blob/9889aedb049622f8bbaab3d24ddb6da922ea07af/jetson_containers/l4t_version.py#L231)
> 
> Line 231 in [9889aed](/dusty-nv/jetson-containers/commit/9889aedb049622f8bbaab3d24ddb6da922ea07af)
> 
>  return "ubuntu:22.04" #"nvcr.io/ea-linux4tegra/l4t-jetpack:r36.0.0" 
> For Jetpack 5, `nvcr.io/nvidia/l4t-jetpack:r35.4.1` is used but for Jetpack 6, a bare `ubuntu:22.04` image is used although `nvcr.io/nvidia/l4t-jetpack:r36.4.0` exists and I'm wondering why. I couldn't find the Dockerfile of `nvcr.io/nvidia/l4t-jetpack:r36.4.0` to understand the difference but I assume there are some pretty important changes (?)

more clean. I think that Jetpack6 dockers comes with everything installed, it is like plug and play

dusty-nv, 2025-02-28:
Yes, it was no longer needed to use l4t-base, as l4t-base itself had just become simple to the point it made more sense to eliminate it as a dependency, and now portability is also improved with some equally long-term changes having also recently come into effect that enable 24.04 to work, which we are building for now.

closed 2025-02-28 by tonynajjar