dusty-nv/jetson-containers #921: Ollama phi4-mini model fails on tesnor 'output.weight' on orin super nano

OriNachum, 2025-03-08:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### jetson-containers Component

Packages

### Bug

Error:
root@ori-desktop:/# ollama run phi4-mini
pulling manifest 
pulling 3c168af1dea0... 100% ▕... ▏ 2.5 GB                         
pulling 813f53fdc6e5... 100% ▕...▏  655 B                         
pulling fa8235e5b48f... 100% ▕...▏ 1.1 KB                         
pulling 8c2539a423c4... 100% ▕...▏  411 B                         
verifying sha256 digest 
writing manifest 
success 
Error: llama runner process has terminated: error loading model: check_tensor_dims: tensor 'output.weight' not found
llama_load_model_from_file: failed to load model

Steps:
1. Install ollama from tag
2. Install phi4-mini

### Environment

jetson orin nano super

### Are you willing to submit a PR?

- [x] Yes I'd like to help by submitting a PR!

johnnynunez, 2025-03-11:
solved

closed 2025-03-11 by johnnynunez

lpalbou, 2025-03-14:
@johnnynunez sorry if I missed something ,but how is it solved ?
Just tried on OSX Sonoma 14.2 on Apple silicon and:

ollama run phi4-mini
Error: llama runner process has terminated: error loading model: missing tensor 'output.weight'

johnnynunez, 2025-03-14:
> [@johnnynunez](https://github.com/johnnynunez) sorry if I missed something ,but how is it solved ? Just tried on OSX Sonoma 14.2 on Apple silicon and:
> 
> ollama run phi4-mini Error: llama runner process has terminated: error loading model: missing tensor 'output.weight'

use the last ollama that we upload it. we no support macos, it uses mlx backend