dusty-nv/jetson-containers #939: SGLang container in cu128

chit-ang, 2025-03-12:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Question

I ran dustynv/sglang: 0.44-r36.4.0-CU128-24.04 through jetson-containers and reported an error：
```
root@ubuntu:/# python -m sglang.launch_server --model-path mistralai/Mistral-7B-Instruct-v0.3 --port 3000
/opt/venv/lib/python3.12/site-packages/transformers/utils/hub.py:128: FutureWarning: Using `TRANSFORMERS_CACHE` is deprecated and will be removed in v5 of Transformers. Use `HF_HOME` instead.
  warnings.warn(
INFO 03-12 16:28:44 [__init__.py:207] Automatically detected platform cuda.
[2025-03-12 16:28:47] server_args=ServerArgs(model_path='mistralai/Mistral-7B-Instruct-v0.3', tokenizer_path='mistralai/Mistral-7B-Instruct-v0.3', tokenizer_mode='auto', skip_tokenizer_init=False, load_format='auto', trust_remote_code=False, dtype='auto', kv_cache_dtype='auto', quantization=None, quantization_param_path=None, context_length=None, device='cuda', served_model_name='mistralai/Mistral-7B-Instruct-v0.3', chat_template=None, is_embedding=False, revision=None, host='127.0.0.1', port=3000, mem_fraction_static=0.88, max_running_requests=None, max_total_tokens=None, chunked_prefill_size=8192, max_prefill_tokens=16384, schedule_policy='fcfs', schedule_conservativeness=1.0, cpu_offload_gb=0, prefill_only_one_req=False, tp_size=1, stream_interval=1, stream_output=False, random_seed=177342676, constrained_json_whitespace_pattern=None, watchdog_timeout=300, dist_timeout=None, download_dir=None, base_gpu_id=0, gpu_id_step=1, log_level='info', log_level_http=None, log_requests=False, log_requests_level=0, show_time_cost=False, enable_metrics=False, decode_log_interval=40, api_key=None, file_storage_path='sglang_storage', enable_cache_report=False, dp_size=1, load_balance_method='round_robin', ep_size=1, dist_init_addr=None, nnodes=1, node_rank=0, json_model_override_args='{}', lora_paths=None, max_loras_per_batch=8, lora_backend='triton', attention_backend='flashinfer', sampling_backend='flashinfer', grammar_backend='outlines', speculative_algorithm=None, speculative_draft_model_path=None, speculative_num_steps=5, speculative_eagle_topk=4, speculative_num_draft_tokens=8, speculative_accept_threshold_single=1.0, speculative_accept_threshold_acc=1.0, speculative_token_map=None, enable_double_sparsity=False, ds_channel_config_path=None, ds_heavy_channel_num=32, ds_heavy_token_num=256, ds_heavy_channel_type='qk', ds_sparse_decode_threshold=4096, disable_radix_cache=False, disable_cuda_graph=False, disable_cuda_graph_padding=False, enable_nccl_nvls=False, disable_outlines_disk_cache=False, disable_custom_all_reduce=False, disable_mla=False, disable_overlap_schedule=False, enable_mixed_chunk=False, enable_dp_attention=False, enable_ep_moe=False, enable_torch_compile=False, torch_compile_max_bs=32, cuda_graph_max_bs=160, cuda_graph_bs=None, torchao_config='', enable_nan_detection=False, enable_p2p_check=False, triton_attention_reduce_in_fp32=False, triton_attention_num_kv_splits=8, num_continuous_decode_steps=1, delete_ckpt_after_loading=False, enable_memory_saver=False, allow_auto_truncate=False, enable_custom_logit_processor=False, tool_call_parser=None, enable_hierarchical_cache=False, enable_flashinfer_mla=False, flashinfer_mla_disable_ragged=False, warmups=None, debug_tensor_dump_output_folder=None, debug_tensor_dump_input_file=None, debug_tensor_dump_inject=False)
/opt/venv/lib/python3.12/site-packages/transformers/utils/hub.py:128: FutureWarning: Using `TRANSFORMERS_CACHE` is deprecated and will be removed in v5 of Transformers. Use `HF_HOME` instead.
  warnings.warn(
/opt/venv/lib/python3.12/site-packages/transformers/utils/hub.py:128: FutureWarning: Using `TRANSFORMERS_CACHE` is deprecated and will be removed in v5 of Transformers. Use `HF_HOME` instead.
  warnings.warn(
INFO 03-12 16:28:54 [__init__.py:207] Automatically detected platform cuda.
INFO 03-12 16:28:54 [__init__.py:207] Automatically detected platform cuda.
[2025-03-12 16:28:59 TP0] Init torch distributed begin.
[2025-03-12 16:28:59 TP0] Failed to import pynvml with ModuleNotFoundError("No module named 'pynvml'")
[2025-03-12 16:28:59 TP0] Load weight begin. avail mem=47.80 GB
[2025-03-12 16:28:59 TP0] The following error message 'operation scheduled before its operands' can be ignored.
[2025-03-12 16:29:02 TP0] Using model weights format ['*.safetensors']
Loading safetensors checkpoint shards:   0% Completed | 0/3 [00:00<?, ?it/s]
Loading safetensors checkpoint shards:  33% Completed | 1/3 [00:01<00:03,  1.80s/it]
Loading safetensors checkpoint shards:  67% Completed | 2/3 [00:03<00:01,  1.88s/it]
Loading safetensors checkpoint shards: 100% Completed | 3/3 [00:06<00:00,  2.19s/it]
Loading safetensors checkpoint shards: 100% Completed | 3/3 [00:06<00:00,  2.10s/it]

[2025-03-12 16:29:09 TP0] Load weight end. type=MistralForCausalLM, dtype=torch.bfloat16, avail mem=41.72 GB
[2025-03-12 16:29:17 TP0] KV Cache is allocated. K size: 17.99 GB, V size: 17.99 GB.
[2025-03-12 16:29:17 TP0] Memory pool end. avail mem=5.74 GB
[2025-03-12 16:29:17 TP0] Capture cuda graph begin. This can take up to several minutes. avail mem=4.77 GB
  0%|                                                                                                            | 0/23 [00:00<?, ?it/s]
[2025-03-12 16:29:18 TP0] Scheduler hit an exception: Traceback (most recent call last):
  File "/opt/venv/lib/python3.12/site-packages/triton/backends/nvidia/compiler.py", line 292, in make_cubin
    subprocess.run(cmd, shell=True, check=True)
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '/usr/local/cuda/bin/ptxas -lineinfo -v --gpu-name=sm_87 /tmp/tmpma1fkmf6.ptx -o /tmp/tmpma1fkmf6.ptx.o 2> /tmp/tmptm_ofd1d.log' returned non-zero exit status 255.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/venv/lib/python3.12/site-packages/sglang/srt/model_executor/cuda_graph_runner.py", line 245, in __init__
    self.capture()
  File "/opt/venv/lib/python3.12/site-packages/sglang/srt/model_executor/cuda_graph_runner.py", line 312, in capture
    ) = self.capture_one_batch_size(bs, forward)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.12/site-packages/sglang/srt/model_executor/cuda_graph_runner.py", line 372, in capture_one_batch_size
    self.model_runner.attn_backend.init_forward_metadata_capture_cuda_graph(
  File "/opt/venv/lib/python3.12/site-packages/sglang/srt/layers/attention/flashinfer_backend.py", line 299, in init_forward_metadata_capture_cuda_graph
    self.indices_updater_decode.update(
  File "/opt/venv/lib/python3.12/site-packages/sglang/srt/layers/attention/flashinfer_backend.py", line 555, in update_single_wrapper
    self.call_begin_forward(
  File "/opt/venv/lib/python3.12/site-packages/sglang/srt/layers/attention/flashinfer_backend.py", line 646, in call_begin_forward
    create_flashinfer_kv_indices_triton[(bs,)](
  File "/opt/venv/lib/python3.12/site-packages/triton/runtime/jit.py", line 345, in <lambda>
    return lambda *args, **kwargs: self.run(grid=grid, warmup=False, *args, **kwargs)
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.12/site-packages/triton/runtime/jit.py", line 662, in run
    kernel = self.compile(
             ^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.12/site-packages/triton/compiler/compiler.py", line 282, in compile
    next_module = compile_ir(module, metadata)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.12/site-packages/triton/backends/nvidia/compiler.py", line 320, in <lambda>
    stages["cubin"] = lambda src, metadata: self.make_cubin(src, metadata, options, self.capability)
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.12/site-packages/triton/backends/nvidia/compiler.py", line 297, in make_cubin
    raise RuntimeError(f'Internal Triton PTX codegen error: \n{log}')
RuntimeError: Internal Triton PTX codegen error: 
ptxas /tmp/tmpma1fkmf6.ptx, line 5; fatal   : Unsupported .version 8.8; current version is '8.7'
ptxas fatal   : Ptx assembly aborted due to errors
```

dusty-nv, 2025-03-12:
Thanks @chit-ang - it looks like the CUDA_ARCHITECTURES or TORCH_CUDA_ARCH_LIST aren't getting overridden or patched correctly somewhere - what do you think @shahizat ?

shahizat, 2025-03-13:
Hi @chit-ang, Unfortunately, I was not able to reproduce this. Would you be able to build it yourself using the command: `LSB_RELEASE=24.04 CUDA_VERSION=12.8 PYTHON_VERSION=3.12 PYTORCH_VERSION=2.6   jetson-containers build sglang`

johnnynunez, 2025-03-13:
me either

johnnynunez, 2025-03-14:
seems that 12.8 docker must be updated. reinstall wheels with pip3 install --force-reinstall vllm sglang flashinfer sgl-kernel --index-url https://pypi.jetson-ai-lab.dev/jp6/cu128

chit-ang, 2025-03-17:
How can i solve this problem?

```
root@ubuntu:/# pip3 install --force-reinstall vllm sglang flashinfer sgl-kernel --index-url https://pypi.jetson-ai-lab.dev/jp6/cu128
Using pip 25.0.1 from /opt/venv/lib/python3.12/site-packages/pip (python 3.12)
Non-user install because site-packages writeable
Created temporary directory: /tmp/pip-build-tracker-jm_maoix
Initialized build tracking at /tmp/pip-build-tracker-jm_maoix
Created build tracker: /tmp/pip-build-tracker-jm_maoix
Entered build tracker: /tmp/pip-build-tracker-jm_maoix
Created temporary directory: /tmp/pip-install-azqdp7g4
Created temporary directory: /tmp/pip-ephem-wheel-cache-vs5o1keb
Looking in indexes: https://pypi.jetson-ai-lab.dev/jp6/cu128
1 location(s) to search for versions of vllm:
* https://pypi.jetson-ai-lab.dev/jp6/cu128/vllm/
Fetching project page and analyzing links: https://pypi.jetson-ai-lab.dev/jp6/cu128/vllm/
Getting page https://pypi.jetson-ai-lab.dev/jp6/cu128/vllm/
Found index url https://pypi.jetson-ai-lab.dev/jp6/cu128/
Starting new HTTPS connection (1): pypi.jetson-ai-lab.dev:443
https://pypi.jetson-ai-lab.dev:443 "GET /jp6/cu128/vllm/ HTTP/1.1" 302 213
https://pypi.jetson-ai-lab.dev:443 "GET /jp6/cu128/vllm HTTP/1.1" 200 None
Fetched page https://pypi.jetson-ai-lab.dev/jp6/cu128/vllm/ as application/vnd.pypi.simple.v1+json
  Skipping link: none of the wheel's tags (cp310-cp310-linux_aarch64) are compatible (run pip debug --verbose to show compatible tags): https://pypi.jetson-ai-lab.dev/jp6/cu128/%2Bf/e77/ab94e199794e9/vllm-0.7.3%2Bcu128-cp310-cp310-linux_aarch64.whl (from https://pypi.jetson-ai-lab.dev/jp6/cu128/vllm) (requires-python:>=3.9)
  Skipping link: none of the wheel's tags (cp310-cp310-linux_aarch64) are compatible (run pip debug --verbose to show compatible tags): https://pypi.jetson-ai-lab.dev/jp6/cu128/%2Bf/e47/32dbe63aac7a3/vllm-0.7.4%2Bcu128-cp310-cp310-linux_aarch64.whl (from https://pypi.jetson-ai-lab.dev/jp6/cu128/vllm) (requires-python:>=3.9)
  Found link https://pypi.jetson-ai-lab.dev/jp6/cu128/%2Bf/e56/3ad23096873db/vllm-0.7.4%2Bcu128-cp312-cp312-linux_aarch64.whl (from https://pypi.jetson-ai-lab.dev/jp6/cu128/vllm) (requires-python:>=3.9), version: 0.7.4+cu128
Skipping link: not a file: https://pypi.jetson-ai-lab.dev/jp6/cu128/vllm/
Given no hashes to check 1 links for project 'vllm': discarding no candidates
Collecting vllm
  Created temporary directory: /tmp/pip-unpack-03d4bgki
  Found index url https://pypi.jetson-ai-lab.dev/jp6/cu128/
  https://pypi.jetson-ai-lab.dev:443 "GET /jp6/cu128/%2Bf/e56/3ad23096873db/vllm-0.7.4%2Bcu128-cp312-cp312-linux_aarch64.whl HTTP/1.1" 200 296057548
  Downloading https://pypi.jetson-ai-lab.dev/jp6/cu128/%2Bf/e56/3ad23096873db/vllm-0.7.4%2Bcu128-cp312-cp312-linux_aarch64.whl (296.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 296.1/296.1 MB 1.5 MB/s eta 0:00:00
1 location(s) to search for versions of sglang:
* https://pypi.jetson-ai-lab.dev/jp6/cu128/sglang/
Fetching project page and analyzing links: https://pypi.jetson-ai-lab.dev/jp6/cu128/sglang/
Getting page https://pypi.jetson-ai-lab.dev/jp6/cu128/sglang/
Found index url https://pypi.jetson-ai-lab.dev/jp6/cu128/
https://pypi.jetson-ai-lab.dev:443 "GET /jp6/cu128/sglang/ HTTP/1.1" 302 215
https://pypi.jetson-ai-lab.dev:443 "GET /jp6/cu128/sglang HTTP/1.1" 200 None
Fetched page https://pypi.jetson-ai-lab.dev/jp6/cu128/sglang/ as application/vnd.pypi.simple.v1+json
  Found link https://pypi.jetson-ai-lab.dev/jp6/cu128/%2Bf/f52/e171f8a43e9e9/sglang-0.4.3.post4-py3-none-any.whl (from https://pypi.jetson-ai-lab.dev/jp6/cu128/sglang) (requires-python:>=3.8), version: 0.4.3.post4
  Found link https://pypi.jetson-ai-lab.dev/jp6/cu128/%2Bf/ea4/b792ce1bff587/sglang-0.4.3.post2-py3-none-any.whl (from https://pypi.jetson-ai-lab.dev/jp6/cu128/sglang) (requires-python:>=3.8), version: 0.4.3.post2
Skipping link: not a file: https://pypi.jetson-ai-lab.dev/jp6/cu128/sglang/
Given no hashes to check 2 links for project 'sglang': discarding no candidates
Collecting sglang
  Created temporary directory: /tmp/pip-unpack-0rz37hdv
  Found index url https://pypi.jetson-ai-lab.dev/jp6/cu128/
  https://pypi.jetson-ai-lab.dev:443 "GET /jp6/cu128/%2Bf/f52/e171f8a43e9e9/sglang-0.4.3.post4-py3-none-any.whl HTTP/1.1" 200 913274
  Downloading https://pypi.jetson-ai-lab.dev/jp6/cu128/%2Bf/f52/e171f8a43e9e9/sglang-0.4.3.post4-py3-none-any.whl (913 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 913.3/913.3 kB 2.3 MB/s eta 0:00:00
1 location(s) to search for versions of flashinfer:
* https://pypi.jetson-ai-lab.dev/jp6/cu128/flashinfer/
Fetching project page and analyzing links: https://pypi.jetson-ai-lab.dev/jp6/cu128/flashinfer/
Getting page https://pypi.jetson-ai-lab.dev/jp6/cu128/flashinfer/
Found index url https://pypi.jetson-ai-lab.dev/jp6/cu128/
https://pypi.jetson-ai-lab.dev:443 "GET /jp6/cu128/flashinfer/ HTTP/1.1" 302 219
https://pypi.jetson-ai-lab.dev:443 "GET /jp6/cu128/flashinfer HTTP/1.1" 404 None
Could not fetch URL https://pypi.jetson-ai-lab.dev/jp6/cu128/flashinfer/: 404 Client Error: Not Found for url: https://pypi.jetson-ai-lab.dev/jp6/cu128/flashinfer - skipping
Skipping link: not a file: https://pypi.jetson-ai-lab.dev/jp6/cu128/flashinfer/
Given no hashes to check 0 links for project 'flashinfer': discarding no candidates
ERROR: Could not find a version that satisfies the requirement flashinfer (from versions: none)
ERROR: No matching distribution found for flashinfer
Exception information:
[... cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.12/site-packages/pip/_internal/commands/install.py", line 386, in run
    requirement_set = resolver.resolve(
                      ^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.12/site-packages/pip/_internal/resolution/resolvelib/resolver.py", line 104, in resolve
    raise error from e
pip._internal.exceptions.DistributionNotFound: No matching distribution found for flashinfer
Removed build tracker: '/tmp/pip-build-tracker-jm_maoix'
```

shahizat, 2025-03-22:
Hello @chit-ang, Sorry for the late response. If you want to try it without docker. Here are the instructions using conda or venv. You can follow the same steps to build for CUDA 12.8 by yourself.

```
git clone  https://github.com/sgl-project/sglang.git --recursive
cd ./sglang/sgl-kernel
```
Modify setup.py. Add the following line:

`"-gencode=arch=compute_87,code=sm_87",`

Modify this line:
`options={"bdist_wheel": {"py_limited_api": "cp39"}},
`
to
`options={"bdist_wheel": {"py_limited_api": False}},
`

Build from source the sgl-kernel
```
export CUDA_HOME=/usr/local/cuda-12.6
export CUDA_PATH=$CUDA_HOME
MAX_JOBS=$(nproc) TORCH_CUDA_ARCH_LIST="8.7" python setup.py bdist_wheel  --plat-name linux-aarch64
cd dist
pip3 install *.whl
```
Then 

```
cd ..
cd ./python/sglang/srt
```
Modify utils.py in /python/sglang/srt
```
def get_nvgpu_memory_capacity():
    try:
        # Run nvidia-smi and capture the output
        result = subprocess.run(
            ["nvidia-smi", "--query-gpu=memory.total", "--format=csv,noheader,nounits"],
            stdout=subprocess.PIPE,
            stderr=subprocess.PIPE,
            text=True,
        )

        if result.returncode != 0:
            raise RuntimeError(f"nvidia-smi error: {result.stderr.strip()}")

        # Parse the output to extract memory values
        memory_values = [
            float(mem)
            for mem in result.stdout.strip().split("\n")
            if re.match(r"^\d+(\.\d+)?$", mem.strip())
        ]

        if not memory_values:
            raise ValueError("No GPU memory values found.")

        # Return the minimum memory value
        return min(memory_values)

    except FileNotFoundError:
        raise RuntimeError(
            "nvidia-smi not found. Ensure NVIDIA drivers are installed and accessible."
        )
```
to

```
def get_nvgpu_memory_capacity():
    try:
        # Run nvidia-smi and capture the output
        result = subprocess.run(
            ["nvidia-smi", "--query-gpu=memory.total", "--format=csv,noheader,nounits"],
            stdout=subprocess.PIPE,
            stderr=subprocess.PIPE,
            text=True,
        )

        if result.returncode != 0:
            raise RuntimeError(f"nvidia-smi error: {result.stderr.strip()}")

        # Parse the output to extract memory values
        memory_values = [
            float(mem)
            for mem in result.stdout.strip().split("\n")
            if re.match(r"^\d+(\.\d+)?$", mem.strip())
        ]

        # Return the minimum memory value
        return None

    except FileNotFoundError:
        raise RuntimeError(
            "nvidia-smi not found. Ensure NVIDIA drivers are installed and accessible."
        )
```

Modify pyproject.toml file. Remove the following dependencies:
```
"flashinfer_python==0.2.3",
"torch==2.5.1",
"vllm>=0.6.4.post1,<=0.7.2",
```
 Finally install sglang

`pip install -e "python[all]"
`

Then build flashinfer from source 

```
git clone https://github.com/flashinfer-ai/flashinfer.git --recursive
cd flashinfer
FLASHINFER_ENABLE_AOT=1  MAX_JOBS=$(nproc) TORCH_CUDA_ARCH_LIST="8.7" python setup.py bdist_wheel
cd dist
pip3 install *.whl
```

Then, download other packages like Triton or vLLM from [JP6 CUDA 12.6](https://pypi.jetson-ai-lab.dev/jp6/cu126) or [JP6 CUDA 12.8](https://pypi.jetson-ai-lab.dev/jp6/cu128). Hope it helps.

johnnynunez, 2025-04-09:
this is fixed

closed 2025-04-09 by johnnynunez