dusty-nv/jetson-containers #942: Gemma 3 on ollama on Jetson

davidADSP, 2025-03-12:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Bug

With the latest ollama Jetson version (ollama:0.5.1-r36.4.0), I get the following error when running Gemma 3:

Error: llama runner process has terminated: this model is not supported by your version of Ollama. You may need to upgrade

Will there be an upgraded ollama image soon that supports Gemma 3?

### Environment

ollama:0.5.1-r36.4.0

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

johnnynunez, 2025-03-13:
that is not last version, newer version is 0.6.0 compatible with gemma3

closed 2025-03-13 by johnnynunez

josolso, 2025-03-31:
Support for gemma3 seems to be broken again, even though I'm using main-r36.4.0:

```
~/jetson-containers$ jetson-containers run dustynv/ollama:main-r36.4.0

docker run --runtime nvidia ... --name jetson_container_20250331_140814 dustynv/ollama:main-r36.4.0

Starting ollama server

OLLAMA_HOST   0.0.0.0
OLLAMA_LOGS   /data/logs/ollama.log
OLLAMA_MODELS /data/models/ollama/models

ollama server is now started, and you can run commands here like 'ollama run gemma3'

root:/# ollama run gemma3
pulling manifest 
Error: pull model manifest: 412: 

The model you are attempting to pull requires a newer version of Ollama.

Please download the latest version at:

	https://ollama.com/download

root:/# ollama --version
ollama version is 0.6.0
```

josolso, 2025-03-31:
Managed now to build a container that can run gemma like this:

`CUDA_VERSION=12.8 jetson-containers build ollama:0.6.3`

dusty-nv, 2025-03-31:
Thanks @josolso, my guess is the ollama model registry changed upstream to require that updated version - I have just rebuild and pushed these images, and confirmed `gemma3` to be working:

```
dustynv/ollama:0.6.3-r36.4.0
dustynv/ollama:main-r36.4.0
```