dusty-nv/jetson-containers #944: Failing with ollama on newest jetson version

tobsch, 2025-03-13:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### jetson-containers Component

Docker

### Bug

I am getting this when starting Ollama on my Jetson orin nano with newest Jetpack version (6.2).
Is this still not compatible? Any ideas on how I can work around this issue?

Tobias
```
025/03/13 06:16:01 routes.go:1125: INFO server config env="map[CUDA_VISIBLE_DEVICES: GPU_DEVICE_ORDINAL: HIP_VISIBLE_DEVICES: HSA_OVERRIDE_GFX_VERSION: OLLAMA_DEBUG:false OLLAMA_FLASH_ATTENTION:false OLLAMA_HOST:http://0.0.0.0:11434 OLLAMA_INTEL_GPU:false OLLAMA_KEEP_ALIVE:5m0s OLLAMA_LLM_LIBRARY: OLLAMA_MAX_LOADED_MODELS:0 OLLAMA_MAX_QUEUE:512 OLLAMA_MODELS:/ollama OLLAMA_NOHISTORY:false OLLAMA_NOPRUNE:false OLLAMA_NUM_PARALLEL:0 OLLAMA_ORIGINS:[http://localhost https://localhost http://localhost:* https://localhost:* http://127.0.0.1 https://127.0.0.1 http://127.0.0.1:* https://127.0.0.1:* http://0.0.0.0 https://0.0.0.0 http://0.0.0.0:* https://0.0.0.0:* app://* file://* tauri://*] OLLAMA_RUNNERS_DIR: OLLAMA_SCHED_SPREAD:false OLLAMA_TMPDIR: ROCR_VISIBLE_DEVICES:]"
time=2025-03-13T06:16:01.244Z level=INFO source=images.go:753 msg="total blobs: 0"
time=2025-03-13T06:16:01.244Z level=INFO source=images.go:760 msg="total unused blobs removed: 0"
time=2025-03-13T06:16:01.245Z level=INFO source=routes.go:1172 msg="Listening on [::]:11434 (version 5f7b4a5)"
time=2025-03-13T06:16:01.247Z level=INFO source=payload.go:30 msg="extracting embedded files" dir=/tmp/ollama1022912176/runners
time=2025-03-13T06:16:03.584Z level=INFO source=payload.go:44 msg="Dynamic LLM libraries [cpu cuda_v12]"
time=2025-03-13T06:16:03.585Z level=INFO source=gpu.go:200 msg="looking for compatible GPUs"
time=2025-03-13T06:16:03.586Z level=WARN source=gpu.go:669 msg="unable to locate gpu dependency libraries"
time=2025-03-13T06:16:03.586Z level=WARN source=gpu.go:669 msg="unable to locate gpu dependency libraries"
double free or corruption (out)
SIGABRT: abort
PC=0xffff8af1f200 m=8 sigcode=18446744073709551610
signal arrived during cgo execution

goroutine 1 gp=0x40000021c0 m=8 mp=0x4000100808 [syscall]:
runtime.cgocall(0xe47020, 0x4000048658)
	runtime/cgocall.go:157 +0x44 fp=0x4000048620 sp=0x40000485e0 pc=0x421af4
github.com/ollama/ollama/gpu._Cfunc_nvcuda_init(0xffff28000b70, 0x400021d680)
	_cgo_gotypes.go:545 +0x30 fp=0x4000048650 sp=0x4000048620 pc=0x7cf140
github.com/ollama/ollama/gpu.LoadNVCUDAMgmt.func2(0xffff28000b70, 0x400021d680)
	github.com/ollama/ollama/gpu/gpu.go:552 +0x50 fp=0x4000048680 sp=0x4000048650 pc=0x7d40d0
github.com/ollama/ollama/gpu.LoadNVCUDAMgmt({0x40004b2880, 0x2, 0x410f760?})
	github.com/ollama/ollama/gpu/gpu.go:552 +0x10c fp=0x4000048840 sp=0x4000048680 pc=0x7d3bac
github.com/ollama/ollama/gpu.initCudaHandles()
	github.com/ollama/ollama/gpu/gpu.go:117 +0x390 fp=0x4000048a20 sp=0x4000048840 pc=0x7cfa20
github.com/ollama/ollama/gpu.GetGPUInfo()
	github.com/ollama/ollama/gpu/gpu.go:231 +0x358 fp=0x4000049b80 sp=0x4000048a20 pc=0x7d0408
github.com/ollama/ollama/server.Serve({0x39fc718, 0x40004b23e0})
	github.com/ollama/ollama/server/routes.go:1205 +0x510 fp=0x4000049d00 sp=0x4000049b80 pc=0xe1e640
github.com/ollama/ollama/cmd.RunServer(0x4000247400?, {0x426fa00?, 0x4?, 0x1158f5b?})
	github.com/ollama/ollama/cmd/cmd.go:1160 +0x44 fp=0x4000049d40 sp=0x4000049d00 pc=0xe3e7c4
github.com/spf13/cobra.(*Command).execute(0x40004ea608, {0x426fa00, 0x0, 0x0})
	github.com/spf13/cobra@v1.7.0/command.go:940 +0x654 fp=0x4000049e60 sp=0x4000049d40 pc=0x5d6204
github.com/spf13/cobra.(*Command).ExecuteC(0x400014b208)
	github.com/spf13/cobra@v1.7.0/command.go:1068 +0x320 fp=0x4000049f20 sp=0x4000049e60 pc=0x5d6970
github.com/spf13/cobra.(*Command).Execute(...)
	github.com/spf13/cobra@v1.7.0/command.go:992
github.com/spf13/cobra.(*Command).ExecuteContext(...)
	github.com/spf13/cobra@v1.7.0/command.go:985
main.main()
	github.com/ollama/ollama/main.go:12 +0x54 fp=0x4000049f40 sp=0x4000049f20 pc=0xe45e64
```

### Environment

Jetpack 6.2 on Ubuntu 22.04

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

johnnynunez, 2025-03-13:
do you try last docker from dockerhub?

tobsch, 2025-03-13:
@johnnynunez this one? https://hub.docker.com/r/dustynv/ollama

johnnynunez, 2025-03-13:
> [@johnnynunez](https://github.com/johnnynunez) this one? https://hub.docker.com/r/dustynv/ollama

[image]

tobsch, 2025-03-14:
@johnnynunez solved. seems to run on gpu now, sorry...!

closed 2025-03-14 by tobsch