dusty-nv/jetson-containers #947: Add support for Python 3.9

grep-safi, 2025-03-14:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Description

I notice there's no Docker images for L4T Pytorch Python 3.9: https://github.com/dusty-nv/jetson-containers/blob/master/packages/build/python/config.py

Can we add support for that version of Python? Happy to do it myself too if you have some guidance. Thanks!

### Use case

Our services run on Python 3.9, so we need Docker images that support this version.

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

grep-safi, 2025-03-14:
So I started looking into system setup and need a Jetson to actually build the image. Unfortunately I don't have direct access to one, as I am deploying to a system via a rather complicated release process.

Naively, it seems like adding Python 3.9 to that file should be all we need to build it?

johnnynunez, 2025-03-14:
Why?

Ubuntu 22.04 uses python 3.10
Ubuntu 24.04 uses pytjon 3.12

3.9 will be deprecated this year.
And also give wheels is hard to compile everything

grep-safi, 2025-03-18:
If it's too difficult that's fine. My company is slow to upgrade Python versions, we're still on 3.9 until the end of the year. I'll use the Python 3.8 image for now until we upgrade and then switch to the 3.10 image.

johnnynunez, 2025-03-18:
> If it's too difficult that's fine and feel free to close request. My company is slow to upgrade Python versions, we're still on 3.9 until the end of the year. I can use the Python 3.8 image for now until we upgrade and then switch to the 3.10 image.
> 
> If it's easy though, I'd appreciate it. Also just curious: why was 3.9 skipped? I see there are images for 3.8, 3.10, 3.11, etc. But no 3.9

use docker instead of

closed 2025-03-18 by johnnynunez