dusty-nv/jetson-containers #955: Building faiss v1.10.0 fails - gflags

m-gangloff, 2025-03-17:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### jetson-containers Component

Build

### Bug

Building the faiss 1.10.0 package with `jetson-containers build faiss` fails on the master branch.

The problem originates from [`find_package(gflags REQUIRED)`](https://github.com/facebookresearch/faiss/blob/189a9d4461233de798c2eb17fb6d354ceda542e6/perf_tests/CMakeLists.txt#L20) which was added after v1.7.4. This raises a CMake Error because it cannot find `gflags`:

```bash
CMake Error at perf_tests/CMakeLists.txt:20 (find_package):
  By not providing "Findgflags.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "gflags", but
  CMake did not find one.

  Could not find a package configuration file provided by "gflags" with any
  of the following names:

    gflagsConfig.cmake
    gflags-config.cmake

  Add the installation prefix of "gflags" to CMAKE_PREFIX_PATH or set
  "gflags_DIR" to a directory containing one of the above files.  If "gflags"
  provides a separate development package or SDK, be sure it has been
  installed.
```

I'm running on a Jetson Orin AGX with Jetpack 6.1 and cuda 12.2.

Here is the full log:

[faiss_r36.3.0-faiss.txt](https://github.com/user-attachments/files/19285778/faiss_r36.3.0-faiss.txt)

### Environment

Jetson Orin AGX with Jetpack 6.1 and cuda 12.2

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

mtinnes, 2025-03-19:
Same issue on Orin Nano with latest firmware.

```
CMake Error at perf_tests/CMakeLists.txt:20 (find_package):
  By not providing "Findgflags.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "gflags", but
  CMake did not find one.

  Could not find a package configuration file provided by "gflags" with any
  of the following names:

    gflagsConfig.cmake
    gflags-config.cmake

  Add the installation prefix of "gflags" to CMAKE_PREFIX_PATH or set
  "gflags_DIR" to a directory containing one of the above files.  If "gflags"
  provides a separate development package or SDK, be sure it has been
  installed.
```

Package: nvidia-jetpack
Source: nvidia-jetpack (6.2)
Version: 6.2+b77
Architecture: arm64
Maintainer: NVIDIA Corporation
Installed-Size: 194
Depends: nvidia-jetpack-runtime (= 6.2+b77), nvidia-jetpack-dev (= 6.2+b77)
Homepage: http://developer.nvidia.com/jetson
Priority: standard
Section: metapackages
Filename: pool/main/n/nvidia-jetpack/nvidia-jetpack_6.2+b77_arm64.deb
Size: 29298
SHA256: 70553d4b5a802057f9436677ef8ce255db386fd3b5d24ff2c0a8ec0e485c59cd
SHA1: 9deab64d12eef0e788471e05856c84bf2a0cf6e6
MD5sum: 4db65dc36434fe1f84176843384aee23
Description: NVIDIA Jetpack Meta Package
Description-md5: ad1462289bdbc54909ae109d1d32c0a8

dusty-nv, 2025-03-19:
It has been a few months since I built faiss or needed to - it is a powerful and relevant library though.   I would just get to patching the dockerfile to install gflags or whatever is needed to get it building again.  A PR would  be appreciated, and I will merge it.

Otherwise will fit when I get back to working on vectorDBs and RAG, I would update it then.

mtinnes, 2025-03-19:
Thanks Dusty. Just a note that my error originated from attempting to build a custom jetson container using the following:

`jetson-containers build --name custom-nanollm nano_llm whisper_trt`

I'm assuming FAISS is a dependency in this build chain.

Thanks for your work - it's a great help.

m-gangloff, 2025-03-20:
@mtinnes I'm currently waiting for https://github.com/pypa/distutils/pull/346 to be merged because it resolves another build problem with faiss due to an [deprecated import](https://github.com/facebookresearch/faiss/blob/0f2035cc8369d6475166a564e650fdf9c1cb3299/faiss/python/loader.py#L43). I'll open a PR with the bugfixes as soon as setuptools has been updated. In the meantime, you can simply use faiss v1.7.4 by editing it's [config.py](https://github.com/dusty-nv/jetson-containers/blob/6b5040e59a53a7fa32a21360df26f4e780d61978/packages/vectordb/faiss/config.py#L33) file

m-gangloff, 2025-03-20:
@mtinnes in the meantime, you can checkout https://github.com/m-gangloff/jetson-containers/tree/dev and build from there. I've pinned setuptools for now to 75.8.2 (the latest version where the AttributeError caused by the deprecated import does not occur) see https://github.com/pypa/setuptools/issues/4876#issuecomment-2718126135

m-gangloff, 2025-03-21:
PR was created https://github.com/dusty-nv/jetson-containers/pull/962
@mtinnes it's possible that you need to delete older images to make sure that the correct setuptools version is used

closed 2025-03-21 by m-gangloff