dusty-nv/jetson-containers #963: build onnxruntime:1.22  TensorrtExecutionProvider test failed

Asa-Dong, 2025-03-21:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Bug

**command**
jetson-containers build onnxruntime:1.22

**build log**
```
jetson-containers build onnxruntime:1.22

Namespace(packages=['onnxruntime:1.22'], name='', base='', multiple=False, build_flags='', build_args='', use_proxy=False, package_dirs=[''], list_packages=False, show_packages=False, skip_packages=[''], skip_errors=False, skip_tests=[''], test_only=[''], simulate=False, push='', logs='', verbose=False, no_github_api=False)

-- L4T_VERSION=36.4.3 JETPACK_VERSION=6.2 CUDA_VERSION=12.8 PYTHON_VERSION=3.10 LSB_RELEASE=22.04 (jammy)
-- jetson-containers onnxruntime:1.22

//................//

-- Building container onnxruntime:1.22-r36.4.3-onnxruntime_1.22

DOCKER_BUILDKIT=0 docker build --network=host --tag onnxruntime:1.22-r36.4.3-onnxruntime_1.22 \
--file /data/source/jetson-containers/packages/ml/onnxruntime/Dockerfile \
--build-arg BASE_IMAGE=onnxruntime:1.22-r36.4.3-onnx \
--build-arg ONNXRUNTIME_VERSION="1.22.0" \
--build-arg ONNXRUNTIME_BRANCH="main" \
--build-arg ONNXRUNTIME_FLAGS="--allow_running_as_root" \
/data/source/jetson-containers/packages/ml/onnxruntime \
2>&1 | tee /data/source/jetson-containers/logs/20250321_091025/build/onnxruntime_1.22-r36.4.3-onnxruntime_1.22.txt; exit ${PIPESTATUS[0]}

DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            BuildKit is currently disabled; enable it by removing the DOCKER_BUILDKIT=0
            environment-variable.

Sending build context to Docker daemon  26.62kB
Step 1/5 : ARG BASE_IMAGE
Step 2/5 : FROM ${BASE_IMAGE}
 ---> fe8ed7d43837
Step 3/5 : ARG ONNXRUNTIME_VERSION     ONNXRUNTIME_BRANCH     ONNXRUNTIME_FLAGS     FORCE_BUILD=off
 ---> Using cache
 ---> 40076dab129d
Step 4/5 : COPY install.sh build.sh /tmp/onnxruntime/
 ---> Using cache
 ---> 556a28290ee0
Step 5/5 : RUN /tmp/onnxruntime/install.sh || /tmp/onnxruntime/build.sh
 ---> Using cache
 ---> cb198f39b5ae
Successfully built cb198f39b5ae
Successfully tagged onnxruntime:1.22-r36.4.3-onnxruntime_1.22
-- Testing container onnxruntime:1.22-r36.4.3-onnxruntime_1.22 (onnxruntime:1.22/test.py)

docker run -t --rm --runtime=nvidia --network=host \
--volume /data/source/jetson-containers/packages/ml/onnxruntime:/test \
--volume /data/source/jetson-containers/data:/data \
--workdir /test \
onnxruntime:1.22-r36.4.3-onnxruntime_1.22 \
/bin/bash -c 'python3 test.py' \
2>&1 | tee /data/source/jetson-containers/logs/20250321_091025/test/onnxruntime_1.22-r36.4.3-onnxruntime_1.22_test.py.txt; exit ${PIPESTATUS[0]}

testing onnxruntime...
onnxruntime version: 1.22.0
ORT Build Info: git-branch=main, git-commit-id=0a6b05fb2d, build type=Release, cmake cxx flags: -Wno-unused-variable -I/usr/local/cuda/include -ffunction-sections -fdata-sections -DCPUINFO_SUPPORTED
execution providers:  ['TensorrtExecutionProvider', 'CUDAExecutionProvider', 'CPUExecutionProvider']
Namespace(model='/data/models/onnx/cat_dog_epoch_100/resnet18.onnx', model_url='https://nvidia.box.com/shared/static/zlvb4y43djygotpjn6azjhwu0r3j0yxc.gz', model_tar='cat_dog_epoch_100.tar.gz', runs=100, warmup=10, verbose=False)

loading /data/models/onnx/cat_dog_epoch_100/resnet18.onnx with 'TensorrtExecutionProvider'
2025-03-21 01:10:34.457049897 [E:onnxruntime:Default, provider_bridge_ort.cc:2022 TryGetProviderInfo_TensorRT] /opt/onnxruntime/onnxruntime/core/session/provider_bridge_ort.cc:1695 onnxruntime::Provider& onnxruntime::ProviderLibrary::Get() [ONNXRuntimeError] : 1 : FAIL : Failed to load library libonnxruntime_providers_tensorrt.so with error: libnvdla_compiler.so: cannot open shared object file: No such file or directory

*************** EP Error ***************
EP Error /opt/onnxruntime/onnxruntime/python/onnxruntime_pybind_state.cc:505 void onnxruntime::python::RegisterTensorRTPluginsAsCustomOps(onnxruntime::python::PySessionOptions&, const ProviderOptions&) Please install TensorRT libraries as mentioned in the GPU requirements page, make sure they're in the PATH or LD_LIBRARY_PATH, and that your GPU is supported.
 when using ['TensorrtExecutionProvider']
Falling back to ['CPUExecutionProvider'] and retrying.
****************************************
session providers:  ['CPUExecutionProvider']
Traceback (most recent call last):
  File "/test/test.py", line 128, in <module>
    perf[provider] = test_infer(provider, args.model, runs=args.runs, warmup=args.warmup, verbose=args.verbose)
  File "/test/test.py", line 65, in test_infer
    raise RuntimeError(f"missing provider '{provider}' from session providers {session_providers}")
RuntimeError: missing provider 'TensorrtExecutionProvider' from session providers ['CPUExecutionProvider']
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/data/source/jetson-containers/jetson_containers/build.py", line 120, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.build_args, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api, args.skip_packages)
  File "/data/source/jetson-containers/jetson_containers/container.py", line 158, in build_container
    test_container(container_name, pkg, simulate)
  File "/data/source/jetson-containers/jetson_containers/container.py", line 331, in test_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'docker run -t --rm --runtime=nvidia --network=host --volume /data/source/jetson-containers/packages/ml/onnxruntime:/test --volume /data/source/jetson-containers/data:/data --workdir /test onnxruntime:1.22-r36.4.3-onnxruntime_1.22 /bin/bash -c 'python3 test.py' 2>&1 | tee /data/source/jetson-containers/logs/20250321_091025/test/onnxruntime_1.22-r36.4.3-onnxruntime_1.22_test.py.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.

```

### Environment

**board version**
```
cat /etc/nv_tegra_release                                                                                                   130 ↵
# R36 (release), REVISION: 4.3, GCID: 38968081, BOARD: generic, EABI: aarch64, DATE: Wed Jan  8 01:49:37 UTC 2025
# KERNEL_VARIANT: oot
```

**cuda version**
```
nvidia-smi 
Fri Mar 21 09:22:51 2025       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 540.4.0                Driver Version: 540.4.0      CUDA Version: 12.6 
```

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

tokk-nv, 2025-04-10:
Created this PR
https://github.com/dusty-nv/jetson-containers/pull/1044

tokk-nv, 2025-04-11:
Hi @Asa-Dong,

Do you think you can test this now?

```
cd jetson-containers
git pull
git checkout dev
LSB_RELEASE=24.04 jetson-containers build --build-flags='--no-cache' onnxruntime
```