dusty-nv/jetson-containers #979: parameters on build time not always work correctly

OriNachum, 2025-03-23:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### jetson-containers Component

Build

### Bug

I think there is an issuer with skip tests - it should only run last build, but I see some internal builds running
Also, comma-delimited packages won't work correctly for me.

Also, --build-args doesn't correctly pass the environment variable.

Shouldn't be related - I'm using a jc shortcut instead of jetson-containers (I switched on initial install)
Other parameters work correctly for me

[image]

### Environment

AGX Orin with nvme build (agx-nvme) 

### Additional

I will take a look at this after current PR for spark-tts

### Are you willing to submit a PR?

- [x] Yes I'd like to help by submitting a PR!

ms1design, 2025-03-25:
Is this a real thing? I had no issues when using the `jetson-containers` cli

OriNachum, 2025-03-26:
Can you confirm tests are skipped correctly for you? It skipped for me only on --skip-tests all
So I refer to the other configurations