dusty-nv/jetson-containers #980: Failed to fetch information - all the time

venkatarameshgudapati-rios, 2025-03-24:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Question

Hi, 
I'm getting the following for the any operation.

Was wondering if im missing something.

nvidia@tegra-ubuntu:~/jetson-containers/jetson-containers$ jetson-containers list

Namespace(base='', build_args='', build_flags='', list_packages=True, logs='', multiple=False, name='', no_github_api=False, package_dirs=[''], packages=[], push='', show_packages=False, simulate=False, skip_errors=False, skip_packages=[''], skip_tests=[''], test_only=[''], use_proxy=False, verbose=False)

-- L4T_VERSION=35.3.1 JETPACK_VERSION=5.1.1 CUDA_VERSION=11.4 PYTHON_VERSION=3.8 LSB_RELEASE=20.04 (focal)
-- jetson-containers --list

Failed to fetch version information. Status code: 404
-- Copying /etc/nv_tegra_release to /home/nvidia/jetson-containers/jetson-containers/packages/llm/ollama/nv_tegra_release
Failed to fetch version information. Status code: 404

I'm upto date with the latest codebase

nvidia@tegra-ubuntu:~/jetson-containers/jetson-containers$ jetson-containers update
Already up to date.
nvidia@tegra-ubuntu:~/jetson-containers/jetson-containers$ 

Thanks in advance,
Venkat

venkatarameshgudapati-rios, 2025-03-24:
I do a curl to see if i have reachability, this looks good

nvidia@tegra-ubuntu:~/jetson-containers/jetson-containers$ curl -I https://api.github.com/repos/dusty-nv/jetson-containers/tags
HTTP/2 200 
date: Mon, 24 Mar 2025 03:19:09 GMT
content-type: application/json; charset=utf-8
content-length: 5
cache-control: public, max-age=60, s-maxage=60
vary: Accept,Accept-Encoding, Accept, X-Requested-With
etag: "992aaf92b487506974e4305db4a64326866f3ed48b2ba424cef8fbe4e28f72b0"
last-modified: Sun, 23 Mar 2025 22:44:48 GMT
x-github-media-type: github.v3; format=json
x-github-api-version-selected: 2022-11-28
access-control-expose-headers: ETag, Link, Location, Retry-After, X-GitHub-OTP, X-RateLimit-Limit, X-RateLimit-Remaining, X-RateLimit-Used, X-RateLimit-Resource, X-RateLimit-Reset, X-OAuth-Scopes, X-Accepted-OAuth-Scopes, X-Poll-Interval, X-GitHub-Media-Type, X-GitHub-SSO, X-GitHub-Request-Id, Deprecation, Sunset
access-control-allow-origin: *
strict-transport-security: max-age=31536000; includeSubdomains; preload
x-frame-options: deny
x-content-type-options: nosniff
x-xss-protection: 0
referrer-policy: origin-when-cross-origin, strict-origin-when-cross-origin
content-security-policy: default-src 'none'
server: github.com
accept-ranges: bytes
x-ratelimit-limit: 60
x-ratelimit-remaining: 22
x-ratelimit-reset: 1742788628
x-ratelimit-resource: core
x-ratelimit-used: 38
x-github-request-id: BE04:16900F:40A9A5A:437188A:67E0CF2D

nvidia@tegra-ubuntu:~/jetson-containers/jetson-containers$

tokk-nv, 2025-03-25:
It seems that packages under `smart-home` tries to download some `VERSION` file which does not exist any more so erroring out.

- `packages/smart-home/homeassistant-base/config.py`
- `packages/smart-home/wyoming/assist-microphone/config.py`
- `packages/smart-home/wyoming/piper/config.py`
- `packages/smart-home/wyoming/wyonming-whisper/config.py`

like, 

```
wanted_version = handle_text_request(f'https://raw.githubusercontent.com/rhasspy/wyoming-satellite/{branch}/wyoming_satellite/VERSION')
```
https://github.com/dusty-nv/jetson-containers/blob/master/packages/smart-home/wyoming/assist-microphone/config.py#L10C5-L10C140

Also `transformers` package's `config.py` also has this which tries to download a file but tends to fail.

```
data = handle_json_request('https://pypi.org/pypi/transformers/json')
```
https://github.com/dusty-nv/jetson-containers/blob/master/packages/llm/transformers/config.py#L26

tokk-nv, 2025-03-25:
Hi @venkatarameshgudapati-rios 

If possible, please try this.
https://github.com/dusty-nv/jetson-containers/pull/984