dusty-nv/jetson-containers #989: The hf.co/Qwen/Qwen2.5-VL-3B-Instruct model does not seem to work properly in the dustynv/vllm:0.7.4-r36.4.0-cu128-24.04 container.

nainaigetuide, 2025-03-27:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Bug

Based on the model deployment tutorial for hf.co/Qwen/Qwen2.5-VL-3B-Instruct on Jetson Orin from https://www.jetson-ai-lab.com/models.html, I have deployed the model. 
The command to load the model is 
`vllm serve /home/orin/disk/data/models/huggingface/qwen2.5_3b/       --host=0.0.0.0       --port=9000       --dtype=auto       --max-num-seqs=1       --max-model-len=1024       --chat-template-content-format=openai       --trust-remote-code       --gpu-memory-utilization=0.75       --uvicorn-log-level=debug       --quantization=bitsandbytes       --load-format=bitsandbytes`

When testing with the following command:

`curl http://0.0.0.0:9000/v1/chat/completions \ 
  -H "Content-Type: application/json" \
  -d '{
    "messages": [{
      "role": "user",
      "content": [{
        "type": "text",
        "text": "What is in this image?"
      },
      {
        "type": "image_url",
        "image_url": {
          "url": "https://upload.wikimedia.org/wikipedia/commons/thumb/d/dd/Gfp-wisconsin-madison-the-nature-boardwalk.jpg/2560px-Gfp-wisconsin-madison-the-nature-boardwalk.jpg"
        }
      }
    ]}],
    "max_tokens": 300
  }'
`

The following error occurred. Where is the issue?

Error in preprocessing prompt inputs
ERROR 03-26 02:23:27 [serving_chat.py:204] Traceback (most recent call last):
ERROR 03-26 02:23:27 [serving_chat.py:204]   File "/opt/venv/lib/python3.12/site-packages/vllm/entrypoints/openai/serving_chat.py", line 185, in create_chat_completion
ERROR 03-26 02:23:27 [serving_chat.py:204]     ) = await self._preprocess_chat(
ERROR 03-26 02:23:27 [serving_chat.py:204]         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ERROR 03-26 02:23:27 [serving_chat.py:204]   File "/opt/venv/lib/python3.12/site-packages/vllm/entrypoints/openai/serving_engine.py", line 411, in _preprocess_chat
ERROR 03-26 02:23:27 [serving_chat.py:204]     request_prompt = apply_hf_chat_template(
ERROR 03-26 02:23:27 [serving_chat.py:204]                      ^^^^^^^^^^^^^^^^^^^^^^^
ERROR 03-26 02:23:27 [serving_chat.py:204]   File "/opt/venv/lib/python3.12/site-packages/vllm/entrypoints/chat_utils.py", line 1092, in apply_hf_chat_template
ERROR 03-26 02:23:27 [serving_chat.py:204]     return tokenizer.apply_chat_template(
ERROR 03-26 02:23:27 [serving_chat.py:204]            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ERROR 03-26 02:23:27 [serving_chat.py:204]   File "/opt/venv/lib/python3.12/site-packages/transformers/tokenization_utils_base.py", line 1695, in apply_chat_template
ERROR 03-26 02:23:27 [serving_chat.py:204]     rendered_chat = compiled_template.render(
ERROR 03-26 02:23:27 [serving_chat.py:204]                     ^^^^^^^^^^^^^^^^^^^^^^^^^
ERROR 03-26 02:23:27 [serving_chat.py:204]   File "/opt/venv/lib/python3.12/site-packages/jinja2/environment.py", line 1295, in render
ERROR 03-26 02:23:27 [serving_chat.py:204]     self.environment.handle_exception()
ERROR 03-26 02:23:27 [serving_chat.py:204]   File "/opt/venv/lib/python3.12/site-packages/jinja2/environment.py", line 942, in handle_exception
ERROR 03-26 02:23:27 [serving_chat.py:204]     raise rewrite_traceback_stack(source=source)
ERROR 03-26 02:23:27 [serving_chat.py:204]   File "<template>", line 23, in top-level template code
ERROR 03-26 02:23:27 [serving_chat.py:204] TypeError: can only concatenate str (not "list") to str

### Environment

os: ubuntu 22.04
docker : dustynv/vllm:0.7.4-r36.4.0-cu128-24.04
jetpack version:6.1

### Additional

Is it because the 24.04 version image should not be used?

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

xiaohuihuige, 2025-04-01:
I encountered the same problem, it could be that the version has not been updated @nainaigetuide

nainaigetuide, 2025-04-01:
What version do you mean when you say that the version is not updated? Is it the operating system or the docker version? Did you resolve this issue? @xiaohuihuige

xiaohuihuige, 2025-04-02:
@dusty-nv  Hi Pen friend, I am glad to ask for your help. I have encountered the same situation and hope to receive some guidance

Environment：
    os: ubuntu 22.04
    docker : dustynv/vllm:0.7.4-r36.4.0-cu128-24.04
    jetpack version:6.1
    jetson  NX(8G)

root@jetson:/data/Qwen2.5-VL-3B-Instruct# vllm serve /data/Qwen2.5-VL-3B-Instruct --host=0.0.0.0       --port=9000       --dtype=auto       --max-num-seqs=1       --max-model-len=256       --chat-template-content-format=openai       --trust-remote-code       --gpu-memory-utilization=0.75       --uvicorn-log-level=debug       --quantization=bitsandbytes       --load-format=bitsandbytes
/opt/venv/lib/python3.12/site-packages/transformers/utils/hub.py:105: FutureWarning: Using `TRANSFORMERS_CACHE` is deprecated and will be removed in v5 of Transformers. Use `HF_HOME` instead.
  warnings.warn(
INFO 04-02 09:21:09 [__init__.py:256] Automatically detected platform cuda.
INFO 04-02 09:21:12 [api_server.py:940] vLLM API server version 0.7.4
INFO 04-02 09:21:12 [api_server.py:941] args: Namespace(subparser='serve', model_tag='/data/Qwen2.5-VL-3B-Instruct', config='', host='0.0.0.0', port=9000, uvicorn_log_level='debug', allow_credentials=False, allowed_origins=['*'], allowed_methods=['*'], allowed_headers=['*'], api_key=None, lora_modules=None, prompt_adapters=None, chat_template=None, chat_template_content_format='openai', response_role='assistant', ssl_keyfile=None, ssl_certfile=None, ssl_ca_certs=None, enable_ssl_refresh=False, ssl_cert_reqs=0, root_path=None, middleware=[], return_tokens_as_token_ids=False, disable_frontend_multiprocessing=False, enable_request_id_headers=False, enable_auto_tool_choice=False, tool_call_parser=None, tool_parser_plugin='', model='/data/Qwen2.5-VL-3B-Instruct', task='auto', tokenizer=None, hf_config_path=None, skip_tokenizer_init=False, revision=None, code_revision=None, tokenizer_revision=None, tokenizer_mode='auto', trust_remote_code=True, allowed_local_media_path=None, download_dir=None, load_format='bitsandbytes', config_format=<ConfigFormat.AUTO: 'auto'>, dtype='auto', kv_cache_dtype='auto', max_model_len=256, guided_decoding_backend='xgrammar', logits_processor_pattern=None, model_impl='auto', distributed_executor_backend=None, pipeline_parallel_size=1, tensor_parallel_size=1, enable_expert_parallel=False, max_parallel_loading_workers=None, ray_workers_use_nsight=False, block_size=None, enable_prefix_caching=None, disable_sliding_window=False, use_v2_block_manager=True, num_lookahead_slots=0, seed=None, swap_space=4, cpu_offload_gb=0, gpu_memory_utilization=0.75, num_gpu_blocks_override=None, max_num_batched_tokens=None, max_num_partial_prefills=1, max_long_partial_prefills=1, long_prefill_token_threshold=0, max_num_seqs=1, max_logprobs=20, disable_log_stats=False, quantization='bitsandbytes', rope_scaling=None, rope_theta=None, hf_overrides=None, enforce_eager=False, max_seq_len_to_capture=8192, disable_custom_all_reduce=False, tokenizer_pool_size=0, tokenizer_pool_type='ray', tokenizer_pool_extra_config=None, limit_mm_per_prompt=None, mm_processor_kwargs=None, disable_mm_preprocessor_cache=False, enable_lora=False, enable_lora_bias=False, max_loras=1, max_lora_rank=16, lora_extra_vocab_size=256, lora_dtype='auto', long_lora_scaling_factors=None, max_cpu_loras=None, fully_sharded_loras=False, enable_prompt_adapter=False, max_prompt_adapters=1, max_prompt_adapter_token=0, device='auto', num_scheduler_steps=1, use_tqdm_on_load=True, multi_step_stream_outputs=True, scheduler_delay_factor=0.0, enable_chunked_prefill=None, speculative_model=None, speculative_model_quantization=None, num_speculative_tokens=None, speculative_disable_mqa_scorer=False, speculative_draft_tensor_parallel_size=None, speculative_max_model_len=None, speculative_disable_by_batch_size=None, ngram_prompt_lookup_max=None, ngram_prompt_lookup_min=None, spec_decoding_acceptance_method='rejection_sampler', typical_acceptance_sampler_posterior_threshold=None, typical_acceptance_sampler_posterior_alpha=None, disable_logprobs_during_spec_decoding=None, model_loader_extra_config=None, ignore_patterns=[], preemption_mode=None, served_model_name=None, qlora_adapter_name_or_path=None, show_hidden_metrics_for_version=None, otlp_traces_endpoint=None, collect_detailed_traces=None, disable_async_output_proc=False, scheduling_policy='fcfs', scheduler_cls='vllm.core.scheduler.Scheduler', override_neuron_config=None, override_pooler_config=None, compilation_config=None, kv_transfer_config=None, worker_cls='auto', worker_extension_cls='', generation_config='auto', override_generation_config=None, enable_sleep_mode=False, calculate_kv_scales=False, additional_config=None, enable_reasoning=False, reasoning_parser=None, disable_log_requests=False, max_log_len=None, disable_fastapi_docs=False, enable_prompt_tokens_details=False, enable_server_load_tracking=False, dispatch_function=<function ServeSubcommand.cmd at 0xffff342fb1a0>)
INFO 04-02 09:21:12 [api_server.py:209] Started engine process with PID 189
/opt/venv/lib/python3.12/site-packages/transformers/utils/hub.py:105: FutureWarning: Using `TRANSFORMERS_CACHE` is deprecated and will be removed in v5 of Transformers. Use `HF_HOME` instead.
  warnings.warn(
INFO 04-02 09:21:18 [__init__.py:256] Automatically detected platform cuda.
INFO 04-02 09:21:24 [config.py:583] This model supports multiple tasks: {'embed', 'generate', 'classify', 'reward', 'score'}. Defaulting to 'generate'.
[... cut ...]
  File "/opt/venv/lib/python3.12/site-packages/vllm/entrypoints/openai/api_server.py", line 233, in build_async_engine_client_from_engine_args
    raise RuntimeError(
RuntimeError: Engine process failed to start. See stack trace for the root cause.
root@jetson:/data/Qwen2.5-VL-3B-Instruct# /usr/lib/python3.12/multiprocessing/resource_tracker.py:254: UserWarning: resource_tracker: There appear to be 1 leaked semaphore objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '

nainaigetuide, 2025-04-02:
@dusty-nv Do you want to build a new docker image that can run vllm and qwen2.5vl correctly?

dusty-nv, 2025-04-04:
> [@dusty-nv](https://github.com/dusty-nv) Do you want to build a new docker image that can run vllm and qwen2.5vl correctly?

Hi @nainaigetuide , the latest vLLM 0.8.3 container image is here - `dustynv/vllm:0.8.3-r36.4.0-cu128-24.04`
`

nainaigetuide, 2025-04-07:
Thank you. @dusty-nv

closed 2025-04-07 by nainaigetuide

xiaohuihuige, 2025-04-08:
My board is Jetson NX (8G) and Jetpack 6.2. I did not use Docker to run VLLM, but rather https://pypi.jetson-ai-lab.dev/jp6/cu126 The package inside was configured with the environment. At that time, when running Qwen2-VL-2B-Structure-GPTQ-Int4 using vllm, there was not enough memory, triggering OOM. The parameters of vllm were also configured as gpu memory tilization 0.7 cpu offloading gb 0 swap space 8. I tried different configurations and applied for enough swapping memory, but I was able to run it successfully using pytorch. Could you please advise on how to solve the problem of vllm memory explosion triggering oom? Thank you very much
@dusty-nv @nainaigetuide @michaelgruner