dusty-nv/jetson-containers #995: fast cdr failing with cmake error in ros2-humble-desktop build

smithmh6, 2025-03-29:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Question

fast cdr is failing with cmake errors in colcon build. Can someone help me with this? I am on the latest commit on the master branch.

Docker build command:
```sh
jetson-containers build --name ros:humble-desktop ros:humble-desktop
```

Version info:
-- L4T_VERSION=36.4.0
-- JETPACK_VERSION=6.1
-- CUDA_VERSION=12.6
-- PYTHON_VERSION=3.10
-- LSB_RELEASE=22.04 (jammy)

stdout:
```sh
#All required rosdeps installed successfully
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> gtest_vendor
Starting >>> ament_pycodestyle
Starting >>> ament_cppcheck
Starting >>> osrf_pycommon
Starting >>> google_benchmark_vendor
Starting >>> osrf_testing_tools_cpp
Starting >>> fastcdr
Starting >>> iceoryx_hoofs
Starting >>> urdfdom_headers
/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/dist.py:289: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
/usr/local/lib/python3.10/dist-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
!!

        ********************************************************************************
        Please consider removing the following classifiers in favor of a SPDX license expression:

        License :: OSI Approved :: Apache Software License

        See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
        ********************************************************************************

!!
  self._finalize_license_expression()
/usr/local/lib/python3.10/dist-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
!!

        ********************************************************************************
        Please consider removing the following classifiers in favor of a SPDX license expression:

        License :: OSI Approved :: Apache Software License

        See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
        ********************************************************************************

!!
  self._finalize_license_expression()
/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/dist.py:289: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
/usr/local/lib/python3.10/dist-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
!!

        ********************************************************************************
        Please consider removing the following classifiers in favor of a SPDX license expression:

        License :: OSI Approved :: Apache Software License

        See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
        ********************************************************************************

!!
  self._finalize_license_expression()
/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/dist.py:289: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
/usr/local/lib/python3.10/dist-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
!!

        ********************************************************************************
        Please consider removing the following classifiers in favor of a SPDX license expression:

        License :: OSI Approved :: Apache Software License

        See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
        ********************************************************************************

!!
  self._finalize_license_expression()
/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/dist.py:289: UserWarning: Unknown distribution option: 'test_suite'
  warnings.warn(msg)
/usr/local/lib/python3.10/dist-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
!!

        ********************************************************************************
        Please consider removing the following classifiers in favor of a SPDX license expression:

        License :: OSI Approved :: Apache Software License

        See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
        ********************************************************************************

!!
  self._finalize_license_expression()
--- stderr: fastcdr
CMake Error at CMakeLists.txt:18 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

---
Failed   <<< fastcdr [1.57s, exited with code 1]
Aborted  <<< urdfdom_headers [1.57s]
Aborted  <<< gtest_vendor [1.61s]
Aborted  <<< google_benchmark_vendor [2.00s]
Aborted  <<< iceoryx_hoofs [2.06s]
Aborted  <<< osrf_testing_tools_cpp [2.08s]
Aborted  <<< ament_pycodestyle [2.22s]
Aborted  <<< ament_lint [2.23s]
Aborted  <<< ament_package [2.23s]
Aborted  <<< ament_cppcheck [2.24s]
Aborted  <<< osrf_pycommon [2.24s]

Summary: 0 packages finished [4.70s]
  1 package failed: fastcdr
  10 packages aborted: ament_cppcheck ament_lint ament_package ament_pycodestyle google_benchmark_vendor gtest_vendor iceoryx_hoofs osrf_pycommon osrf_testing_tools_cpp urdfdom_headers
  5 packages had stderr output: fastcdr google_benchmark_vendor gtest_vendor osrf_testing_tools_cpp urdfdom_headers
  317 packages not processed
The command '/bin/bash -c /tmp/ros2_build.sh' returned a non-zero code: 1
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/mnt/ssd/src/jetson-containers/jetson_containers/build.py", line 112, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.build_args, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api, args.skip_packages)
  File "/mnt/ssd/src/jetson-containers/jetson_containers/container.py", line 147, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host --tag ros:humble-desktop-l4t-r36.4.0-ros_humble-desktop --file /mnt/ssd/src/jetson-containers/packages/ros/Dockerfile.ros2 --build-arg BASE_IMAGE=ros:humble-desktop-l4t-r36.4.0-cmake --build-arg ROS_VERSION="humble" --build-arg ROS_PACKAGE="desktop" /mnt/ssd/src/jetson-containers/packages/ros 2>&1 | tee /mnt/ssd/src/jetson-containers/logs/20250329_122847/build/ros_humble-desktop-l4t-r36.4.0-ros_humble-desktop.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.
```

I did try adding
```sh
 -DCMAKE_POLICY_VERSION_MINIMUM=3.5
```
as suggested in the cmake output, but that also fails.

mtinnes, 2025-03-31:
Similar issue building the whisper_trt package:

```
Successfully built torch2trt
Installing collected packages: torch2trt

Successfully installed torch2trt-0.5.0
Removed build tracker: '/tmp/pip-build-tracker-ee5mf_pf'
CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

-- Configuring incomplete, errors occurred!
The command '/bin/sh -c cd /opt &&     git clone --depth=1 https://github.com/NVIDIA-AI-IOT/torch2trt &&     cd torch2trt &&     cp /tmp/patches/flattener.py torch2trt &&     pip3 install . &&     sed 's|^set(CUDA_ARCHITECTURES.*|#|g' -i CMakeLists.txt &&     sed 's|Catch2_FOUND|False|g' -i CMakeLists.txt &&     cmake -B build -DCUDA_ARCHITECTURES=${CUDA_ARCHITECTURES} . &&     cmake --build build --target install &&     ldconfig &&     pip3 install nvidia-pyindex &&     pip3 install onnx-graphsurgeon' returned a non-zero code: 1
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/mtinnes/jetson-containers/jetson_containers/tag.py", line 58, in <module>
    image = find_container(args.packages[0], prefer_sources=args.prefer, disable_sources=args.disable, user=args.user, quiet=args.quiet)
  File "/home/mtinnes/jetson-containers/jetson_containers/container.py", line 542, in find_container
    return build_container('', package) #, simulate=True)
  File "/home/mtinnes/jetson-containers/jetson_containers/container.py", line 151, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host --tag whisper_trt:r36.4.0-torch2trt --file /home/mtinnes/jetson-containers/packages/pytorch/torch2trt/Dockerfile --build-arg BASE_IMAGE=whisper_trt:r36.4.0-tensorrt /home/mtinnes/jetson-containers/packages/pytorch/torch2trt 2>&1 | tee /home/mtinnes/jetson-containers/logs/20250331_165337/build/whisper_trt_r36.4.0-torch2trt.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.
-- Error:  return code 1
```

smithmh6, 2025-04-01:
I tried this again on a fresh jetson orin with these specs using commit id 29fb8672:
-- L4T_VERSION=36.3.0
-- JETPACK_VERSION=6.0
-- CUDA_VERSION=12.2
-- PYTHON_VERSION=3.10
-- LSB_RELEASE=22.04 (jammy)

I received the following error again:

```sh
#All required rosdeps installed successfully
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> gtest_vendor
Starting >>> ament_pycodestyle
Starting >>> ament_cppcheck
Starting >>> osrf_pycommon
Starting >>> google_benchmark_vendor
Starting >>> osrf_testing_tools_cpp
Starting >>> fastcdr
Starting >>> iceoryx_hoofs
Starting >>> urdfdom_headers
/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/dist.py:289: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
/usr/local/lib/python3.10/dist-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
!!

        ********************************************************************************
        Please consider removing the following classifiers in favor of a SPDX license expression:

        License :: OSI Approved :: Apache Software License

        See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
        ********************************************************************************

!!
  self._finalize_license_expression()
/usr/local/lib/python3.10/dist-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
!!

        ********************************************************************************
        Please consider removing the following classifiers in favor of a SPDX license expression:

        License :: OSI Approved :: Apache Software License

        See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
        ********************************************************************************

!!
  self._finalize_license_expression()
/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/dist.py:289: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
/usr/local/lib/python3.10/dist-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
!!

        ********************************************************************************
        Please consider removing the following classifiers in favor of a SPDX license expression:

        License :: OSI Approved :: Apache Software License

        See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
        ********************************************************************************

!!
  self._finalize_license_expression()
/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/dist.py:289: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
/usr/local/lib/python3.10/dist-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
!!

[... 50 lines cut ...]
  10 packages aborted: ament_cppcheck ament_lint ament_package ament_pycodestyle google_benchmark_vendor gtest_vendor iceoryx_hoofs osrf_pycommon osrf_testing_tools_cpp urdfdom_headers
  5 packages had stderr output: fastcdr google_benchmark_vendor gtest_vendor osrf_testing_tools_cpp urdfdom_headers
  317 packages not processed
The command '/bin/bash -c /tmp/ros2_build.sh' returned a non-zero code: 1
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/hsmith/jetson-containers/jetson_containers/build.py", line 112, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.build_args, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api, args.skip_packages)
  File "/home/hsmith/jetson-containers/jetson_containers/container.py", line 147, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host --tag ros:humble-desktop-l4t-r36.3.0-ros_humble-desktop --file /home/hsmith/jetson-containers/packages/ros/Dockerfile.ros2 --build-arg BASE_IMAGE=ros:humble-desktop-l4t-r36.3.0-cmake --build-arg ROS_VERSION="humble" --build-arg ROS_PACKAGE="desktop" /home/hsmith/jetson-containers/packages/ros 2>&1 | tee /home/hsmith/jetson-containers/logs/20250401_122528/build/ros_humble-desktop-l4t-r36.3.0-ros_humble-desktop.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.
```

It seems the problem is with fastcdr. Can someone advise on what knobs I can turn to try to troubleshoot this?  I also tried adding 

```sh
-DCMAKE_POLICY_VERSION_MINIMUM=3.5
```

To the ros2_build.sh script in the ros package, but the error persists.

smithmh6, 2025-04-01:
> Similar issue building the whisper_trt package:
>
> ```
> Successfully built torch2trt
> Installing collected packages: torch2trt
>
> Successfully installed torch2trt-0.5.0
> Removed build tracker: [...]

I was able to solve this issue by changing the `ros2_build.sh`  to use cmake 3.28 by building it from the source. It appears that cmake 4.0 was released on March 28th and officially no longer supports <3.5. When using `apt-get install cmake` it defaults to the current version.

 This is what fixed the issue for me:
```sh
# add the ROS deb repo to the apt sources list
apt-get update
apt-get install -y --no-install-recommends \
                build-essential \
                curl \
                libssl-dev \
                wget \
                gnupg2 \
                lsb-release \
                ca-certificates

wget https://cmake.org/files/v3.28/cmake-3.28.0.tar.gz
tar -xzvf cmake-3.28.0.tar.gz
cd cmake-3.28.0
echo "Current directory: $(pwd)"
./bootstrap
make
make install
echo "Installed cmake version: "
cmake --version
```