dusty-nv/jetson-containers #1002: https://pypi.jetson-ai-lab.dev down?

chrmel, 2025-04-01:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### jetson-containers Component

Build

### Bug

It seems https://pypi.jetson-ai-lab.dev/ is down. It seems similar to #811 

When building i.e. the python package i get the following error (also cant be reached manually):
```bash
$ curl https://pypi.jetson-ai-lab.dev
curl: (7) Failed to connect to pypi.jetson-ai-lab.dev port 443: Connection refused
```
```bash
$ ./build.sh --name=ros:jp5.1.2 ros:humble-ros-base
.
.
.
Looking in indexes: https://pypi.jetson-ai-lab.dev/jp5/cu114
1 location(s) to search for versions of pip:
* https://pypi.jetson-ai-lab.dev/jp5/cu114/pip/
Fetching project page and analyzing links: https://pypi.jetson-ai-lab.dev/jp5/cu114/pip/
Getting page https://pypi.jetson-ai-lab.dev/jp5/cu114/pip/
Found index url https://pypi.jetson-ai-lab.dev/jp5/cu114/
Starting new HTTPS connection (1): pypi.jetson-ai-lab.dev:443
Incremented Retry for (url='/jp5/cu114/pip/'): Retry(total=4, connect=None, read=None, redirect=None, status=None)
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff7de27e80>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp5/cu114/pip/
Starting new HTTPS connection (2): pypi.jetson-ai-lab.dev:443
Incremented Retry for (url='/jp5/cu114/pip/'): Retry(total=3, connect=None, read=None, redirect=None, status=None)
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff7de53310>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp5/cu114/pip/
Starting new HTTPS connection (3): pypi.jetson-ai-lab.dev:443
Incremented Retry for (url='/jp5/cu114/pip/'): Retry(total=2, connect=None, read=None, redirect=None, status=None)
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff7de53400>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp5/cu114/pip/
Starting new HTTPS connection (4): pypi.jetson-ai-lab.dev:443
Incremented Retry for (url='/jp5/cu114/pip/'): Retry(total=1, connect=None, read=None, redirect=None, status=None)
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff7de535b0>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp5/cu114/pip/
Starting new HTTPS connection (5): pypi.jetson-ai-lab.dev:443
Incremented Retry for (url='/jp5/cu114/pip/'): Retry(total=0, connect=None, read=None, redirect=None, status=None)
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff7de53760>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp5/cu114/pip/
Starting new HTTPS connection (6): pypi.jetson-ai-lab.dev:443
Could not fetch URL https://pypi.jetson-ai-lab.dev/jp5/cu114/pip/: connection error: HTTPSConnectionPool(host='pypi.jetson-ai-lab.dev', port=443): Max retries exceeded with url: /jp5/cu114/pip/ (Caused by NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff7de53910>: Failed to establish a new connection: [Errno 111] Connection refused')) - skipping
Skipping link: not a file: https://pypi.jetson-ai-lab.dev/jp5/cu114/pip/
Given no hashes to check 0 links for project 'pip': discarding no candidates
ERROR: Could not find a version that satisfies the requirement pip (from versions: none)
ERROR: No matching distribution found for pip
Exception information:
Traceback (most recent call last):
  File "/tmp/tmpoxct7xt8/pip.zip/pip/_vendor/resolvelib/resolvers.py", line 397, in resolve
    self._add_to_criteria(self.state.criteria, r, parent=None)
  File "/tmp/tmpoxct7xt8/pip.zip/pip/_vendor/resolvelib/resolvers.py", line 174, in _add_to_criteria
    raise RequirementsConflicted(criterion)
pip._vendor.resolvelib.resolvers.RequirementsConflicted: Requirements conflict: SpecifierRequirement('pip')
.
.
.
```

Thank you and Regards!

### Environment

current master (2025-04-01)

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

space192, 2025-04-01:
same issue :/

closed 2025-04-01 by johnnynunez

JadTawil77, 2025-07-10:
This is down again!

PawelPeczek-Roboflow, 2025-07-11:
@johnnynunez thanks a lot - this is live-saving index <3

karl-mobiusindustry, 2025-07-11:
@johnnynunez We appreciate the diligence on resolving this issue. Wanted to additionally ask if your team has a more exact time on when the server will be back online again?

johnnynunez, 2025-07-11:
> [@johnnynunez](https://github.com/johnnynunez) We appreciate the diligence on resolving this issue. Wanted to additionally ask if your team has a more exact time on when the server will be back online again?

I don’t know. I’m simple random dev guy. But i hope soon(around next week)

PawelPeczek-Roboflow, 2025-07-14:
@johnnynunez would it be possible for Roboflow to mirror the index?

evikzub, 2025-07-22:
pypi.jetson-ai-lab.dev is not available. When can it be up?

AnatolySamaris, 2025-08-10:
it's still down. When it's gonna be up?

johnnynunez, 2025-08-10:
> it's still down. When it's gonna be up?
Never. That url it was from Dusty