dusty-nv/jetson-containers #1007: opencv build fails

chrmel, 2025-04-02:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### jetson-containers Component

Build

### Bug

Trying to build `opencv:4.8.1` it **fails** at the testing stage (so do 4.5.0, 4.10.0, 4.11.0). `cv2` cannot be imported.

As u can see from the log, the BUILD looks successful. Image tagged, and no error msg (all packages before look fine too).
```bash
.
.
.
-- Building container opencv:4.5.0-r35.4.1-opencv_4.5.0

DOCKER_BUILDKIT=0 docker build --network=host --tag opencv:4.5.0-r35.4.1-opencv_4.5.0 \
--file /home/jetson/jetson-containers/packages/opencv/Dockerfile \
--build-arg BASE_IMAGE=opencv:4.5.0-r35.4.1-numpy \
--build-arg OPENCV_VERSION="4.5.0" \
--build-arg OPENCV_PYTHON="4.x" \
--build-arg CUDA_ARCH_BIN="7.2,8.7" \
/home/jetson/jetson-containers/packages/opencv \
2>&1 | tee /home/jetson/jetson-containers/logs/20250402_082543/build/opencv_4.5.0-r35.4.1-opencv_4.5.0.txt; exit ${PIPESTATUS[0]}

DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            BuildKit is currently disabled; enable it by removing the DOCKER_BUILDKIT=0
            environment-variable.

Sending build context to Docker daemon  35.33kB
Step 1/6 : ARG BASE_IMAGE
Step 2/6 : FROM ${BASE_IMAGE}
 ---> ba1398fd6736
Step 3/6 : ARG OPENCV_VERSION     OPENCV_PYTHON     OPENCV_URL     CUDA_ARCH_BIN     FORCE_BUILD=off
 ---> Using cache
 ---> e7e1ece6d361
Step 4/6 : ENV OPENCV_VERSION=${OPENCV_VERSION}     OPENCV_URL=${OPENCV_URL}
 ---> Using cache
 ---> 1f4311b34752
Step 5/6 : COPY build.sh      install.sh      install_deps.sh      install_deb.sh      patches.diff      /tmp/opencv/
 ---> Using cache
 ---> e1b24815416b
Step 6/6 : RUN cd /tmp/opencv && ./install.sh || ./build.sh || echo "BUILD FAILED (OpenCV ${OPENCV_VERSION})"
 ---> Using cache
 ---> 23b8142ac110
Successfully built 23b8142ac110
Successfully tagged opencv:4.5.0-r35.4.1-opencv_4.5.0
-- Testing container opencv:4.5.0-r35.4.1-opencv_4.5.0 (opencv:4.5.0/test.py)

docker run -t --rm --runtime=nvidia --network=host \
--volume /home/jetson/jetson-containers/packages/opencv:/test \
--volume /home/jetson/jetson-containers/data:/data \
--workdir /test \
opencv:4.5.0-r35.4.1-opencv_4.5.0 \
/bin/bash -c 'python3 test.py' \
2>&1 | tee /home/jetson/jetson-containers/logs/20250402_082543/test/opencv_4.5.0-r35.4.1-opencv_4.5.0_test.py.txt; exit ${PIPESTATUS[0]}

testing OpenCV...
Traceback (most recent call last):
  File "test.py", line 4, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
[... 62 lines cut ...]
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/jetson/jetson-containers/jetson_containers/build.py", line 120, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.build_args, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api, args.skip_packages)
  File "/home/jetson/jetson-containers/jetson_containers/container.py", line 151, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)  
  File "/usr/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host --tag opencv:4.8.1-r35.4.1-opencv_4.8.1 --file /home/jetson/jetson-containers/packages/opencv/Dockerfile --build-arg BASE_IMAGE=opencv:4.8.1-r35.4.1-numpy --build-arg OPENCV_VERSION="4.8.1" --build-arg OPENCV_PYTHON="4.x" --build-arg CUDA_ARCH_BIN="7.2,8.7" /home/jetson/jetson-containers/packages/opencv 2>&1 | tee /home/jetson/jetson-containers/logs/20250402_083006/build/opencv_4.8.1-r35.4.1-opencv_4.8.1.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 100.
```

Help would be much appreciated, thanks!

### Environment

-- L4T_VERSION=35.4.1 JETPACK_VERSION=5.1.2 CUDA_VERSION=11.4 PYTHON_VERSION=3.8 LSB_RELEASE=20.04 (focal)

### Additional

Is there a way to tell the build to use a pre-built opencv container instead of building it?
`jetson-containers build --base=dustynv/opencv:r35.2.1 ros:humble-ros-base` still tries to build `opencv:4.8.1`

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

sei-jgwohlbier, 2025-05-14:
Did you solve this?

chrmel, 2025-05-15:
Unfortunately not.

johnnynunez, 2025-06-27:
> same here.... `Encountered exception while importing cv2: libtesseract.so.5: cannot open shared object file: No such file or directory`

you have to add that dependencies in install_deps. We only support right now jp6 and jp7

closed 2025-07-03 by johnnynunez