dusty-nv/jetson-containers #1013: pypi.jetson-ai-lab.dev unreachable

samanhmi, 2025-04-03:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Bug

I am trying to run the Whisper model on the AGX Orin board with Jetpack 6.2 and CUDA 12.6 installed. However after running the _jetson-containers run $(autotag whisper)_  command, the process stops due to a 443 error in the connection. It seems like the destination host is unreachable or link is not valid. 

### Environment

Fetching project page and analyzing links: https://pypi.jetson-ai-lab.dev/jp6/cu126/pip/
Getting page https://pypi.jetson-ai-lab.dev/jp6/cu126/pip/
Found index url https://pypi.jetson-ai-lab.dev/jp6/cu126/
Starting new HTTPS connection (1): pypi.jetson-ai-lab.dev:443
Incremented Retry for (url='/jp6/cu126/pip/'): Retry(total=4, connect=None, read=None, redirect=None, status=None)
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffb7512770>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /jp6/cu126/pip/
****

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

dusty-nv, 2025-04-03:
Hi @samanhmi, are you able to try again in the event it was a temporary issue?

samanhmi, 2025-04-07:
> Hi [@samanhmi](https://github.com/samanhmi), are you able to try again in the event it was a temporary issue?

Hi, I tried again and the error still persists.

samanhmi, 2025-04-09:
I am adding this comment here to let you know that the issue is resolved. The reason was that my Network was blocking the DNS of this site even though I didn't have firewall enabled on my system.

closed 2025-04-11 by johnnynunez