dusty-nv/jetson-containers #1016: `smart-home` group packages trying to download non-existing VERSION file

tokk-nv, 2025-04-04:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### jetson-containers Component

Build

### Bug

Today, if you try to run a `jetson-containers build` command, you will see lines like the one below indicating failures in downloading files like VERSION, which seems to no longer exist on each respective repo.

You can replicate this by

```bash
git clone https://github.com/dusty-nv/jetson-containers/
./jetson-containers/install.sh
jetson-containers build --list
```

and you will see lines like following.

```text
[info] GET https://api.github.com/repos/home-assistant/docker-base/tags (attempt 1)                                                                                                                                                   
[info] GET https://api.github.com/repos/closeio/ciso8601/tags (attempt 1)                                                                                                                                                             
[info] GET https://api.github.com/repos/home-assistant-libs/psutil-home-assistant/tags (attempt 1)                                                                                                                                    
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-faster-whisper/v2.4.0/wyoming_faster_whisper/VERSION (attempt 1)                                                                                                   
[info] GET https://version.home-assistant.io/stable.json (attempt 1)                                                                                                                                                                  
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-piper/v1.5.2/wyoming_piper/VERSION (attempt 1)                                                                                                                     
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-openwakeword/master/wyoming_openwakeword/VERSION (attempt 1)                                                                                                       
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-satellite/master/wyoming_satellite/VERSION (attempt 1)                                                                                                             
[info] GET https://pypi.org/pypi/transformers/json (attempt 1)                                                                                                                                                                        
[warn] Request error on https://pypi.org/pypi/transformers/json: ('Connection aborted.', ConnectionResetError(104, 'Connection reset by peer'))                                                                                       
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-openwakeword/v1.10.0/wyoming_openwakeword/VERSION (attempt 1)                                                                                                      
[WARN] Failed to fetch https://raw.githubusercontent.com/rhasspy/wyoming-satellite/master/wyoming_satellite/VERSION: 404 Client Error: Not Found for url: https://raw.githubusercontent.com/rhasspy/wyoming-satellite/master/wyoming_s
atellite/VERSION                                                                                                                                                                                                                      
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-piper/v1.5.0/wyoming_piper/VERSION (attempt 1)                                                                                                                     
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-faster-whisper/v2.3.0/wyoming_faster_whisper/VERSION (attempt 1)                                                                                                   
[INFO] Fetching: https://raw.githubusercontent.com/home-assistant/docker-base/2025.02.0/ubuntu/build.yaml (attempt 1)                                                                                                                 
[WARN] Failed to fetch https://raw.githubusercontent.com/home-assistant/docker-base/2025.02.0/ubuntu/build.yaml: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))                         
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-faster-whisper/v2.2.0/wyoming_faster_whisper/VERSION (attempt 1)                                                                                                   
[WARN] Failed to fetch https://raw.githubusercontent.com/rhasspy/wyoming-faster-whisper/v2.2.0/wyoming_faster_whisper/VERSION: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))           
[info] GET https://pypi.org/pypi/transformers/json (attempt 2)                                                                                                                                                                        
[info] GET https://api.github.com/repos/huggingface/transformers/tags (attempt 1)                                                                                                                                                     
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-satellite/master/wyoming_satellite/VERSION (attempt 2)                                                                                                             
[WARN] Failed to fetch https://raw.githubusercontent.com/rhasspy/wyoming-satellite/master/wyoming_satellite/VERSION: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))                     
[INFO] Fetching: https://raw.githubusercontent.com/home-assistant/docker-base/2025.02.0/ubuntu/build.yaml (attempt 2)                                                                                                                 
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-faster-whisper/v2.2.0/wyoming_faster_whisper/VERSION (attempt 2)                                                                                                   
[WARN] Failed to fetch https://raw.githubusercontent.com/rhasspy/wyoming-faster-whisper/v2.2.0/wyoming_faster_whisper/VERSION: ('Connection aborted.', ConnectionResetError(104, 'Connection reset by peer'))                         
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-satellite/master/wyoming_satellite/VERSION (attempt 3)                                                                                                             
[WARN] Failed to fetch https://raw.githubusercontent.com/rhasspy/wyoming-satellite/master/wyoming_satellite/VERSION: ('Connection aborted.', ConnectionResetError(104, 'Connection reset by peer'))                                   
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-satellite/master/wyoming_satellite/VERSION (attempt 1)                                                                                                             
[WARN] Failed to fetch https://raw.githubusercontent.com/rhasspy/wyoming-satellite/master/wyoming_satellite/VERSION: 404 Client Error: Not Found for url: https://raw.githubusercontent.com/rhasspy/wyoming-satellite/master/wyoming_s
atellite/VERSION                                                                                                                                                                                                                      
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-faster-whisper/v2.2.0/wyoming_faster_whisper/VERSION (attempt 3)                                                                                                   
[WARN] Failed to fetch https://raw.githubusercontent.com/rhasspy/wyoming-faster-whisper/v2.2.0/wyoming_faster_whisper/VERSION: ('Connection aborted.', ConnectionResetError(104, 'Connection reset by peer'))                         
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-satellite/master/wyoming_satellite/VERSION (attempt 2)                                                                                                             
[... cut ...]
mba', 'kokoro-tts:onnx', 'speaches']                                                                                                                                                                                                  
-- Building container speaches:r36.4.3-cu128-24.04-build-essential      
```

### Environment

Tested on the following

- Jetson AGX Orin 64GB Developer Kit
- JetPack 6.2
- (NVIDIA corporate network)

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

ms1design, 2025-04-04:
Hi @tokk-nv, thanks for reminding me with this, I'm aware of this upstream issue. I briefly checked the repos and it seems that @Mike changed build process for `wyoming_satellite` and `wyoming_faster_whisper` packagesand that is hitting us hard here. 

I can patch that, but ETA is for now unknown atm.

ms1design, 2025-04-04:
@tokk-nv https://github.com/dusty-nv/jetson-containers/pull/1018 should resolve issues:

```log
jetson-containers build wyoming-assist-microphone

Namespace(packages=['wyoming-assist-microphone'], name='', base='', multiple=False, build_flags='', build_args='', use_proxy=False, package_dirs=[''], list_packages=False, show_packages=False, skip_packages=[''], skip_errors=False, skip_tests=[''], test_only=[''], simulate=False, push='', logs='', verbose=False, version=False, no_github_api=False)

┌──────────────┬─────────┬─────────────────┬───────┐
│ L4T_VERSION  │ 36.4.0  │ JETPACK_VERSION │ 6.1   │
├──────────────┼─────────┼─────────────────┼───────┤
│ CUDA_VERSION │ 12.6    │ PYTHON_VERSION  │ 3.1   │
├──────────────┼─────────┼─────────────────┼───────┤
│ SYSTEM_ARCH  │ aarch64 │ LSB_RELEASE     │ 22.04 │
└──────────────┴─────────┴─────────────────┴───────┘

$ jetson-containers wyoming-assist-microphone

[info] GET https://api.github.com/repos/home-assistant/docker-base/tags (attempt 1)
[info] GET https://api.github.com/repos/home-assistant-libs/psutil-home-assistant/tags (attempt 1)
[info] GET https://pypi.org/pypi/transformers/json (attempt 1)
[info] GET https://version.home-assistant.io/stable.json (attempt 1)
[info] GET https://api.github.com/repos/closeio/ciso8601/tags (attempt 1)
[info] GET https://api.github.com/repos/rhasspy/wyoming-satellite/tags (attempt 1)
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-faster-whisper/v2.4.0/wyoming_faster_whisper/VERSION (attempt 1)
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-openwakeword/master/wyoming_openwakeword/VERSION (attempt 1)
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-piper/v1.5.2/wyoming_piper/VERSION (attempt 1)
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-faster-whisper/v2.3.0/wyoming_faster_whisper/VERSION (attempt 1)
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-openwakeword/v1.10.0/wyoming_openwakeword/VERSION (attempt 1)
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-piper/v1.5.0/wyoming_piper/VERSION (attempt 1)
[INFO] Fetching: https://raw.githubusercontent.com/home-assistant/docker-base/2025.02.0/ubuntu/build.yaml (attempt 1)
[info] GET https://api.github.com/repos/huggingface/transformers/tags (attempt 1)
[INFO] Fetching: https://raw.githubusercontent.com/rhasspy/wyoming-faster-whisper/v2.2.0/wyoming_faster_whisper/VERSION (attempt 1)
-- Building containers  ['build-essential', 'homeassistant-base', 'pip_cache', 'python:3.11', 'wyoming-assist-microphone']
```

synesthesiam, 2025-04-04:
I'm slowly migrating everything to pyproject.toml since I've started getting deprecation warnings about using setup.py.

closed 2025-04-04 by ms1design