dusty-nv/jetson-containers #1017: Add MCPs support

OriNachum, 2025-04-04:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Description

MCPs are all the rage now.
It is the accepted integration protocol between tools and models.
OpenAI, Cursor IDE, Windsurf, VSCode and Amazon support it already.

We need 2 parts:

- [ ] At least 1 LLM server to support MCPs, preferably vllm (mainly verify)
- [ ] Add MCPs dockers as a POC for making jetson devices tools for LLMs.

### Use case

Jetson Super Nano as a smart camera for an LLM on another device AGX or a computer 

Jetson AGX Orin as training tool for models by agents

Jetson based robots as a swarm controlled by a master agent

### Additional

This places Jetson-containers on the edge (p/i) of technology.
MCPs had become a standard that engineers will look to.
Jetson-containers itself is a standard for Jetson and we need to place a high bar for what a library should cover.

### Are you willing to submit a PR?

- [x] Yes I'd like to help by submitting a PR!

ms1design, 2025-04-04:
@OriNachum something to skyrocket you into right direction: https://github.com/sparfenyuk/mcp-proxy

What I can recommend here is to follow how NabuCasa implemented MCP Server in Home Assistant, allowing to integrate any Agent/LLM with their MCP Server following the industry standard. In addition to that users can also connect other official/community MCP Servers that are aggregated under the main HA MCE Server.

OriNachum, 2025-04-18:
MCP actually worked for me
What didn't work was Vllm models that supported the tool use 

I'll test again with Ollama, after my own support for Codex - it already works (a global adapter) but missing a major component to make it powerful.

Currently no one intend to support it properly