dusty-nv/jetson-containers #1055: Failing to build text-generation-webui on Jetpack 6.2

AndBobsYourUncle, 2025-04-14:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Bug

Running `jetson-containers build $(autotag text-generation-webui)` gives me this output:

```
┌─────────────────────────────────────────────────────┐
│ > BUILDING  text-generation-webui:r36.4.3-auto_gptq │
└─────────────────────────────────────────────────────┘

DOCKER_BUILDKIT=0 docker build --network=host \
  --tag text-generation-webui:r36.4.3-auto_gptq \
  --file /home/nicholas/jetson-containers/packages/llm/gptqmodel/Dockerfile \
  --build-arg BASE_IMAGE=text-generation-webui:r36.4.3-transformers \
  --build-arg GPTQMODEL_VERSION="2.2.0" \
  --build-arg GPTQMODEL_BRANCH="2.2.0" \
   /home/nicholas/jetson-containers/packages/llm/gptqmodel

[12:15:34] [14/21] Building auto_gptq (text-generation-webui:r36.4.3-auto_gptq)
DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            BuildKit is currently disabled; enable it by removing the DOCKER_BUILDKIT=0
            environment-variable.

Sending build context to Docker daemon  16.38kB
Step 1/5 : ARG BASE_IMAGE
Step 2/5 : FROM ${BASE_IMAGE}
 ---> 1e3bb01eb6a3
Step 3/5 : ARG GPTQMODEL_VERSION     GPTQMODEL_BRANCH     FORCE_BUILD="off"
 ---> Running in f09b0faa66f6
 ---> Removed intermediate container f09b0faa66f6
 ---> ce4360793adc
Step 4/5 : COPY build.sh /tmp/build.sh
 ---> 8b4ef4d15f53
Step 5/5 : RUN pip3 install gekko &&     pip3 install auto-gptq==${GPTQMODEL_VERSION} ||     /tmp/build.sh
 ---> Running in 976ef6c4fef1
Using pip 25.0.1 from /usr/local/lib/python3.10/dist-packages/pip (python 3.10)
Looking in indexes: https://pypi.jetson-ai-lab.dev/jp6/cu126
Collecting gekko
  Downloading https://pypi.jetson-ai-lab.dev/root/pypi/%2Bf/78e/6e51e1e3ea2ad/gekko-1.3.0-py3-none-any.whl (13.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 13.2/13.2 MB 11.6 MB/s eta 0:00:00
Requirement already satisfied: numpy>=1.8 in /usr/local/lib/python3.10/dist-packages (from gekko) (1.26.4)
Installing collected packages: gekko
Successfully installed gekko-1.3.0
Using pip 25.0.1 from /usr/local/lib/python3.10/dist-packages/pip (python 3.10)
Looking in indexes: https://pypi.jetson-ai-lab.dev/jp6/cu126
ERROR: Could not find a version that satisfies the requirement auto-gptq==2.2.0 (from versions: 0.7.1+cu126, 0.8.0.dev0+cu126)
ERROR: No matching distribution found for auto-gptq==2.2.0
/bin/sh: 1: /tmp/build.sh: Permission denied
The command '/bin/sh -c pip3 install gekko &&     pip3 install auto-gptq==${GPTQMODEL_VERSION} ||     /tmp/build.sh' returned a non-zero code: 126
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/nicholas/jetson-containers/jetson_containers/tag.py", line 58, in <module>
    image = find_container(args.packages[0], prefer_sources=args.prefer, disable_sources=args.disable, user=args.user, quiet=args.quiet)
  File "/home/nicholas/jetson-containers/jetson_containers/container.py", line 569, in find_container
    return build_container('', package) #, simulate=True)
  File "/home/nicholas/jetson-containers/jetson_containers/container.py", line 177, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host   --tag text-generation-webui:r36.4.3-auto_gptq   --file /home/nicholas/jetson-containers/packages/llm/gptqmodel/Dockerfile   --build-arg BASE_IMAGE=text-generation-webui:r36.4.3-transformers   --build-arg GPTQMODEL_VERSION="2.2.0"   --build-arg GPTQMODEL_BRANCH="2.2.0"    /home/nicholas/jetson-containers/packages/llm/gptqmodel 2>&1 | tee /home/nicholas/jetson-containers/logs/20250414_115844/build/text-generation-webui_r36.4.3-auto_gptq.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 126.
-- Error:  return code 1

Namespace(packages=[], name='', base='', multiple=False, build_flags='', build_args='', use_proxy=False, package_dirs=[''], list_packages=False, show_packages=False, skip_packages=[''], skip_errors=False, skip_tests=[''], test_only=[''], simulate=False, push='', no_github_api=False, log_dir=None, log_level=None, log_colors=None, log_status=None, debug=False, verbose=False, version=False)

┌───────────────────────┬────────────────────────┐
│ L4T_VERSION   36.4.3  │ JETPACK_VERSION  6.2   │
│ CUDA_VERSION  12.6    │ PYTHON_VERSION   3.10  │
│ SYSTEM_ARCH   aarch64 │ LSB_RELEASE      22.04 │
└───────────────────────┴────────────────────────┘

$ jetson-containers

[12:15:39] Failed building:

Traceback (most recent call last):
  File "/home/nicholas/jetson-containers/jetson_containers/build.py", line 129, in <module>
    build_container(**vars(args))
  File "/home/nicholas/jetson-containers/jetson_containers/container.py", line 64, in build_container
    raise ValueError("must specify at least one package to build")
ValueError: must specify at least one package to build
```

It looks like this is the relevant error:
```
Looking in indexes: https://pypi.jetson-ai-lab.dev/jp6/cu126
ERROR: Could not find a version that satisfies the requirement auto-gptq==2.2.0 (from versions: 0.7.1+cu126, 0.8.0.dev0+cu126)
ERROR: No matching distribution found for auto-gptq==2.2.0
/bin/sh: 1: /tmp/build.sh: Permission denied
```

Not sure how I can get this to build properly on my machine. Is Jetpack 6.2 not supported yet?

### Environment

┌───────────────────────┬────────────────────────┐
│ L4T_VERSION   36.4.3  │ JETPACK_VERSION  6.2   │
│ CUDA_VERSION  12.6    │ PYTHON_VERSION   3.10  │
│ SYSTEM_ARCH   aarch64 │ LSB_RELEASE      22.04 │
└───────────────────────┴────────────────────────┘

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

Hugo415, 2025-04-17:
Hi @AndBobsYourUncle  and @Flank1e ,
Me too. It's seems like auto-gptq==2.2.0 version error. 
I used this command to check the latest tag.
`$  git ls-remote --tags https://github.com/AutoGPTQ/AutoGPTQ`
```
4f84d2168424b3cec3441a3378ad5366f68354b3        refs/tags/v0.0.2
73606a3a161ae563e693f0ee3304eb214be13336        refs/tags/v0.0.3
82815471fdf9b6e6877cecf98c7ee67efd547eec        refs/tags/v0.0.4
a87d0b6e08508e7541f282cd4639c5ee08f6fcd7        refs/tags/v0.0.5
374ce21066142cec9977464f0c037afd0d3049bf        refs/tags/v0.1.0
6a37f7c26623e48995dd0573cc329a0d75a7ce70        refs/tags/v0.2.0
7206705456cb7c97b411db6d1b9b6fcaa3887a04        refs/tags/v0.2.1
53cf80a9894212908bc2b5d82677c1ab67eeecaa        refs/tags/v0.2.2
45576f0933f5e9ef7c1617006d5db359e1669155        refs/tags/v0.3.0
a7167b108c438f570938f0ced46a52fe515f4a59        refs/tags/v0.3.2
69cdfe80fdd3802151b815c6009c33047310e2c7        refs/tags/v0.4.0
eea67b7e130b92605f5ace33cc93e8b95e9c12a5        refs/tags/v0.4.1
62187bf843726bdd473e0685ec9098639fe340c0        refs/tags/v0.4.2
ef442d9f70c50a9c1d9f2d4aa8622d1b89c83ea2        refs/tags/v0.4.2^{}
200844d26191ba841e7af4bb6ffaa1cb9d7cb137        refs/tags/v0.5.0
c70d29f05586d0221626e167c4e75265efc24967        refs/tags/v0.5.0^{}
ddbfb04ca3567b0dd2a9b24d1b8167841d6e9b22        refs/tags/v0.5.1
f4c9681020f23909faecc50ddaf24ee538e90395        refs/tags/v0.5.1^{}
cae3caf0a73e2842d0a0b2348482888dbb07f470        refs/tags/v0.6.0
323950bcb14059d7154109ee8b189f16cfc925d3        refs/tags/v0.6.0^{}
a59d3d50349e06200b733ff70113932b260bee37        refs/tags/v0.7.0
bb4c06fe795980f26748687d70d16a1ac7c61298        refs/tags/v0.7.0^{}
b2e918c9968ea46796acc87d801121dd25c9301e        refs/tags/v0.7.1
b8b41270f3dd5d657aee049d7b64538ab5dd244e        refs/tags/v0.7.1^{}
```

And I edited the  `config.py` to change gptqmodel version as below(2.2.0 --> 0.7.1):
```
def gptqmodel(version, branch=None, default=False):
    pkg = package.copy()

    pkg['name'] = f'gptqmodel:{version}'

    if not branch:
        branch = version

    pkg['build_args'] = {
        ''
        'GPTQMODEL_VERSION': version,
        'GPTQMODEL_BRANCH': branch,
    }

    builder = pkg.copy()

    builder['name'] = f'auto_gptq:{version}-builder'
    builder['build_args'] = {**pkg['build_args'], **{'FORCE_BUILD': 'on'}}

    if default:
        pkg['alias'] = 'auto_gptq'
        builder['alias'] = 'auto_gptq:builder'

    return pkg, builder

package = [
    #gptqmodel('2.2.0', default=True),
      gptqmodel('0.7.1', default=True),
]
```
Although it can be success to` build text-generation-webui:r36.4.3-auto_gptq`, But the next step `TESTING  text-generation-webui:r36.4.3-auto_gptq` still get the error by `from gptqmodel import GPTQModel ModuleNotFoundError: No module named 'gptqmodel'` .

Flank1e, 2025-04-17:
Thanks, one step forward - will try this later this evening and have a bit of a dig

khicks01, 2025-04-19:
Same issues here.  Where is the config.py file located?

johnnynunez, 2025-06-27:
it is fixed! That library was deprecated and move to new one

closed 2025-06-27 by johnnynunez