dusty-nv/jetson-containers #1056: Whether I can build TensorRT-LLM v0.18.1 on Jetson Orin?

gesanqiu, 2025-04-15:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Question

Hi, thanks for your amazing works. I have noticed that TensorRT-LLM had released their cpp source codes since v0.18.0, and this allows me to transplant newest model(like Qwen2.5-VL) to TensorRT-LLM.
According to https://github.com/dusty-nv/jetson-containers/blob/b56a41b60f46db65314eaf1e2192725cdd465bf0/packages/llm/tensorrt_optimizer/tensorrt_llm/config.py#L50 we can know jetson-container supports v0.17 without applying any patch right now. I'm wandering whether I can build v0.18.1 in the same container environment?

johnnynunez, 2025-04-15:
> ### Search before asking
> * [x]  I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.
> 
> ### Question
> Hi, thanks for your amazing works. I have noticed that TensorRT-LLM had released their cpp source codes since v0.18.0, and this allows me to transplant newest model(like Qwen2.5-VL) to TensorRT-LLM. According to
> 
> [jetson-containers/packages/llm/tensorrt_optimizer/tensorrt_llm/config.py](https://github.com/dusty-nv/jetson-containers/blob/b56a41b60f46db65314eaf1e2192725cdd465bf0/packages/llm/tensorrt_optimizer/tensorrt_llm/config.py#L50)
> 
> Line 50 in [b56a41b](/dusty-nv/jetson-containers/commit/b56a41b60f46db65314eaf1e2192725cdd465bf0)
> 
>  tensorrt_llm('0.17.0', requires='>=cu128', default=True), 
> we can know jetson-container supports v0.17 without applying any patch right now. I'm wandering whether I can build v0.18.1 in the same container environment?
> ### Additional
> _No response_

We have not tested yet

gesanqiu, 2025-04-16:
@johnnynunez Thanks for your reply, I think I can try this, any suggestion? BTW, I found that the latest JetPack is 6.2 with cuda12.6, when will the new version release so I can try TensorRT-LLM 0.17 first?

johnnynunez, 2025-06-27:
I've tried and fails, we will look that in the future with Thor

closed 2025-06-27 by johnnynunez