dusty-nv/jetson-containers #1068: Tests will not run with --gpus=all

WhoWouldaThunk, 2025-04-23:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### jetson-containers Component

Build

### Bug

 Build command:
	LSB_RELEASE=24.04 CUDA_VERSION=12.8 jetson-containers build --name=jetson_base_cu128  build-essential cuda cudnn cmake rust tensorrt onnxruntime python numpy pycuda pytorch torchvision torchaudio opencv ffmpeg
 
Fails (attempting to run tests on the cuda container with this command):
	docker run -t --rm --gpus=all --network=host   --env NVIDIA_DRIVER_CAPABILITIES=all   --volume /jetson-containers/packages/cuda/cuda:/test   --volume /jetson-containers/data:/data   --workdir /test   jetson_base_cu128:latest     /bin/bash -c '/bin/bash test.sh' 2>&1 | tee /jetson-containers/logs/20250422_193833/test/log_test.sh.txt

Which throws this error:
	docker: Error response from daemon: failed to create task for container: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error running prestart hook #0: exit status 1, stdout: , stderr: Auto-detected mode as 'csv'
	invoking the NVIDIA Container Runtime Hook directly (e.g. specifying the docker --gpus flag) is not supported. Please use the NVIDIA Container Runtime (e.g. specify the --runtime=nvidia flag) instead.: unknown

If you remove:
	--gpus=all or replace it with --runtime=nvidia 

Then tests will run fine.

I'm not sure why it this error just started showing up as it was working fine last week. Although I did a new clone of the repo today 
and then ran apt update to see if it would fix this issue.

Solution:
	If you look at line 338 in jetson-containers/jetson_containers/container.py, this is ultimately where the docker run command is created. 
	Change it to --runtime=nvidia ?

### Environment

NVIDIA Container Toolkit CLI version 1.17.5

R36 (release), REVISION: 4.3

Client: Docker Engine - Community
 Version:           28.1.1
 API version:       1.49
 Go version:        go1.23.8
 Git commit:        4eba377
 Built:             Fri Apr 18 09:52:37 2025
 OS/Arch:           linux/arm64
 Context:           default

Server: Docker Engine - Community
 Engine:
  Version:          28.1.1
  API version:      1.49 (minimum version 1.24)
  Go version:       go1.23.8
  Git commit:       01f442b
  Built:            Fri Apr 18 09:52:37 2025
  OS/Arch:          linux/arm64
  Experimental:     false
 containerd:
  Version:          1.7.27
  GitCommit:        05044ec0a9a75232cad458027ca83437aae3f4da
 runc:
  Version:          1.2.5
  GitCommit:        v1.2.5-0-g59923ef
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

johnnynunez, 2025-04-23:
> Thanks for reporting this, I had made this change to use --gpus=all and it was working on JP6 here.  I will revert the change 👍
> […](#)

FYI: I don't have this issue

ToeiRei, 2025-04-26:
Docker daemon.json
```
{
    "runtimes": {
        "nvidia": {
            "path": "/usr/bin/nvidia-container-runtime",
            "runtimeArgs": []
         } 
    },
    "default-runtime": "nvidia"  <---- THIS is important
}
```

Without the default runtime, things go haywire fast.

hillct, 2025-04-30:
It's worth pointing out here that 6.2 as flashed via the SDK manager 30 days ago to an AGX Orin Deb kit(32GB) does not ship with the proper `/etc/docker/daemon.json` file/configuration out of the box.

ToeiRei, 2025-04-30:
None of them are shipped with proper settings. It's usually a full day for me to reflash and set up my Orin NX and figure what I forgot to configure. Alternatively I could do my own images or even better write some documentation

hillct, 2025-05-02:
> It's noted in the jetson-containers setup docs along with other tweaks for setting up docker that actually have nothing to do with jetson-containers: https://github.com/dusty-nv/jetson-containers/blob/master/docs/setup.md#docker-default-runtime

@dusty-nv I'm quite sure nobody was blaming you or the other contributors to the jetson-containers project. To the contrary, the documentation has vastly improved over the past two years which appears to be all your work. 

The thing worth considering here is that the hardware should ship with the nvidia runtime already being the default, and your being an employee of Nvidia, puts you in a better position to coordinate with the image maintainers to introduce that default to future images.

Certainly, the remaining jetson-containers setup tips are very helpful and more workflow specific so should possibly not be deploymant defaults. While we're discussing those tips, and absent opening a separate issue, I've found it helpful to when adding an M.2 SSD, to create an lvm2 volume group, containing only that drive, and in the VG, createing two volumes labeled HOME and DOCKER, where each are mounted by label, for convenience, at /home and /var/lib/docker respectively. This approach has several advantages:

- Flexibility of volume size
- Convenience of configuration
  - Use of labels for ease of mounting references
  - Avoiding changing docker data root, for ease of cross-deployment compatibility
- Convenience of migration
  - Use of LVM2 facilitates migation of storage media as SSD capacity requirements change, with negligible overhead

ToeiRei, 2025-05-02:
> The thing worth considering here is that the hardware should ship with the nvidia runtime already being the default, and your being an employee of Nvidia, puts you in a better position to coordinate with the image maintainers to introduce that default to future images.

You can get nvidia jetson devices from various other vendors as well like seedstudio who had provided their own images that you had to build. To keep the jetson-containers as universal as possible, the documentation is key. You cannot ask other venors who "package" a carrier board to set defaults.

In the end everyone of us has their private notes as well on setting up their particiular piece of hardware (at least I have.)

hillct, 2025-05-02:
I didn't realize the likes of seed studio actually provided their own images. That seems sub-optimal.