dusty-nv/jetson-containers #1075: https://pypi.jetson-ai-lab.dev broken again

bl33m, 2025-04-30:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Bug

Trying to build a container and it seems https://pypi.jetson-ai-lab.dev is down again as I am getting the same errors as 
#1002 and https://forums.developer.nvidia.com/t/pypi-jetson-ai-lab-dev-is-down/323972. Is there an alternative link at the moment? Thanks! 

### Environment

Jetson AGX Orin 64 GB,

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

cyhasuka, 2025-04-30:
The same! Hope u guys fix it asap, THX!

annto99, 2025-04-30:
Same issue here! I thought I was getting crazy. Hope you'll fix it soon 🙏

dusty-nv, 2025-04-30:
It is back up again, sorry about that!