dusty-nv/jetson-containers #1076: Build containers` test.sh being copied to final container erroneously causing final test stage failures

hillct, 2025-04-30:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### jetson-containers Component

Other, Build

### Bug

It appears, in the event that a final package has a sitabled test stage where `test.sh` was renamed to `test.sh.disabled` (as can be seen in the llm/ollama package build), the test.sh from the immediate dependency build container is being eroniously copied in, or not replaced by the lower level container test stage script. In the example case of the Ollama package, the `test.sh` script from the cuda container (dependency most previously built) is installed in the runtime  and not replaced, because the runtime test script was disabled (a commit from 3 months back). The result in this example case is that the test of the ollama runtime container fails, because it's testing for the presence of the nvcc compiler, which is (correctly) not present

[ollama-build-logs-partial.txt](https://github.com/user-attachments/files/19981680/ollama-build-logs-partial.txt)

 in the ollama runtime container.

### Environment

Tested on Jetson AGX Orin Dev Kit 32GB
Jetson-container Master Commit 3e1c8f040ddf03fba65e44c0bc087f99ef2af253
Command: `jetson-containers run --name ollama $(autotag ollama)`

### Additional

It should be noted that this appears to be a result of the recent refactoring of the build scripts, not specific to the Ollama package, but would probably be exhibited when building any package which depends upon the CUDA build container, and has a locally disabled `test.sh` script. It likely also occurts in similar cases with other dependency build containers where comonents are being rested for in the build environment that are not present in the runtime environment. Whily I haven't sorted through the newly refactored build scripts, the optimal solution is to not copy the test.sh script from a dependency build, intto the next container, and properly handle the lack of a local test.sh script (ie: test.sh.disabled)

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

dusty-nv, 2025-04-30:
Thanks, there is no `COPY` or `ADD` in the ollama dockerfile, so I not sure how they are getting in there:  https://github.com/dusty-nv/jetson-containers/blob/master/packages/llm/ollama/Dockerfile

The ollama build keeps changing upstream and we put a best-effort into keeping it going.

hillct, 2025-04-30:
Certainly reviewing the docker file was the first thing I did. If you review the specific error message relating to line 12 of test.sh and compare it to test.sh that exists within the cuda build container, you'll see that nvcc version check, which is what's actually failing. Somewhere within the build scripts or directory structure. There must be a wild card copy taking place. It's just a matter of tracking it down.