dusty-nv/jetson-containers #1089: RuntimeError building jetson-inference: torchvision::nms does not exist

m6f13, 2025-05-07:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### jetson-containers Component

Build

### Bug

when running: jetson-containers build jetson-inference

Error in: Testing torchvision (jetson-inference:r36.4-cu126-22.04-torchvision)

error:
`testing torchvision...
Traceback (most recent call last):
  File "/test/test.py", line 12, in <module>
    import torchvision
  File "/usr/local/lib/python3.10/dist-packages/torchvision/__init__.py", line 10, in <module>
    from torchvision import _meta_registrations, datasets, io, models, ops, transforms, utils  # usort:skip
  File "/usr/local/lib/python3.10/dist-packages/torchvision/_meta_registrations.py", line 164, in <module>
    def meta_nms(dets, scores, iou_threshold):
  File "/usr/local/lib/python3.10/dist-packages/torch/library.py", line 828, in register
    use_lib._register_fake(op_name, func, _stacklevel=stacklevel + 1)
  File "/usr/local/lib/python3.10/dist-packages/torch/library.py", line 198, in _register_fake
    handle = entry.fake_impl.register(func_to_register, source)
  File "/usr/local/lib/python3.10/dist-packages/torch/_library/fake_impl.py", line 31, in register
    if torch._C._dispatch_has_kernel_for_dispatch_key(self.qualname, "Meta"):
RuntimeError: operator torchvision::nms does not exist
[18:56:05] Failed building:  jetson-inference`

### Environment

[18:55:43] [10/19] Building torchvision (jetson-inference:r36.4-cu126-22.04-torchvision)                                                       9 stages completed in 02m52s at 18:55:43
DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            BuildKit is currently disabled; enable it by removing the DOCKER_BUILDKIT=0
            environment-variable.

Sending build context to Docker daemon   34.3kB
Step 1/6 : ARG BASE_IMAGE
Step 2/6 : FROM ${BASE_IMAGE}
 ---> 9b96cd52febf
Step 3/6 : ARG TORCHVISION_VERSION     FORCE_BUILD=off
 ---> Running in 7f73324fa1ab
 ---> Removed intermediate container 7f73324fa1ab
 ---> 46a22af19b46
Step 4/6 : COPY build.sh install.sh /tmp/torchvision/
 ---> c53b10238a4e
Step 5/6 : RUN apt-get update &&     apt-get install -y --no-install-recommends         libjpeg-dev         libpng-dev         zlib1g-dev       && rm -rf /var/lib/apt/lists/*      && apt-get clean
 ---> Running in 13ddbf393858
Get:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease [270 kB]
Get:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease [128 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease [127 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease [129 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports jammy/universe arm64 Packages [17.2 MB]
Get:6 http://ports.ubuntu.com/ubuntu-ports jammy/restricted arm64 Packages [24.2 kB]
Get:7 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 Packages [1,758 kB]
Get:8 http://ports.ubuntu.com/ubuntu-ports jammy/multiverse arm64 Packages [224 kB]
Get:9 http://ports.ubuntu.com/ubuntu-ports jammy-updates/restricted arm64 Packages [3,737 kB]
Get:10 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 Packages [2,882 kB]
Get:11 http://ports.ubuntu.com/ubuntu-ports jammy-updates/multiverse arm64 Packages [32.8 kB]
Get:12 http://ports.ubuntu.com/ubuntu-ports jammy-updates/universe arm64 Packages [1,516 kB]
Get:13 http://ports.ubuntu.com/ubuntu-ports jammy-backports/universe arm64 Packages [33.3 kB]
Get:14 http://ports.ubuntu.com/ubuntu-ports jammy-backports/main arm64 Packages [82.7 kB]
Get:15 http://ports.ubuntu.com/ubuntu-ports jammy-security/universe arm64 Packages [1,222 kB]
Get:16 http://ports.ubuntu.com/ubuntu-ports jammy-security/restricted arm64 Packages [3,593 kB]
Get:17 http://ports.ubuntu.com/ubuntu-ports jammy-security/multiverse arm64 Packages [26.5 kB]
Get:18 http://ports.ubuntu.com/ubuntu-ports jammy-security/main arm64 Packages [2,578 kB]
Fetched 35.6 MB in 5s (7,342 kB/s)
[... 108 lines cut ...]
  File "/usr/local/lib/python3.10/dist-packages/torchvision/_meta_registrations.py", line 164, in <module>
    def meta_nms(dets, scores, iou_threshold):
  File "/usr/local/lib/python3.10/dist-packages/torch/library.py", line 828, in register
    use_lib._register_fake(op_name, func, _stacklevel=stacklevel + 1)
  File "/usr/local/lib/python3.10/dist-packages/torch/library.py", line 198, in _register_fake
    handle = entry.fake_impl.register(func_to_register, source)
  File "/usr/local/lib/python3.10/dist-packages/torch/_library/fake_impl.py", line 31, in register
    if torch._C._dispatch_has_kernel_for_dispatch_key(self.qualname, "Meta"):
RuntimeError: operator torchvision::nms does not exist
[18:56:05] Failed building:  jetson-inference

Traceback (most recent call last):
  File "/home/martin/git/jetson-containers/jetson_containers/build.py", line 129, in <module>
    build_container(**vars(args))
  File "/home/martin/git/jetson-containers/jetson_containers/container.py", line 243, in build_container
    test_container(container_name, pkg, simulate)
  File "/home/martin/git/jetson-containers/jetson_containers/container.py", line 429, in test_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'docker run -t --rm --network=host --runtime=nvidia   --volume /home/martin/git/jetson-containers/packages/pytorch/torchvision:/test   --volume /home/martin/git/jetson-containers/data:/data   --workdir /test   jetson-inference:r36.4-cu126-22.04-torchvision     /bin/bash -c 'python3 test.py' 2>&1 | tee /home/martin/git/jetson-containers/logs/20250507_185248/test/jetson-inference_r36.4-cu126-22.04-torchvision_test.py.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

UserName-wang, 2025-05-08:
I have the same issue when trying to build a container of nemo. same OS environment and latest jetson-container.

adamjuhasz, 2025-05-12:
This also affect the vllm container

YuZhong-Chen, 2025-05-15:
I encountered the same issue in the same environment and versions. 
As a temporary solution, I resolved it by forcing `jetson-containers` to build `torchvision` from the source. 
The necessary flag for this is in `packages -> pytorch -> torchvision -> Dockerfile`. I hope this helps!

ju-leon, 2025-05-20:
It seems like this is only a problem in the tests for torchvision.

Adding the `--skip-tests=all` flag to the container build seems to fix the issue, i.e.:
```
jetson-containers build --skip-tests=all pytorch
```

Hugo415, 2025-05-29:
Hi @YuZhong-Chen ,
Did you mean modify the `Dockerfile` under the `jetson-containers/packages/pytorch/torchvision/` right?
Make the FORCE_BUILD=`off` to `on` 

[image]

Hugo415, 2025-05-29:
Hi @m6f13 ,
Thank you for reply. 
Have you tried the `text-generation-webui` in **NVIDIA Jetson AI Lab** tutorial?

mtinnes, 2025-05-30:
Same issue when building a custom container from an existing base container (`FROM dustynv/whisper_trt:r36.3.0`):

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/transformers/utils/import_utils.py", line 2045, in __getattr__
    module = self._get_module(self._class_to_module[name])
  File "/usr/local/lib/python3.10/dist-packages/transformers/utils/import_utils.py", line 2075, in _get_module
    raise e
  File "/usr/local/lib/python3.10/dist-packages/transformers/utils/import_utils.py", line 2073, in _get_module
    return importlib.import_module("." + module_name, self.__name__)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/usr/local/lib/python3.10/dist-packages/transformers/modeling_utils.py", line 73, in <module>
    from .loss.loss_utils import LOSS_MAPPING
  File "/usr/local/lib/python3.10/dist-packages/transformers/loss/loss_utils.py", line 21, in <module>
    from .loss_d_fine import DFineForObjectDetectionLoss
  File "/usr/local/lib/python3.10/dist-packages/transformers/loss/loss_d_fine.py", line 21, in <module>
    from .loss_for_object_detection import (
  File "/usr/local/lib/python3.10/dist-packages/transformers/loss/loss_for_object_detection.py", line 32, in <module>
    from transformers.image_transforms import center_to_corners_format
  File "/usr/local/lib/python3.10/dist-packages/transformers/image_transforms.py", line 21, in <module>
    from .image_utils import (
  File "/usr/local/lib/python3.10/dist-packages/transformers/image_utils.py", line 59, in <module>
    from torchvision.transforms import InterpolationMode
  File "/usr/local/lib/python3.10/dist-packages/torchvision/__init__.py", line 6, in <module>
    from torchvision import _meta_registrations, datasets, io, models, ops, transforms, utils
  File "/usr/local/lib/python3.10/dist-packages/torchvision/_meta_registrations.py", line 164, in <module>
    def meta_nms(dets, scores, iou_threshold):
  File "/usr/local/lib/python3.10/dist-packages/torch/library.py", line 654, in register
    use_lib._register_fake(op_name, func, _stacklevel=stacklevel + 1)
  File "/usr/local/lib/python3.10/dist-packages/torch/library.py", line 154, in _register_fake
    handle = entry.abstract_impl.register(func_to_register, source)
  File "/usr/local/lib/python3.10/dist-packages/torch/_library/abstract_impl.py", line 31, in register
    if torch._C._dispatch_has_kernel_for_dispatch_key(self.qualname, "Meta"):
RuntimeError: operator torchvision::nms does not exist
```

ManuelZ, 2025-07-01:
Happens to me when trying to build `l4t-pytorch` after `gstreamer`.

I'm using commit 0f66b900d55200cb61bd50d3c685dfb239fc93dd, Apr 25th. Maybe I'll update to the latest commit and test it out again.

```
┌───────────────────────────────────────────────┐
│ > TESTING  container_dev:r36.4.3-torchvision │
└───────────────────────────────────────────────┘

docker run -t --rm --gpus=all --network=host \
  --env NVIDIA_DRIVER_CAPABILITIES=all \
  --volume /home/jorinnx/jetson-containers/packages/pytorch/torchvision:/test \
  --volume /home/jorinnx/jetson-containers/data:/data \
  --workdir /test \
  container_dev:r36.4.3-torchvision \
    /bin/bash -c 'python3 test.py

testing torchvision...
Traceback (most recent call last):
  File "/test/test.py", line 12, in <module>
    import torchvision
  File "/usr/local/lib/python3.10/dist-packages/torchvision/__init__.py", line 10, in <module>
    from torchvision import _meta_registrations, datasets, io, models, ops, transforms, utils  # usort:skip
  File "/usr/local/lib/python3.10/dist-packages/torchvision/_meta_registrations.py", line 164, in <module>
    def meta_nms(dets, scores, iou_threshold):
  File "/usr/local/lib/python3.10/dist-packages/torch/library.py", line 828, in register
    use_lib._register_fake(op_name, func, _stacklevel=stacklevel + 1)
  File "/usr/local/lib/python3.10/dist-packages/torch/library.py", line 198, in _register_fake
    handle = entry.fake_impl.register(func_to_register, source)
  File "/usr/local/lib/python3.10/dist-packages/torch/_library/fake_impl.py", line 31, in register
    if torch._C._dispatch_has_kernel_for_dispatch_key(self.qualname, "Meta"):
RuntimeError: operator torchvision::nms does not exist
[22:44:41] Failed building:  gstreamer, l4t-pytorch

Traceback (most recent call last):
  File "/home/jorinnx/jetson-containers/jetson_containers/build.py", line 129, in <module>
    build_container(**vars(args))
  File "/home/jorinnx/jetson-containers/jetson_containers/container.py", line 243, in build_container
    test_container(container_name, pkg, simulate)
  File "/home/jorinnx/jetson-containers/jetson_containers/container.py", line 423, in test_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'docker run -t --rm --gpus=all --network=host   --env NVIDIA_DRIVER_CAPABILITIES=all   --volume /home/jorinnx/jetson-containers/packages/pytorch/torchvision:/test   --volume /home/jorinnx/jetson-containers/data:/data   --workdir /test   container_dev:r36.4.3-torchvision     /bin/bash -c 'python3 test.py' 2>&1 | tee /home/jorinnx/jetson-containers/logs/20250630_224415/test/container_dev_r36.4.3-torchvision_test.py.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.
```

closed 2025-07-08 by johnnynunez