dusty-nv/jetson-containers #1090: Failed building jetson-inference: No such element or plugin 'nvarguscamerasrc'

m6f13, 2025-05-07:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### jetson-containers Component

Build

### Bug

when running: LSB_RELEASE=24.04 jetson-containers build jetson-inference

error:
**No such element or plugin 'nvarguscamerasrc'**
[19:42:33] Failed building:  jetson-inference

Traceback (most recent call last):
  File "/home/martin/git/jetson-containers/jetson_containers/build.py", line 129, in <module>
    build_container(**vars(args))
  File "/home/martin/git/jetson-containers/jetson_containers/container.py", line 243, in build_container
    test_container(container_name, pkg, simulate)
  File "/home/martin/git/jetson-containers/jetson_containers/container.py", line 429, in test_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'docker run -t --rm --network=host --runtime=nvidia   --volume /home/martin/git/jetson-containers/packages/multimedia/gstreamer:/test   --volume /home/martin/git/jetson-containers/data:/data   --workdir /test   jetson-inference:r36.4-cu128-24.04-gstreamer     /bin/bash -c '/bin/bash test.sh' 2>&1 | tee /home/martin/git/jetson-containers/logs/20250507_191239/test/jetson-inference_r36.4-cu128-24.04-gstreamer_test.sh.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 255.

### Environment

Namespace(packages=['jetson-inference'], name='', base='', multiple=False, build_flags='', build_args='', use_proxy=False, package_dirs=[''], list_packages=False, show_packages=False, skip_packages=[''], skip_errors=False, skip_tests=[''], test_only=[''], simulate=False, push='', no_github_api=False, log_dir=None, log_level=None, log_colors=None, log_status=None, debug=False, verbose=False, version=False)

┌───────────────────────┬────────────────────────┐
│ L4T_VERSION   36.4.3  │ JETPACK_VERSION  6.2   │
│ CUDA_VERSION  12.8    │ PYTHON_VERSION   3.12  │
│ SYSTEM_ARCH   aarch64 │ LSB_RELEASE      24.04 │
└───────────────────────┴────────────────────────┘

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

fivemtebex, 2025-05-10:
Great implementation of this feature. I particularly like the approach to error handling.

scriptforfivem, 2025-05-10:
The performance optimizations here are quite clever. Nice engineering work!

fivemeyestore, 2025-05-10:
The performance optimizations here are quite clever. Nice engineering work!

fivemserverhostings, 2025-05-10:
Excellent contribution to open source! The community will benefit greatly from this.

fivemhostingserver, 2025-05-10:
Great implementation of this feature. I particularly like the approach to error handling.

johnnynunez, 2025-06-27:
> ### Search before asking
> * [x]  I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.
> 
> ### jetson-containers Component
> Build
> 
> ### Bug
> when running: LSB_RELEASE=24.04 jetson-containers build jetson-inference
> 
> error: **No such element or plugin 'nvarguscamerasrc'** [19:42:33] Failed building: jetson-inference
> 
> Traceback (most recent call last): File "/home/martin/git/jetson-containers/jetson_containers/build.py", line 129, in build_container(**vars(args)) File "/home/martin/git/jetson-containers/jetson_containers/container.py", line 243, in build_container test_container(container_name, pkg, simulate) File "/home/martin/git/jetson-containers/jetson_containers/container.py", line 429, in test_container status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True) File "/usr/lib/python3.10/subprocess.py", line 526, in run raise CalledProcessError(retcode, process.args, subprocess.CalledProcessError: Command 'docker run -t --rm --network=host --runtime=nvidia --volume /home/martin/git/jetson-containers/packages/multimedia/gstreamer:/test --volume /home/martin/git/jetson-containers/data:/data --workdir /test jetson-inference:r36.4-cu128-24.04-gstreamer /bin/bash -c '/bin/bash test.sh' 2>&1 | tee /home/martin/git/jetson-containers/logs/20250507_191239/test/jetson-inference_r36.4-cu128-24.04-gstreamer_test.sh.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 255.
> 
> ### Environment
> Namespace(packages=['jetson-inference'], name='', base='', multiple=False, build_flags='', build_args='', use_proxy=False, package_dirs=[''], list_packages=False, show_packages=False, skip_packages=[''], skip_errors=False, skip_tests=[''], test_only=[''], simulate=False, push='', no_github_api=False, log_dir=None, log_level=None, log_colors=None, log_status=None, debug=False, verbose=False, version=False)
> 
> ┌───────────────────────┬────────────────────────┐ │ L4T_VERSION 36.4.3 │ JETPACK_VERSION 6.2 │ │ CUDA_VERSION 12.8 │ PYTHON_VERSION 3.12 │ │ SYSTEM_ARCH aarch64 │ LSB_RELEASE 24.04 │ └───────────────────────┴────────────────────────┘
> 
> ### Additional
> _No response_
> 
> ### Are you willing to submit a PR?
> * [ ]  Yes I'd like to help by submitting a PR!

jetson-inference is deprecated