dusty-nv/jetson-containers #1112: An issue while building a specific CUDA version of PyTorch using Jetson containers on JetPack 6.2.

kerokero-kero, 2025-05-21:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Question

Hi, I’m trying to build a simple PyTorch image using jetson-containers. I was able to successfully build it without specifying a CUDA_VERSION, which was great.

Encouraged by that, I then tried to specify the CUDA version explicitly using the following command:

`CUDA_VERSION=12.2 jetson-containers build pytorch`

Unfortunately, the build failed with the following error message:

```
#define CUDNN_MAJOR 8
#define CUDNN_MINOR 9
#define CUDNN_VERSION (CUDNN_MAJOR * 1000 + CUDNN_MINOR * 100 + CUDNN_PATCHLEVEL)
#define CUDNN_MAX_SM_MAJOR_NUMBER 9
#define CUDNN_MAX_SM_MINOR_NUMBER 0
#define CUDNN_MAX_DEVICE_VERSION (CUDNN_MAX_SM_MAJOR_NUMBER * 100 + CUDNN_MAX_SM_MINOR_NUMBER * 10)
#define CUDNN_MIN_DEVICE_VERSION CUDNN_SM_50
Executing: conv_sample
double free or corruption (out)
test.sh: line 14:    22 Aborted                 (core dumped) ./conv_sample
[14:40:46] Failed building:  pytorch 
```
Device: Jetson AGX Orin 32GB
JetPack version: 6.2

I’m not sure if this is related to compatibility between PyTorch and the specified CUDA version, or if I’m missing something else. Any advice or insight would be greatly appreciated—thank you!

j3soon, 2025-05-23:
I can reproduce this issue on two separate Jetson AGX Orin 32GB with JetPack 6.2.

Does anyone have any ideas on how to fix it? I'd be happy to open a pull request but may need some guidance.

johnnynunez, 2025-06-27:
upgrade your repository and jetson-containers to 36.4.4 and use last cuda 12.6