dusty-nv/jetson-containers #1119: Live Llava webrtc connection makes the pipeline crash

akatzi77, 2025-05-29:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### jetson-containers Component

Other

### Bug

I'm running Live Llava demo:
``
jetson-containers run $(autotag nano_llm) \
  python3 -m nano_llm.agents.video_query --api=mlc \
    --model Efficient-Large-Model/VILA1.5-3b \
    --max-context-len 256 \
    --max-new-tokens 32 \
    --video-input /dev/video0 \
    --video-output webrtc://@:8554/output
``
Everything runs well and I'm getting image description from the model.

When browsing to any of the following (either from jetson itself or another host):
https://JETSON_IP:8554/ , https://JETSON_IP:8050/ 

the GST pipeline crashes:
``
A wire shelf is in13:24:25 | INFO | 172.26.11.28 - - [29/May/2025 13:24:25] "GET / HTTP/1.1" 200 -
 a garage.
13:24:25 | INFO | refresh rate:  3.15 FPS (317.0 ms)
A wire shelf is13:24:25 | INFO | 172.26.11.28 - - [29/May/2025 13:24:25] "GET /static/chat.css HTTP/1.1" 200 -
 in13:24:25 | INFO | 172.26.11.28 - - [29/May/2025 13:24:25] "GET /static/bootstrap.css HTTP/1.1" 200 -
 a gar13:24:25 | INFO | 172.26.11.28 - - [29/May/2025 13:24:25] "GET /static/select2.min.css HTTP/1.1" 200 -
age.13:24:26 | INFO | 172.26.11.28 - - [29/May/2025 13:24:26] "GET /static/webrtc.js HTTP/1.1" 200 -

13:24:26 | INFO | refresh rate:  3.45 FPS (289.9 ms)
13:24:26 | INFO | 172.26.11.28 - - [29/May/2025 13:24:26] "GET /static/rest.js HTTP/1.1" 200 -
13:24:26 | INFO | 172.26.11.28 - - [29/May/2025 13:24:26] "GET /static/debounce.js HTTP/1.1" 200 -
13:24:26 | INFO | 172.26.11.28 - - [29/May/2025 13:24:26] "GET /static/websocket.js HTTP/1.1" 200 -
13:24:26 | INFO | 172.26.11.28 - - [29/May/2025 13:24:26] "GET /static/jquery-3.6.3.min.js HTTP/1.1" 200 -
13:24:26 | INFO | 172.26.11.28 - - [29/May/2025 13:24:26] "GET /static/bootstrap.bundle.min.js HTTP/1.1" 200 -
13:24:26 | INFO | 172.26.11.28 - - [29/May/2025 13:24:26] "GET /static/select2.min.js HTTP/1.1" 200 -
A wire shelf13:24:26 | INFO | 172.26.11.28 - - [29/May/2025 13:24:26] "GET /static/favicon.ico HTTP/1.1" 200 -
 is13:24:26 | INFO | connection open
13:24:26 | INFO | new websocket connection from ('172.26.11.28', 58109)
13:24:26 | INFO | listening on websocket connection from ('172.26.11.28', 58109)
 in[webrtc] websocket /output -- new connection opened by 172.26.11.28 (peer_id=0)
[webrtc] new WebRTC peer connecting (172.26.11.28, peer_id=0)
 a**
ERROR:/opt/jetson-utils/codec/gstEncoder.cpp:876:static void gstEncoder::onWebsocketMessage(WebRTCPeer*, const char*, size_t, void*): 'sinkpad' should not be nullptr
Bail out! ERROR:/opt/jetson-utils/codec/gstEncoder.cpp:876:static void gstEncoder::onWebsocketMessage(WebRTCPeer*, const char*, size_t, void*): 'sinkpad' should not be nullptr
Fatal Python error: Aborted
``

I've disabled chrome://flags/#enable-webrtc-hide-local-ips-with-mdns following instructions in https://forums.developer.nvidia.com/t/webrtc-issues-with-jetson-utils/283864.

### Environment

Orin AGX 32G
Jetpack 6.2
dustynv/nano_llm   r36.4.0
USB camera attached
browser: chrome (remote host) / chromium (jetson)

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

johnnynunez, 2025-06-27:
I will check

enginger, 2025-07-02:
hi i have the same issue on my end, is there any fix to this? 
[webrtc] websocket /output -- new connection opened by 172.20.10.2 (peer_id=0)
[webrtc] new WebRTC peer connecting (172.20.10.2, peer_id=0)
**
ERROR:/opt/jetson-utils/codec/gstEncoder.cpp:876:static void gstEncoder::onWebsocketMessage(WebRTCPeer*, const char*, size_t, void*): 'sinkpad' should not be nullptr
Bail out! ERROR:/opt/jetson-utils/codec/gstEncoder.cpp:876:static void gstEncoder::onWebsocketMessage(WebRTCPeer*, const char*, size_t, void*): 'sinkpad' should not be nullptr
Fatal Python error: Aborted

johnnynunez, 2025-07-02:
try to build it again