dusty-nv/jetson-containers #1159: Failed to build ollama docker image

kentsuiGitHub, 2025-06-17:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Bug

┌─────────────────────────────────────────────┐
│ > BUILDING  ollama:r36.4-cu126-22.04-python │
└─────────────────────────────────────────────┘

DOCKER_BUILDKIT=0 docker build --network=host \
  --tag ollama:r36.4-cu126-22.04-python \
  --file /opt/jetson-containers/packages/build/python/Dockerfile \
  --build-arg BASE_IMAGE=ollama:r36.4-cu126-22.04-cuda \
  --build-arg PYTHON_VERSION="3.10" \
  --no-cache \
   /opt/jetson-containers/packages/build/python

[14:16:36] [4/5] Building python (ollama:r36.4-cu126-22.04-python)3 stages compl             eted in 08m35s at 14:16:36
DEPRECATED: The legacy builder is deprecated and will be removed in a future rel             ease.
            BuildKit is currently disabled; enable it by removing the DOCKER_BUI             LDKIT=0
            environment-variable.

Sending build context to Docker daemon  23.04kB
Step 1/6 : ARG BASE_IMAGE
Step 2/6 : FROM ${BASE_IMAGE}
 ---> 20838c69e638
Step 3/6 : ARG PYTHON_VERSION     TMP=/tmp/python
 ---> Running in 219320e6de82
 ---> Removed intermediate container 219320e6de82
 ---> 31f0c9ef875a
Step 4/6 : ENV PYTHON_VERSION=${PYTHON_VERSION}     PYTHONFAULTHANDLER=1     PYT             HONUNBUFFERED=1     PYTHONIOENCODING=utf-8     PYTHONHASHSEED=random     PIP_NO_             CACHE_DIR=true     PIP_CACHE_PURGE=true     PIP_ROOT_USER_ACTION=ignore     PIP_             DISABLE_PIP_VERSION_CHECK=on     PIP_DEFAULT_TIMEOUT=100     PIP_WHEEL_DIR=/opt/             wheels     PIP_VERBOSE=1     TWINE_NON_INTERACTIVE=1     DEBIAN_FRONTEND=noninte             ractive     PATH=/opt/venv/bin:$PATH
 ---> Running in ef8ec7d71691
 ---> Removed intermediate container ef8ec7d71691
 ---> 05e86bd7d726
Step 5/6 : COPY install.sh $TMP/
 ---> d3e579088627
Step 6/6 : RUN $TMP/install.sh
 ---> Running in d24d5aa2c138
+ apt-get update
Get:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease [270 kB]
Get:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease [128 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease [127 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease [129 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports jammy/universe arm64 Packages [17.2 M             B]
Get:6 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 Packages [1,758 kB]
Get:7 http://ports.ubuntu.com/ubuntu-ports jammy/multiverse arm64 Packages [224              kB]
Get:8 http://ports.ubuntu.com/ubuntu-ports jammy/restricted arm64 Packages [24.2              kB]
Get:9 http://ports.ubuntu.com/ubuntu-ports jammy-updates/multiverse arm64 Packag             es [32.8 kB]
Get:10 http://ports.ubuntu.com/ubuntu-ports jammy-updates/restricted arm64 Packa             ges [3,922 kB]
Get:11 http://ports.ubuntu.com/ubuntu-ports jammy-updates/universe arm64 Package             s [1,532 kB]
Get:12 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 Packages [2             ,972 kB]
Get:13 http://ports.ubuntu.com/ubuntu-ports jammy-backports/universe arm64 Packa             ges [33.3 kB]
Get:14 http://ports.ubuntu.com/ubuntu-ports jammy-backports/main arm64 Packages              [82.8 kB]
Get:15 http://ports.ubuntu.com/ubuntu-ports jammy-security/universe arm64 Packag             es [1,233 kB]
Get:16 http://ports.ubuntu.com/ubuntu-ports jammy-security/multiverse arm64 Pack             ages [26.5 kB]
Get:17 http://ports.ubuntu.com/ubuntu-ports jammy-security/main arm64 Packages [             2,669 kB]
Get:18 http://ports.ubuntu.com/ubuntu-ports jammy-security/restricted arm64 Pack             ages [3,779 kB]
Fetched 36.2 MB in 11s (3,332 kB/s)
Reading package lists...
+ apt-get install -y --no-install-recommends python3.10 python3.10-dev
Reading package lists...
Building dependency tree...
Reading state information...
python3.10 is already the newest version (3.10.12-1~22.04.10).
python3.10 set to manually installed.
The following additional packages will be installed:
  libexpat1-dev libpython3.10-dev zlib1g-dev
The following NEW packages will be installed:
  libexpat1-dev libpython3.10-dev python3.10-dev zlib1g-dev
0 upgraded, 4 newly installed, 0 to remove and 8 not upgraded.
Need to get 5,467 kB of archives.
After this operation, 21.0 MB of additional disk space will be used.
Get:1 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 libexpat1-de             v arm64 2.4.7-1ubuntu0.6 [130 kB]
Get:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 zlib1g-dev a             rm64 1:1.2.11.dfsg-2ubuntu9.2 [163 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 libpython3.1             0-dev arm64 3.10.12-1~22.04.10 [4,666 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 python3.10-d             ev arm64 3.10.12-1~22.04.10 [508 kB]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 5,467 kB in 3s (1,848 kB/s)
Selecting previously unselected package libexpat1-dev:arm64.
(Reading database ... 29336 files and directories currently installed.)
Preparing to unpack .../libexpat1-dev_2.4.7-1ubuntu0.6_arm64.deb ...
Unpacking libexpat1-dev:arm64 (2.4.7-1ubuntu0.6) ...
Selecting previously unselected package zlib1g-dev:arm64.
Preparing to unpack .../zlib1g-dev_1%3a1.2.11.dfsg-2ubuntu9.2_arm64.deb ...
Unpacking zlib1g-dev:arm64 (1:1.2.11.dfsg-2ubuntu9.2) ...
Selecting previously unselected package libpython3.10-dev:arm64.
Preparing to unpack .../libpython3.10-dev_3.10.12-1~22.04.10_arm64.deb ...
Unpacking libpython3.10-dev:arm64 (3.10.12-1~22.04.10) ...
Selecting previously unselected package python3.10-dev.
Preparing to unpack .../python3.10-dev_3.10.12-1~22.04.10_arm64.deb ...
Unpacking python3.10-dev (3.10.12-1~22.04.10) ...
Setting up libexpat1-dev:arm64 (2.4.7-1ubuntu0.6) ...
Setting up zlib1g-dev:arm64 (1:1.2.11.dfsg-2ubuntu9.2) ...
Setting up libpython3.10-dev:arm64 (3.10.12-1~22.04.10) ...
Setting up python3.10-dev (3.10.12-1~22.04.10) ...
+ which python3.10
/usr/bin/python3.10
+ return_code=0
+ set -e
+ '[' 0 '!=' 0 ']'
++ lsb_release -rs
+ distro=22.04
+ '[' 22.04 = 24.04 ']'
+ '[' 22.04 = 20.04 ']'
+ '[' 22.04 = 18.04 ']'
+ python3.10
+ curl -sS https://bootstrap.pypa.io/get-pip.py
Using pip 25.1.1 from /tmp/python/tmpiwevonk5/pip.zip/pip (python 3.10)
Looking in indexes: https://pypi.jetson-ai-lab.dev/jp6/cu126
ERROR: Could not find a version that satisfies the requirement pip (from version             s: none)
ERROR: No matching distribution found for pip
The command '/bin/sh -c $TMP/install.sh' returned a non-zero code: 1
[14:17:11] Failed building:  ollama

Traceback (most recent call last):
  File "/opt/jetson-containers/jetson_containers/build.py", line 129, in <module             >
    build_container(**vars(args))
  File "/opt/jetson-containers/jetson_containers/container.py", line 229, in bui             ld_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash',              shell=True, check=True)
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network             =host   --tag ollama:r36.4-cu126-22.04-python   --file /opt/jetson-containers/pa             ckages/build/python/Dockerfile   --build-arg BASE_IMAGE=ollama:r36.4-cu126-22.04             -cuda   --build-arg PYTHON_VERSION="3.10"   --no-cache    /opt/jetson-containers             /packages/build/python 2>&1 | tee /opt/jetson-containers/logs/20250617_140759/bu             ild/04o5_ollama_r36.4-cu126-22.04-python.txt; exit ${PIPESTATUS[0]}' returned no             n-zero exit status 1.

### Environment

Jetpack 6

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

johnnynunez, 2025-06-27:
fixed

closed 2025-06-27 by johnnynunez

goldyfruit, 2025-06-29:
Still having the issue from a fresh `git clone` of this repo.

```

┌─────────────────────────────────────────────┐
│ > BUILDING  ollama:r36.4-cu126-22.04-ollama │
└─────────────────────────────────────────────┘

DOCKER_BUILDKIT=0 docker build --network=host \
  --tag ollama:r36.4-cu126-22.04-ollama \
  --file /root/jetson-containers/packages/llm/ollama/Dockerfile \
  --build-arg BASE_IMAGE=ollama:r36.4-cu126-22.04-python \
  --build-arg OLLAMA_VERSION="0.9.4" \
  --build-arg JETPACK_VERSION_MAJOR="6" \
   /root/jetson-containers/packages/llm/ollama

[09:27:48] [5/5] Building ollama (ollama:r36.4-cu126-22.04-ollama)                      4 stages completed in 09m31s at 09:27:48 
DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            BuildKit is currently disabled; enable it by removing the DOCKER_BUILDKIT=0
            environment-variable.

Sending build context to Docker daemon  31.74kB
Step 1/11 : ARG BASE_IMAGE
Step 2/11 : FROM ${BASE_IMAGE}
 ---> 647a8e4764eb
Step 3/11 : ARG OLLAMA_VERSION     JETPACK_VERSION_MAJOR     TMP=/tmp/ollama
 ---> Running in adc269f5a732
 ---> Removed intermediate container adc269f5a732
 ---> 9d2c34e008bf
Step 4/11 : ENV OLLAMA_VERSION=${OLLAMA_VERSION}     OLLAMA_HOST=0.0.0.0     OLLAMA_LOGS=/data/logs/ollama.log     OLLAMA_MODELS=/data/models/ollama/models
 ---> Running in 26a64599b628
 ---> Removed intermediate container 26a64599b628
 ---> 1ad1f5e84016
Step 5/11 : COPY nv_tegra_release /etc/nv_tegra_release
 ---> 14279353f6ed
Step 6/11 : COPY install.sh $TMP/
 ---> 85c92c45e725
Step 7/11 : RUN $TMP/install.sh
 ---> Running in 91bef37d1a90
Downloading https://github.com/ollama/ollama/releases/download/v0.9.4/ollama-linux-arm64.tgz
The command '/bin/sh -c $TMP/install.sh' returned a non-zero code: 8
[09:27:49] Failed building:  ollama

Traceback (most recent call last):
  File "/root/jetson-containers/jetson_containers/build.py", line 129, in <module>
    build_container(**vars(args))
  File "/root/jetson-containers/jetson_containers/container.py", line 229, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host   --tag ollama:r36.4-cu126-22.04-ollama   --file /root/jetson-containers/packages/llm/ollama/Dockerfile   --build-arg BASE_IMAGE=ollama:r36.4-cu126-22.04-python   --build-arg OLLAMA_VERSION="0.9.4"   --build-arg JETPACK_VERSION_MAJOR="6"    /root/jetson-containers/packages/llm/ollama 2>&1 | tee /root/jetson-containers/logs/20250629_091815/build/05o5_ollama_r36.4-cu126-22.04-ollama.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 8.
 ```

johnnynunez, 2025-06-29:
> Still having the issue from a fresh `git clone` of this repo.
>
> ```
>
> ┌─────────────────────────────────────────────┐
> │ > BUILDING  ollama:r36.4-cu126-22.04-ollama │ [...]

Change for you your last public version in config.py

goldyfruit, 2025-06-29:
So with CUDA 12.6, version 0.9.2 compiles.

johnnynunez, 2025-06-29:
i don't have that problem:
https://hub.docker.com/r/dustynv/ollama i pushed 0.9.3