dusty-nv/jetson-containers #1170: Update llama_cpp package to later build to support mmproj

ihubanov, 2025-06-19:
Hey Man,
Is it possible to update llama_cpp to newer build that will support mmproj?
I'm trying `jetson-containers build llama_cpp` but it fails... 
```
 ---> Removed intermediate container 745eed831fbb                  ---> 0815fe60144f
Step 7/7 : RUN if [ ! -f "$TMP/.llama_cpp" ]; then       echo "FAILED to install llama.cpp $LLAMA_CPP_VERSION";       exit 1;     fi
 ---> Running in c07f8da11230                                     FAILED to install llama.cpp 5676                                  The command '/bin/sh -c if [ ! -f "$TMP/.llama_cpp" ]; then       echo "FAILED to install llama.cpp $LLAMA_CPP_VERSION";       exit 1;     fi' returned a non-zero code: 1                            [07:11:52] Failed building:  llama_cpp                                                                                              Traceback (most recent call last):                                  File "/home/sniffski/Documents/jetson-containers/jetson_containers/build.py", line 129, in <module>                                   build_container(**vars(args))                                   File "/home/sniffski/Documents/jetson-containers/jetson_containers/container.py", line 229, in build_container                        status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host   --tag llama_cpp:r36.4-cu126-22.04-llama_cpp   --file /home/sniffski/Documents/jetson-containers/packages/llm/llama_cpp/Dockerfile   --build-arg BASE_IMAGE=llama_cpp:r36.4-cu126-22.04-sudonim   --build-arg LLAMA_CPP_VERSION="5676"   --build-arg LLAMA_CPP_VERSION_PY="0.3.8"   --build-arg LLAMA_CPP_BRANCH="b5676"   --build-arg LLAMA_CPP_BRANCH_PY="main"   --build-arg LLAMA_CPP_FLAGS="-DGGML_CUDA=on -DGGML_CUDA_F16=on -DLLAMA_CURL=on -DGGML_CUDA_FA_ALL_QUANTS=ON"    /home/sniffski/Documents/jetson-containers/packages/llm/llama_cpp 2>&1 | tee /home/sniffski/Documents/jetson-containers/logs/20250619_064951/build/10o10_llama_cpp_r36.4-cu126-22.04-llama_cpp.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.
```

johnnynunez, 2025-06-26:
llama-cpp-python should be updated. Now it is possible to build it

closed 2025-06-26 by johnnynunez

johnnynunez, 2025-07-01:
it is working again:

```bash
LSB_RELEASE=24.04 jetson-containers build llama_cpp
```

ihubanov, 2025-07-01:
I can confirm build works, but if I directly try `jetson-containers run $(autotag llama_cpp)` and let it download the container, it still downloads an older version... 
It is still fine, but I'm preparing an automated install script for another project and it would be way easier if I don't need to build the container on every initial install...

johnnynunez, 2025-07-01:
> I can confirm build works, but if I directly try `jetson-containers run $(autotag llama_cpp)` and let it download the container, it still downloads an older version... It is still fine, but I'm [...]

it is normal, dusty is on vacation, no dockers are updated :)

ihubanov, 2025-07-01:
Thanks for the update! So at some point they will get updated?

ihubanov, 2025-07-21:
Now build doesn't work again...
```
Using pip 25.1.1 from /tmp/python/tmp0nx4fyum/pip.zip/pip (python 3.10)
Looking in indexes: https://pypi.jetson-ai-lab.dev/jp6/cu126, https://pypi.jetson-ai-lab.dev/jp6/cu126
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object
 at 0xffff8e6cec50>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu126/pip/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object
 at 0xffff8e7cad40>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu126/pip/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object
 at 0xffff8e7c9240>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu126/pip/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object
 at 0xffff8e7cbe80>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu126/pip/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object
 at 0xffff8e7c9120>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu126/pip/
ERROR: Could not find a version that satisfies the requirement pip (from versions: none)
ERROR: No matching distribution found for pip
The command '/bin/sh -c $TMP/install.sh' returned a non-zero code: 1
[16:13:21] Failed building:  llama_cpp
```
And it seems the container that comes with autotag doesn't have the latest llama_cpp yet... :(