dusty-nv/jetson-containers #1195: jp6/cu122 removed?

zhifeis, 2025-07-02:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Bug

I'm using jp6/cu122 to build docker images for a Jetson Orin NX running Jetpack 6.0 rev. 2 w/ cuda 12.2 but since yesterday my builds have been failing. After investigation it looks like jp6/cu122 has been removed from the server:

[image]

[image]

Is this a temporary outage or has it been moved to a different address? Thanks!

### Environment

Nvidia Jetson Orin NX 8GB
Jetpack 6.0 rev 2
CUDA 12.2

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

johnnynunez, 2025-07-02:
yes, upgrade to last jetpack 6 that Orin NX is compatible. Cuda 12.6 and 12.9 are very stable.
12.9 will be removed to move to cuda 13.0

closed 2025-07-02 by johnnynunez

zhifeis, 2025-07-02:
Are there any plans for cuda 12.6 to be removed in the future or can I assume it to be available indefinitely, even if it becomes unsupported?