dusty-nv/jetson-containers #1197: Jetson PyPi down

RoroKuso, 2025-07-03:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### jetson-containers Component

Packages

### Bug

Hey, it seems that pypi.jetson-ai-lab.dev is down. Just wanted to notify you.

### Environment

Using a jetson AGX Orin with jp6 and cuda 12.6

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

AimbotParce, 2025-07-03:
Hit the same wall, hope it comes back up soon :)

closed 2025-07-03 by johnnynunez

handysome6, 2025-07-08:
The server seems to be offline agian. Any idea to setup a news board to update maintenance notifications?

AimbotParce, 2025-07-08:
Still down :(

JadTawil77, 2025-07-09:
Any Updates on this?