dusty-nv/jetson-containers #1202: Installing python packages fails inside dustynv/l4t-pytorch:r36.2.0 fails

jpluomajoki-solita, 2025-07-07:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Bug

Trying to install any python package with pip3 inside a container fails, since the the used package index seems to be gone.

```
$ docker run -it dustynv/l4t-pytorch:r36.2.0 bash
WARNING: The requested image's platform (linux/arm64/v8) does not match the detected host platform (linux/amd64/v3) and no specific platform was requested
root@8c11b6b50f83:/# pip3 install opencv-python
Looking in indexes: http://jetson.webredirect.org/jp6/cu122, https://pypi.ngc.nvidia.com
ERROR: Could not find a version that satisfies the requirement opencv-python (from versions: none)
ERROR: No matching distribution found for opencv-python
```

### Environment

dustynv/l4t-pytorch:r36.2.0 running in nvidia jetson Orin NX 16GB

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

johnnynunez, 2025-07-07:
> ### Search before asking
> * [x]  I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.
>
> ### jetson-containers [...]

upgrade to use newer jetpack version

johnnynunez, 2025-07-11:
there is a Nvidia pypi backup until new host comeback:
```bash
pip3 install torch —index-url https://pypi.jetson-ai-lab.io/jp6/cu126
```