dusty-nv/jetson-containers #1203: Unable to install sglang - Python wheel site down

taresh18-ag, 2025-07-08:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Bug

Hi, thanks for building this repository. 

I am trying to build sglang container following the tutorial However, while building the image, it fails when trying to install a package from the python wheel site - "https://pypi.jetson-ai-lab.dev/"

Could you pls take a look into it. 

### Environment

jp6.1 cuda12.6

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

jtaveau, 2025-07-08:
For sglang that's a fix yes, but if I want to install any pip packages in an image that fails as the website is down, for example on this one I want to install `colcon-cache`

```
#23 5.544 Looking in indexes: https://pypi.jetson-ai-lab.dev/jp6/cu128
#23 5.544 Collecting git+https://github.com/ruffsl/colcon-cache.git@a937541bfc496c7a267db7ee9d6cceca61e470ca
#23 5.545   Cloning https://github.com/ruffsl/colcon-cache.git (to revision a937541bfc496c7a267db7ee9d6cceca61e470ca) to /tmp/pip-req-build-e201om3f
#23 5.552   Running command git clone --filter=blob:none --quiet https://github.com/ruffsl/colcon-cache.git /tmp/pip-req-build-e201om3f
#23 6.509   Running command git rev-parse -q --verify 'sha^a937541bfc496c7a267db7ee9d6cceca61e470ca'
#23 6.511   Running command git fetch -q https://github.com/ruffsl/colcon-cache.git a937541bfc496c7a267db7ee9d6cceca61e470ca
#23 6.707   Running command git checkout -q a937541bfc496c7a267db7ee9d6cceca61e470ca
#23 6.963   Resolved https://github.com/ruffsl/colcon-cache.git to commit a937541bfc496c7a267db7ee9d6cceca61e470ca
#23 6.965   Preparing metadata (setup.py): started
#23 7.187   Preparing metadata (setup.py): finished with status 'done'
#23 9.552 WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xf989e1d92930>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu128/colcon-core/
#23 13.17 WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xf989e1d92b70>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu128/colcon-core/
#23 16.79 WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xf989e1d93440>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu128/colcon-core/
#23 20.92 WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xf989e1d93650>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu128/colcon-core/
#23 26.03 WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xf989e1d93860>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu128/colcon-core/
#23 28.44 INFO: pip is looking at multiple versions of colcon-cache to determine which version is compatible with other requirements. This could take a while.
#23 28.44 ERROR: Could not find a version that satisfies the requirement colcon-core>=0.5.2 (from colcon-cache) (from versions: none)
#23 28.44 ERROR: No matching distribution found for colcon-core>=0.5.2
```

Do you have any idea on how to fix that?

johnnynunez, 2025-07-10:
Hello,
We are restoring pypi. It was closed by security. We are upgrading the plattform.