dusty-nv/jetson-containers #1204: PyPI index (http://jetson.webredirect.org) is down again

TeddyZaremba, 2025-07-08:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### jetson-containers Component

Docker, Packages

### Bug

Running into the same problem as this issue (https://github.com/dusty-nv/jetson-containers/issues/811). All the pip registries mentioned there seem to be down. 

[image]

### Environment

Jetson Orin Nano running dustynv/l4t-pytorch:r36.2.0

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

zhifeis, 2025-07-09:
seconded

JoshPoravanthattil, 2025-07-10:
experiencing the same issue as well!

closed 2025-07-10 by johnnynunez

johnnynunez, 2025-07-11:
> Is there a way we can set this as the default temporarily when building a container until the new host comes back? For example, if I wanted to build a container, how can I make all the packages that will be installed point to this new temporary Nvidia pypi without having to update each of the config/dockerfiles?

changing this https://github.com/dusty-nv/jetson-containers/blob/19e6ec3cfcfe106b8e37e9387d38bacb05194125/packages/cuda/cuda/config.py#L137