dusty-nv/jetson-containers #1211: Web direct is down: jp6/cu126

greliefhub, 2025-07-12:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Bug

There are similar requests, but they are outdated. The webdirect is not working. 

WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffb349c6a0>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu126/onnxruntime-gpu/

### Environment

dustynv/l4t-pytorch:r36.4.0 bash

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

qingy1337, 2025-07-13:
Same error:

```
Looking in indexes: https://pypi.jetson-ai-lab.dev/jp6/cu126
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa945fc10>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu126/pip/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa945ecb0>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu126/pip/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa945ef50>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu126/pip/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa945d6c0>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu126/pip/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa945ed10>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu126/pip/
ERROR: Could not find a version that satisfies the requirement pip (from versions: none)
ERROR: No matching distribution found for pip
```

bryanjuho, 2025-07-13:
The pypi index has been down for a few days now. Anyone knows an alternative, or a statuspage perhaps?

closed 2025-07-13 by johnnynunez

ivyinautumn, 2025-07-21:
> > The pypi index has been down for a few days now. Anyone knows an alternative, or a statuspage perhaps?
> 
> pypi.jetson-ai-lab.io

I tried the link [pypi.jetson-ai-lab.io](url) around 10:00(UTC+8), and it worked fine. But when I tried the link around 13:00 later, it did not work fine and showed response time was too long. I want to know whether the backup link [pypi.jetson-ai-lab.io](url) is also down?

[image]