dusty-nv/jetson-containers #1213: jax LSB_RELEASE=24.04

hkortier, 2025-07-14:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Bug

Error in fail: The supported CUDNN versions are ["8.9.4.25", "8.9.6", "8.9.7.29", "9.1.1", "9.2.0", "9.2.1", "9.3.0", "9.4.0", "9.5.0", "9.5.1", "9.6.0", "9.7.0", "9.7.1", "9.8.0"]. Please provide a supported version in ["HERMETIC_CUDNN_VERSION", "TF_CUDNN_VERSION"] environment variable(s) or add JSON URL for CUDNN version=9.11.0.

It seems hardcoded in the build file: https://vscode.dev/github/dusty-nv/jetson-containers/blob/master/packages/ml/jax/build.sh#L16-L17

### Environment

jetson orin nx jetpack 6.2.1

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

johnnynunez, 2025-07-14:
> Thanks for pointing that out. Do you know if it’s also applicable to version 22.04? I haven’t been able to get it working there either.
> 
> More specific:
> 
> jetson-containers build jax
> INFO: Repository python_version_repo instantiated at: /opt/jax/WORKSPACE:21:25: in /root/.cache/bazel/_bazel_root/cfd1b2cc6fe180f3eb424db6004de364/external/xla/third_party/py/python_init_repositories.bzl:13:22: in python_init_repositories Repository rule python_repository defined at: /root/.cache/bazel/_bazel_root/cfd1b2cc6fe180f3eb424db6004de364/external/xla/third_party/py/python_repo.bzl:211:36: in ERROR: /root/.cache/bazel/_bazel_root/cfd1b2cc6fe180f3eb424db6004de364/external/xla/third_party/py/python_repo.bzl:37:13: An error occurred during the fetch of repository 'python_version_repo': Traceback (most recent call last): File "/root/.cache/bazel/_bazel_root/cfd1b2cc6fe180f3eb424db6004de364/external/xla/third_party/py/python_repo.bzl", line 37, column 13, in _python_repository_impl fail(""" Error in fail: Could not find requirements_lock.txt file matching specified Python version. Specified python version: 3.10 Python versions with available requirement_lock.txt files: 3.11, 3.12, 3.13, 3.14, 3.13-ft, 3.14-ft Please check python_init_repositories() in your WORKSPACE file. ERROR: Error computing the main repository mapping: no such package '@@python_version_repo//': Could not find requirements_lock.txt file matching specified Python version. Specified python version: 3.10 Python versions with available requirement_lock.txt files: 3.11, 3.12, 3.13, 3.14, 3.13-ft, 3.14-ft Please check python_init_repositories() in your WORKSPACE file. Traceback (most recent call last): File "/opt/jax/build/build.py", line 782, in asyncio.run(main()) File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run return loop.run_until_complete(main) File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete return future.result() File "/opt/jax/build/build.py", line 473, in main raise RuntimeError(f"Command failed with return code {result.return_code}") RuntimeError: Command failed with return code 1 The command '/bin/sh -c /tmp/JAX/install.sh || /tmp/JAX/build.sh' returned a non-zero code: 1 [20:11:54] Failed building: jax
> 
> Traceback (most recent call last): File "/home/jetson/repos/jetson-containers/jetson_containers/build.py", line 129, in build_container(**vars(args)) File "/home/jetson/repos/jetson-containers/jetson_containers/container.py", line 229, in build_container status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True) File "/usr/lib/python3.10/subprocess.py", line 526, in run raise CalledProcessError(retcode, process.args, subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host --tag jax:r36.4.tegra-aarch64-cu126-22.04-jax --file /home/jetson/repos/jetson-containers/packages/ml/jax/Dockerfile --build-arg BASE_IMAGE=jax:r36.4.tegra-aarch64-cu126-22.04-llvm_20 --build-arg JAX_CUDA_ARCH_ARGS="8.7;9.0" --build-arg JAX_VERSION="0.7.1" --build-arg JAX_BUILD_VERSION="0.7.1" --build-arg ENABLE_NCCL="1" /home/jetson/repos/jetson-containers/packages/ml/jax 2>&1 | tee /home/jetson/repos/jetson-containers/logs/20250714_201127/build/09o9_jax_r36.4.tegra-aarch64-cu126-22.04-jax.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.

i think that now is pointing to bad version of cuda. Look inside build.sh and change the version.

johnnynunez, 2025-07-14:
don't change that lines.
change it here:
https://github.com/dusty-nv/jetson-containers/blob/3f99bd9d791393820ba7091a71fb109eb1c0965a/packages/ml/jax/build.sh#L16

jax is special... use hermetic cuda that is all in one for compile

hkortier, 2025-07-15:
cuda 12.8 and cudnn=9.8 results in:

```bash
Traceback (most recent call last):
  File "/test/test.py", line 41, in <module>
    x = jnp.linalg.solve(b, a)
jaxlib._jax.XlaRuntimeError: INTERNAL: jaxlib/gpu/linalg_kernels.cc:113: operation gpuGetLastError() failed: no kernel image is available for execution on the device
```

closed 2025-08-07 by johnnynunez