dusty-nv/jetson-containers #1216: Can't run stable-diffusion-webui container due to numpy error

arjanvaneersel, 2025-07-18:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### jetson-containers Component

Packages

### Bug

Command: `jetson-containers run $(autotag stable-diffusion-webui)`

Output:
```
Namespace(packages=['stable-diffusion-webui'], prefer=['local', 'registry', 'build'], disable=[''], user='dustynv', output='/tmp/autotag', quiet=False, verbose=False)
-- L4T_VERSION=36.4.4  JETPACK_VERSION=6.2.1  CUDA_VERSION=12.6
-- Finding compatible container image for ['stable-diffusion-webui']
dustynv/stable-diffusion-webui:r36.4.0
V4L2_DEVICES: 
### ARM64 architecture detected
### Jetson Detected
SYSTEM_ARCH=tegra-aarch64
+ docker run --runtime nvidia --env NVIDIA_DRIVER_CAPABILITIES=compute,utility,graphics -it --rm --network host --shm-size=8g --volume /tmp/argus_socket:/tmp/argus_socket --volume /etc/enctune.conf:/etc/enctune.conf --volume /etc/nv_tegra_release:/etc/nv_tegra_release --volume /tmp/nv_jetson_model:/tmp/nv_jetson_model --volume /var/run/dbus:/var/run/dbus --volume /var/run/avahi-daemon/socket:/var/run/avahi-daemon/socket --volume /var/run/docker.sock:/var/run/docker.sock --volume /home/arjan/jetson-containers/data:/data -v /etc/localtime:/etc/localtime:ro -v /etc/timezone:/etc/timezone:ro --device /dev/snd -e PULSE_SERVER=unix:/run/user/1000/pulse/native -v /run/user/1000/pulse:/run/user/1000/pulse --device /dev/bus/usb --device /dev/i2c-0 --device /dev/i2c-1 --device /dev/i2c-2 --device /dev/i2c-3 --device /dev/i2c-4 --device /dev/i2c-5 --device /dev/i2c-6 --device /dev/i2c-7 --device /dev/i2c-8 --name jetson_container_20250718_045547 dustynv/stable-diffusion-webui:r36.4.0
Python 3.10.12 (main, Feb  4 2025, 14:57:36) [GCC 11.4.0]
Version: v1.10.1
Commit hash: 82a973c04367123ae98bd9abdf80d9eda9b910e2
Installing requirements
Traceback (most recent call last):
  File "/opt/stable-diffusion-webui/launch.py", line 48, in <module>
    main()
  File "/opt/stable-diffusion-webui/launch.py", line 39, in main
    prepare_environment()
  File "/opt/stable-diffusion-webui/modules/launch_utils.py", line 423, in prepare_environment
    run_pip(f"install -r \"{requirements_file}\"", "requirements")
  File "/opt/stable-diffusion-webui/modules/launch_utils.py", line 144, in run_pip
    return run(f'"{python}" -m pip {command} --prefer-binary{index_url_line}', desc=f"Installing {desc}", errdesc=f"Couldn't install {desc}", live=live)
  File "/opt/stable-diffusion-webui/modules/launch_utils.py", line 116, in run
    raise RuntimeError("\n".join(error_bits))
RuntimeError: Couldn't install requirements.
Command: "/usr/local/bin/python3" -m pip install -r "requirements_versions.txt" --prefer-binary
Error code: 1
stdout: Using pip 25.0.1 from /usr/local/lib/python3.10/dist-packages/pip (python 3.10)
Looking in indexes: https://pypi.jetson-ai-lab.dev/jp6/cu126
Requirement already satisfied: setuptools==69.5.1 in /usr/local/lib/python3.10/dist-packages (from -r requirements_versions.txt (line 1)) (69.5.1)
Requirement already satisfied: GitPython==3.1.32 in /usr/local/lib/python3.10/dist-packages (from -r requirements_versions.txt (line 2)) (3.1.32)
Requirement already satisfied: Pillow==9.5.0 in /usr/local/lib/python3.10/dist-packages (from -r requirements_versions.txt (line 3)) (9.5.0)
Requirement already satisfied: accelerate==0.21.0 in /usr/local/lib/python3.10/dist-packages (from -r requirements_versions.txt (line 4)) (0.21.0)
Requirement already satisfied: blendmodes==2022 in /usr/local/lib/python3.10/dist-packages (from -r requirements_versions.txt (line 5)) (2022)
Requirement already satisfied: clean-fid==0.1.35 in /usr/local/lib/python3.10/dist-packages (from -r requirements_versions.txt (line 6)) (0.1.35)
Requirement already satisfied: diskcache==5.6.3 in /usr/local/lib/python3.10/dist-packages (from -r requirements_versions.txt (line 7)) (5.6.3)
Requirement already satisfied: einops==0.4.1 in /usr/local/lib/python3.10/dist-packages (from -r requirements_versions.txt (line 8)) (0.4.1)
Requirement already satisfied: facexlib==0.3.0 in /usr/local/lib/python3.10/dist-packages (from -r requirements_versions.txt (line 9)) (0.3.0)
Requirement already satisfied: fastapi==0.94.0 in /usr/local/lib/python3.10/dist-packages (from -r requirements_versions.txt (line 10)) (0.94.0)
Requirement already satisfied: gradio==3.41.2 in /usr/local/lib/python3.10/dist-packages (from -r requirements_versions.txt (line 11)) (3.41.2)
Requirement already satisfied: httpcore==0.15 in /usr/local/lib/python3.10/dist-packages (from -r requirements_versions.txt (line 12)) (0.15.0)
Requirement already satisfied: inflection==0.5.1 in /usr/local/lib/python3.10/dist-packages (from -r requirements_versions.txt (line 13)) (0.5.1)
Requirement already satisfied: jsonmerge==1.8.0 in /usr/local/lib/python3.10/dist-packages (from -r requirements_versions.txt (line 14)) (1.8.0)
Requirement already satisfied: kornia==0.6.7 in /usr/local/lib/python3.10/dist-packages (from -r requirements_versions.txt (line 15)) (0.6.7)
Requirement already satisfied: lark==1.1.2 in /usr/local/lib/python3.10/dist-packages (from -r requirements_versions.txt (line 16)) (1.1.2)

stderr: WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff96197910>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu126/numpy/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff96197c40>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu126/numpy/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff96197df0>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu126/numpy/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff96197fd0>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu126/numpy/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff95fc4190>: Failed to establish a new connection: [Errno 113] No route to host')': /jp6/cu126/numpy/
ERROR: Could not find a version that satisfies the requirement numpy==1.26.2 (from versions: none)
ERROR: No matching distribution found for numpy==1.26.2
```

### Environment

.env has only defaults, except: `INDEX_HOST=jetson-ai-lab.io`

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

johnnynunez, 2025-07-18:
Did you modify config.py from cuda?

arjanvaneersel, 2025-07-19:
@johnnynunez Again, I am not building ollama, but trying to run stable-diffusion-webui. Someone else made a remark about ollama.

Anyway, I tried everything from scratch again. Double checked that .env is sourced, removed the downloaded image and gave the command: `jetson-containers run $(autotag stable-diffusion-webui)`. It downloaded the container, but once it runs it ends with the error: `ERROR: Could not find a version that satisfies the requirement numpy==1.26.2 (from versions: none)
ERROR: No matching distribution found for numpy==1.26.2`. 

The complete log is in my initial post of this ticket.

Corovaneer, 2025-07-24:
Same issue as @arjanvaneersel
I noticed that docker searches at "https://pypi.jetson-ai-lab.dev", which is down, despite the line 
`export PIP_INDEX_URL=https://pypi.jetson-ai-lab.io/jp6/cu126` in the end of bashrc
I grepped everything for "dev/jp6/cu126" and found some in home/jetson-containers/packages/vlm/vila-microservice, however changing the address there to .io did nothing.
Re-running `bash jetson-containers/install.sh` as @ch40s suggested results in a warning 
`The index url "jetson-ai-lab.io" seems invalid, please provide a scheme.`
 and an error 
`Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
    × python setup.py egg_info did not run successfully.`