dusty-nv/jetson-containers #1220: flash-attention - no kernel image

nicolofranceschi, 2025-07-20:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Question

## Flash Attention fails on Jetson AGX Orin with "no kernel image" error

  ### Environment
  - **Device**: NVIDIA Jetson AGX Orin
  - **GPU**: Orin (sm_87)
  - **PyTorch version**: 2.7.0
  - **CUDA version**: 12.6
  - **Flash Attention versions tested**: 2.5.7, 2.5.8, 2.8.1
  - **Container images tested**: 
    - `ghcr.io/intelligencebox-repo/intelligencebox-pro/vectorclient-jetson:pytorch2.7-base`
    - `ghcr.io/intelligencebox-repo/intelligencebox-pro/vectorclient-jetson:pytorch2.7-cuda12.8-u
  buntu24.04`

  ### Problem Description
  Flash Attention installation fails on Jetson AGX Orin with the error:
  RuntimeError: CUDA error: no kernel image is available for execution on the device

  This happens even though:
  1. The Jetson AGX Orin has sm_87 architecture
  2. Flash Attention wheels are built for sm_87
  3. CUDA operations work fine in PyTorch

  ### Steps to Reproduce
  1. Install Flash Attention from PyPI or build from source with `TORCH_CUDA_ARCH_LIST="8.7"`
  2. Run basic test:
  ```python
  import torch
  from flash_attn import flash_attn_func

  batch_size, seq_len, num_heads, head_dim = 2, 128, 8, 64
  q = torch.randn(batch_size, seq_len, num_heads, head_dim, dtype=torch.float16, device='cuda')
  k = torch.randn(batch_size, seq_len, num_heads, head_dim, dtype=torch.float16, device='cuda')
  v = torch.randn(batch_size, seq_len, num_heads, head_dim, dtype=torch.float16, device='cuda')

  output = flash_attn_func(q, k, v)  # Fails here

  Root Cause

  After investigation, I found that PyTorch itself was not compiled with sm_87 support, even
  though it runs on sm_87 hardware. You can verify this:
  import torch
  print(torch.cuda.get_arch_list())  # Does NOT include sm_87

  Expected Behavior

  Flash Attention should work on Jetson AGX Orin since it's officially supported hardware.

  Actual Behavior

  Flash Attention fails with "no kernel image is available for execution" because PyTorch wasn't
  built with sm_87 support.

  Workarounds Attempted

  1. Built Flash Attention from source with TORCH_CUDA_ARCH_LIST="8.7" - still fails
  2. Tried different Flash Attention versions (2.5.7, 2.5.8, 2.8.1) - all fail
  3. Set environment variables CUDA_ARCHITECTURES=87, CUDAARCHS=87 - no effect

  Proposed Solutions

  1. Document PyTorch compatibility requirements more clearly
  2. Provide pre-built wheels that work with common Jetson PyTorch distributions
  3. Add a compatibility check during installation to warn users
  4. Provide guidance on which PyTorch images/builds support sm_87

  Additional Context

  - The error is misleading because it suggests Flash Attention wasn't built correctly, when
  actually PyTorch is the issue
  - This affects all Jetson AGX Orin users trying to use Flash Attention
  - Working PyTorch images include dustynv/pytorch:2.7-r36.4.0 which has sm_87 support

  Questions

  1. Which official PyTorch builds include sm_87 support?
  2. Can Flash Attention detect this incompatibility during installation?
  3. Are there plans to provide Jetson-specific wheels with proper PyTorch dependencies?

  This issue clearly explains:
  - The problem and error message
  - The root cause (PyTorch lacking sm_87 support)
  - Steps to reproduce
  - What you've already tried
  - Proposed solutions
  - Helpful context for other Jetson users

johnnynunez, 2025-07-20:
> ### Search before asking
> 
> * [x]  I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.
> 
> ### Question
> 
> ## Flash Attention fails on Jetson AGX Orin with "no kernel image" error
> 
> ### Environment
> 
> * **Device**: NVIDIA Jetson AGX Orin
> * **GPU**: Orin (sm_87)
> * **PyTorch version**: 2.7.0
> * **CUDA version**: 12.6
> * **Flash Attention versions tested**: 2.5.7, 2.5.8, 2.8.1
> * **Container images tested**:
>   
>   * `ghcr.io/intelligencebox-repo/intelligencebox-pro/vectorclient-jetson:pytorch2.7-base`
>   * `ghcr.io/intelligencebox-repo/intelligencebox-pro/vectorclient-jetson:pytorch2.7-cuda12.8-u   buntu24.04`
> 
> ### Problem Description
> 
> Flash Attention installation fails on Jetson AGX Orin with the error: RuntimeError: CUDA error: no kernel image is available for execution on the device
> 
> This happens even though:
> 
> 1. The Jetson AGX Orin has sm_87 architecture
> 2. Flash Attention wheels are built for sm_87
> 3. CUDA operations work fine in PyTorch
> 
> ### Steps to Reproduce
> 
> 1. Install Flash Attention from PyPI or build from source with `TORCH_CUDA_ARCH_LIST="8.7"`
> 2. Run basic test:
> 
> import torch
> from flash_attn import flash_attn_func
> 
> batch_size, seq_len, num_heads, head_dim = 2, 128, 8, 64
> q = torch.randn(batch_size, seq_len, num_heads, head_dim, dtype=torch.float16, device='cuda')
> k = torch.randn(batch_size, seq_len, num_heads, head_dim, dtype=torch.float16, device='cuda')
> v = torch.randn(batch_size, seq_len, num_heads, head_dim, dtype=torch.float16, device='cuda')
> 
> output = flash_attn_func(q, k, v)  # Fails here
> 
> Root Cause
> 
> After investigation, I found that PyTorch itself was not compiled with sm_87 support, even
> though it runs on sm_87 hardware. You can verify this:
> import torch
> print(torch.cuda.get_arch_list())  # Does NOT include sm_87
> 
> Expected Behavior
> 
> Flash Attention should work on Jetson AGX Orin since it's officially supported hardware.
> 
> Actual Behavior
> 
> Flash Attention fails with "no kernel image is available for execution" because PyTorch wasn't
> built with sm_87 support.
> 
> Workarounds Attempted
> 
> 1. Built Flash Attention from source with TORCH_CUDA_ARCH_LIST="8.7" - still fails
> 2. Tried different Flash Attention versions (2.5.7, 2.5.8, 2.8.1) - all fail
> 3. Set environment variables CUDA_ARCHITECTURES=87, CUDAARCHS=87 - no effect
> 
> Proposed Solutions
> 
> 1. Document PyTorch compatibility requirements more clearly
> 2. Provide pre-built wheels that work with common Jetson PyTorch distributions
> 3. Add a compatibility check during installation to warn users
> 4. Provide guidance on which PyTorch images/builds support sm_87
> 
> Additional Context
> 
> - The error is misleading because it suggests Flash Attention wasn't built correctly, when
> actually PyTorch is the issue
> - This affects all Jetson AGX Orin users trying to use Flash Attention
> - Working PyTorch images include dustynv/pytorch:2.7-r36.4.0 which has sm_87 support
> 
> Questions
> 
> 1. Which official PyTorch builds include sm_87 support?
> 2. Can Flash Attention detect this incompatibility during installation?
> 3. Are there plans to provide Jetson-specific wheels with proper PyTorch dependencies?
> 
> This issue clearly explains:
> - The problem and error message
> - The root cause (PyTorch lacking sm_87 support)
> - Steps to reproduce
> - What you've already tried
> - Proposed solutions
> - Helpful context for other Jetson users
> 
> ### Additional
> 
> _No response_

Is it happens on 12.9 or 12.6?

johnnynunez, 2025-07-20:
Did you tried with pytorch 2.8.0?

johnnynunez, 2025-07-20:
PyTorch 2.7 is not built here

closed 2025-07-20 by johnnynunez

nicolofranceschi, 2025-07-21:
sorry what you mean ? , there is no image with pythotch 2.8 

box@box:~$ jetson-containers run dustynv/pytorch:2.8-r36.4-cu128-24.04
V4L2_DEVICES: 
### ARM64 architecture detected
### Jetson Detected
SYSTEM_ARCH=tegra-aarch64
+ docker run --runtime nvidia --env NVIDIA_DRIVER_CAPABILITIES=compute,utility,graphics -it --rm --network host --shm-size=8g --volume /tmp/argus_socket:/tmp/argus_socket --volume /etc/enctune.conf:/etc/enctune.conf --volume /etc/nv_tegra_release:/etc/nv_tegra_release --volume /tmp/nv_jetson_model:/tmp/nv_jetson_model --volume /var/run/dbus:/var/run/dbus --volume /var/run/avahi-daemon/socket:/var/run/avahi-daemon/socket --volume /var/run/docker.sock:/var/run/docker.sock --volume /home/box/jetson-containers/data:/data -v /etc/localtime:/etc/localtime:ro -v /etc/timezone:/etc/timezone:ro --device /dev/snd -e PULSE_SERVER=unix:/run/user/1000/pulse/native -v /run/user/1000/pulse:/run/user/1000/pulse --device /dev/bus/usb --device /dev/i2c-0 --device /dev/i2c-1 --device /dev/i2c-2 --device /dev/i2c-3 --device /dev/i2c-4 --device /dev/i2c-5 --device /dev/i2c-6 --device /dev/i2c-7 --device /dev/i2c-8 --name jetson_container_20250721_084331 dustynv/pytorch:2.8-r36.4-cu128-24.04
Unable to find image 'dustynv/pytorch:2.8-r36.4-cu128-24.04' locally
docker: Error response from daemon: manifest for dustynv/pytorch:2.8-r36.4-cu128-24.04 not found: manifest unknown: manifest unknown.
See 'docker run --help'