dusty-nv/jetson-containers #1227: Missing jp5/cu114 and jp4/cu102 indexer in pypi.jetson-ai-lab.io

stylish-llama, 2025-07-22:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Bug

With the departure of  `dusty-nv` (many thanks to them) and <https://pypi.jetson-ai-lab.dev/jp5/cu114> unavailable, all container builds fail on JP4/5 systems.

### Environment

```
┌───────────────────────┬────────────────────────┐
│ L4T_VERSION   35.6.2  │ JETPACK_VERSION  5.1.5 │
│ CUDA_VERSION  11.4    │ PYTHON_VERSION   3.8   │
│ SYSTEM_ARCH   aarch64 │ LSB_RELEASE      20.04 │
└───────────────────────┴────────────────────────┘
```

### Additional

As a workaround I managed to build a working image of ollama v0.9.6 on JP5 by hard-coding `pip_path = "root/pypi"`  here: https://github.com/dusty-nv/jetson-containers/blob/696a77f45d3a51d82454e98401ae443cd74a333a/packages/cuda/cuda/config.py#L141-L145

I belive on older platforms that do not support JP6/7, multiple other images could be built without modifying the codebase, by just adding a basic indexers containing only the inherited bases `root/pypi`.

PS: a list of previously available wheels is visible at the [internet archive](https://web.archive.org/web/20250316160127/https://pypi.jetson-ai-lab.dev/), sadly the actual resources are not available

### Are you willing to submit a PR?

- [x] Yes I'd like to help by submitting a PR!

Keval-P, 2025-08-01:
If you need, you can download the packages from here 
=> https://drive.google.com/drive/folders/1macyG38SGCiazkiVfvQDahkAHgjhT6Mi

johnnynunez, 2025-08-27:
I opened that for you! but we don't have the wheels sorry

johnnynunez, 2025-08-29:
> excluding those already published by Keval-P and [blacklisted](https://github.com/dusty-nv/jetson-containers/blob/8cadf109b3fc42a97847700a4acd32a87b3fd949/packages/net/devpi/blacklist.txt) ones , I currently only have llama_cpp. You can find it here https://we.tl/t-xmstTFCKVF (let me know if the link expires before you download it).
> 
> If you have a script to automatically generate the most commonly used ones (or even just a list) I can let it run in the background and send you the result.

i can't upload torch due that is from ngc, so it should be generated from scratch

Oneiroi, 2025-09-01:
Encountering an issue with building `pytorch`; likely I'm presuming for the same related issue seems https://pypi.jetson-ai-lab.dev/jp6/cu126 is not available when running `jetson-containers build pytorch` this step however is needed to support running of other projects where pytorch is a dependency; unsure of what (if any) steps may be taken at this time ?

johnnynunez, 2025-09-02:
> [@johnnynunez](https://github.com/johnnynunez) interesting this is clone of this repo; just ran another `git pull` which brought it to [37956ba](https://github.com/dusty-nv/jetson-containers/commit/37956baec45292af31b61822ad2d521a52cbab67) have the build re-running will update with the results shortly

dev branch always is more updated