dusty-nv/jetson-containers #1286: Architecture Detection Logic: `tegra-aarch64` vs `IS_SBSA` for SBSA System Identification

tokk-nv, 2025-08-15:
## Background
Currently, the jetson-containers build system uses `CUDA_ARCH` values (`tegra-aarch64` vs `aarch64`) to determine whether to install SBSA-specific packages like NVPL. The assumption was that `tegra-aarch64` indicates Jetson devices that don't support SBSA.

## Problem Identified
We've discovered that **Jetson Thor (SBSA system) still returns "tegra" in `uname -a`**, making `CUDA_ARCH = "tegra-aarch64"`. <br>This breaks the current logic because:

- **Current logic**: `if [[ "$CUDA_ARCH" == "aarch64" ]]` (only installs NVPL on pure `aarch64`)
- **Reality**: Thor is `tegra-aarch64` but **IS** an SBSA system
- **Result**: NVPL and other SBSA packages fail to install on Thor

## Proposed Solution
Switch from architecture-based detection to **SBSA capability detection** using the existing `IS_SBSA` flag, which is determined by checking if `nvgpu` appears in the GPU name from `nvidia-smi`.

## Code Changes Needed - Example 1: nvpl
```bash
# Current (problematic):
if [[ "$CUDA_ARCH" == "aarch64" ]]; then
    # Install SBSA packages
fi

# Proposed (robust):
if [[ "$CUDA_ARCH" == "aarch64" ]] || [[ "$IS_SBSA" == "True" ]]; then
    # Install SBSA packages
fi
```

## Known Files Affected
- `packages/cuda/nvpl/install.sh`
- `packages/cuda/cusparselt/install.sh` 
- `packages/cuda/nccl/install.sh`
- Other packages with similar architecture checks

## Questions for Discussion
1. **Architecture Detection Strategy**: Should we completely abandon `CUDA_ARCH`-based SBSA detection in favor of `IS_SBSA`?
2. **Backward Compatibility**: Are there any edge cases where `IS_SBSA` might be incorrect?
3. **Implementation Approach**: Should this be a gradual migration or a comprehensive refactor?
4. **Testing Strategy**: How can we validate this change across different Jetson generations?

## Current Status
- ✅ Fixed `packages/cuda/cusparselt/install.sh` by adding `IS_SBSA` check
- 🔄 NVPL installation fix using `IS_SBSA` veified. PR in progress.
- 🔄 Need to evaluate and standardize across all affected packages
- 📋 Seeking community input on the best long-term approach

## Reference

### System Classification Matrix

| \*OS   | Orin    | Thor | DGX Spark | GH200 |
| ------ | ------- | ---- | --------- | ----- |
| DGX OS |         |      | SBSA      | SBSA  |
| JP 7.2 | SBSA?   | SBSA |           |       |
| JP 7.1 | (nvgpu)? | SBSA |           |       |
| JP 7.0 | \-      | SBSA |           |       |
| JP 6.x | (nvgpu) | \-   |           |       |

### Evidence and Examples

#### Jetson Thor running L4T r38.1 (JetPack 7.0 EA)
```bash
$ uname -a
Linux jat03-iso0807 6.8.12-tegra #1 SMP PREEMPT Thu Aug  7 11:56:42 PDT 2025 aarch64 aarch64 aarch64 GNU/Linux
```
#### Jetson Orin running L4T r36.4 (JetPack 6.2)
```bash
$ uname -a
Linux jao-02-jp62 5.15.148-tegra #1 SMP PREEMPT Tue Jan 7 17:14:38 PST 2025 aarch64 aarch64 aarch64 GNU/Linux
```
#### DGX Spark
```bash
$ uname -a
Linux spark-xxxx 6.11.0-1013-nvidia #13-Ubuntu SMP PREEMPT_DYNAMIC Thu Jul 24 21:58:34 UTC 2025 aarch64 aarch64 aarch64 GNU/Linux
```

## Tags
`architecture-detection`, `sbsa-support`, `jetson-thor`, `build-system`, `enhancement`

ms1design, 2025-08-15:
@tokk-nv thanks for this ticket and vast problem description. The ultimate solution seems to be obvious here - to include Orin on SBSA ;)

Until that happens, we can just rely on more dirty solution where we can:

- rename current IS_SBSA to IS_NVGPU to reflect its true meaning 
- and finally include check for Thor in gpu name to create the IS_SBSA:

IS_SBSA = (CUDA_ARCH == "aarch64") or (
    IS_NVGPU and "Thor" in GPU_NAME
)

This is just my proposal here to maje sure that we don’t need to refactor much.

ms1design, 2025-08-15:
> > All good [@johnnynunez](https://github.com/johnnynunez), just remember to be inclusive here as this repo has also active collaborators that are helping to shape it/fix it and help the community too.
> 
> 🤣🤣🤣❤️

So you don’t agree with above statement?

OriNachum, 2025-08-16:
@tokk-nv 

Thank you for the comprehensive evidence/ background.  

I, generally, prefer variables to say exactly what they mean:

IS_SBSA - would mean that it is SBSA, otherwise it's confusing to me.

But I saw these:
- IS_TEGRA
- IS_NVGPU
- IS_THOR
- (I didn't see for Spark and ARM Blackwell servers - if specifically ***IS_SBSA*** - great!)

I think they should stay - even if we don't use any of them directly.

This leaves us with another variable: USE_SBSA 
This we can override if we like.
It's a decision variable with a clear instruction.

Then we can do something like
```
if ! IS_X86 && USE_SBSA; then 
```

I don't mind using architecture, as long as it's clear.

I know it's semantic, but it reduces confusion.
Then we can naturally remove the use of "architecture" field if relevant.

@tokk-nv @ms1design @johnnynunez  what do you think?

johnnynunez, 2025-08-16:
> [@tokk-nv](https://github.com/tokk-nv)
>
> Thank you for the comprehensive evidence/ background.
>
> I, generally, prefer variables to say exactly what they mean:
>
> IS_SBSA - would mean that it is SBSA, [...]

It easy. Simplified by IS_SBSA IS_TEGRA. There are two options from CUDA. Nvidia compiler not distinguishing any gpu, only codegene to build binaries.

Cuda 13 has:

nvcc --list-gpu-code
sm_75 turing
sm_80 ampere
sm_86 ampere
sm_87 Orin
sm_88
sm_89 Ada
sm_90 H100/200/GH200
sm_100 B100/b200/GB200
sm_110 Thor
sm_103 GB300
sm_120 RTX/PRO
sm_121 Spark

From CUDA 12.9:
[image]

From CUDA 13:
[image]
 The Server Base System Architecture (SBSA) is a hardware system architecture, it is like a template to democratize arm processors.

So, cuda-tegra and cuda arm are the same for the system (AARCH64), but cuda tegra was specialized for jetson.
Thor and spark and cloud are based on grace(cpu) that support sbsa natively.

In summary, we only have to distinguish like actual conditions:
IS_SBSA
IS_TEGRA

The only have to change is how to detect correctly assuming that there is not cuda software installed because by nvidia-smi is easy to detect. Tegra has old driver nvgpu.

But jetson-containers first, install SO, build-essential and inside docker not exist nvidia-smi until install cuda(that install for sbsa, tegra or amd64), so, we have to detect by other ways, like @rbonghi did in jtop

kbenkhaled, 2025-08-25:
@OriNachum @ms1design @johnnynunez 
What do you guys think about Chitoku's proposal? do we want to adapt it?
Personally I am good with:

- `pytorch:arm64-sbsa-cu131-24.04` for Thor, GH200 and maybe DGX Spark
- `pytorch:r38.3.aarch64jetson-cu131-24.04` : for Orin

Happy to proceed with this tagging if there are no objections.

kbenkhaled, 2025-08-25:
@OriNachum I would say as soon as reasonably possible.
@johnnynunez I agree, but given that the current approach correctly distinguishes Orin (Jetson) and Thor (SBSA) for now, I propose we leave it as is and track SBSA vs. Jetson detection as a follow-up.

I will work on the agreed upon re-tagging tomorrow and open a draft PR and we take it from there!

mfruhner, 2026-09-27:
I want to confirm that building images on Jetson Orin with JetPack 7.2 is STILL completly broken at the moment.
As @PavanDendi states, the orin arch is falsely detected as tegra and the package image sbsa/cu132 is missing.
Is there anything in the works to fix that?