dusty-nv/jetson-containers #1345: tritonserver for JP 6.2.1?

turowicz, 2025-08-28:
@dusty-nv is that on the roadmap?

johnnynunez, 2025-08-28:
> [@dusty-nv](https://github.com/dusty-nv) is that on the roadmap?

dusty was retired

turowicz, 2025-08-28:
was retired or has retired?

@johnnynunez perhaps you can help: https://github.com/triton-inference-server/server/issues/8367

johnnynunez, 2025-08-28:
> was retired or has retired?
> 
> [@johnnynunez](https://github.com/johnnynunez) perhaps you can help: [triton-inference-server/server#8367](https://github.com/triton-inference-server/server/issues/8367)

He left nvidia some months ago

turowicz, 2025-08-28:
his profile on github still says NVIDIA... wonder what's he up to now.

@johnnynunez perhaps you can help: https://github.com/triton-inference-server/server/issues/8367

johnnynunez, 2025-08-28:
I will help with this for sure. But I just landed in Nvidia and I don't have time yet

turowicz, 2025-09-02:
also they didn't release the any l4t containers for JP7...

turowicz, 2025-09-03:
> [@turowicz](https://github.com/turowicz) it should work now: https://github.com/triton-inference-server/server/releases/tag/v2.60.0

NOPE

[image]

turowicz, 2025-09-03:
It looks like we need to stay on JP5 if we want the triton to work... the old version of it

turowicz, 2025-09-03:
thanks for pointing me to a non-solution...

turowicz, 2025-09-04:
The forums explicilty say that for triton all questions need to go to GitHub**!**

https://forums.developer.nvidia.com/t/triton-inference-server-support-available-on-github/190337/2