dusty-nv/jetson-containers #1351: Sudonim CUDA test fails with 'NoneType' object has no attribute 'decode' after commit 39e1316e

tokk-nv, 2025-08-30:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Bug

## Issue Summary

The sudonim package test is failing with a CUDA initialization error after commit 39e1316e ("KISS pattern... working again #1350"). The error occurs during CUDA device query and prevents the sudonim package from functioning properly.

## Error Details

```
cudaDeviceQuery() failed: 'NoneType' object has no attribute 'decode'
Traceback (most recent call last):
  File "/opt/venv/bin/sudonim", line 3, in <module>
    from sudonim.update import entrypoint
  File "/opt/sudonim/sudonim/__init__.py", line 17, in <module>
    from .utils.hub import *
  File "/opt/sudonim/sudonim/utils/hub.py", line 7, in <module>
    env, log = getenv()
               ^^^^^^^^
  File "/opt/sudonim/sudonim/env.py", line 99, in getenv
    vars.append(global_env())
                ^^^^^^^^^^
  File "/opt/sudonim/sudonim/env.py", line 32, in global_env
    CUDA_DEVICES = nim.cudaDeviceQuery()
                   ^^^^^^^^^^^^^^^^^^^^^
  File "/opt/sudonim/sudonim/utils/cuda.py", line 14, in cudaDeviceQuery
    raise error
  File "/opt/sudonim/sudonim/utils/cuda.py", line 11, in cudaDeviceQuery
    return _cudaDeviceQuery()
           ^^^^^^^^^^^^^^^^^^^^^
  File "/opt/sudonim/sudonim/utils/cuda.py", line 49, in _cudaDeviceQuery
    raise RuntimeError("cuInit failed with error code %d: %s" % (result, error_str.value.decode()))
                                                                         ^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'decode'
```

## Root Cause Analysis
The error occurs in `/opt/sudonim/sudonim/utils/cuda.py` at line 49 when `cuInit()` fails and `error_str.value` is `None`. This suggests that CUDA initialization is failing at a fundamental level.

## Affected Commits
- ✅ Working: 798d3ce9, 975e6a9a, 5c3436bc
- ❌ Failing: 39e1316e (latest dev)

## Suspicious Changes in Commit `39e1316e`
BuildKit Version Changes: Conditional BuildKit usage based on JetPack version
CUDA Library Path Changes: Prioritizing stub libraries over actual CUDA runtime libraries
Major Build System Refactoring: "KISS pattern... working again" suggests significant changes

## Impact
- Sudonim package: Cannot initialize CUDA, rendering it non-functional
- Other packages: CUDA package itself works, suggesting the issue is in dependency resolution or linking
- Build system: May affect other CUDA-dependent packages

## Reproduction Steps
1. Checkout commit `39e1316e`
1. Build sudonim package: `./build.sh --build-flags="--no-cache" --no-github-api sudonim`
1. Run sudonim test: The test will fail with the CUDA error above

## Expected Behavior
Sudonim should successfully initialize CUDA and display device information, as it does in earlier commits.

### Environment

- Jetson Thor (JetPack 7.0)
- Jetson AGX Orin (JetPack 6.2.1)

### Additional

- The issue persists even after reverting the CUDA library path changes
- The problem appears to be systemic rather than package-specific
- BuildKit version changes may be affecting library linking behavior

### Are you willing to submit a PR?

- [x] Yes I'd like to help by submitting a PR!

johnnynunez, 2025-08-30:
> ### Search before asking
> 
> * [x]  I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.
> 
> ### jetson-containers Component
> 
> _No response_
> 
> ### Bug
> 
> ## Issue Summary
> 
> The sudonim package test is failing with a CUDA initialization error after commit [39e1316](https://github.com/dusty-nv/jetson-containers/commit/39e1316ea5ce280809470a3af47fe5da8e4149ea) ("KISS pattern... working again [#1350](https://github.com/dusty-nv/jetson-containers/pull/1350)"). The error occurs during CUDA device query and prevents the sudonim package from functioning properly.
> 
> ## Error Details
> 
> ```
> cudaDeviceQuery() failed: 'NoneType' object has no attribute 'decode'
> Traceback (most recent call last):
>   File "/opt/venv/bin/sudonim", line 3, in <module>
>     from sudonim.update import entrypoint
>   File "/opt/sudonim/sudonim/__init__.py", line 17, in <module>
>     from .utils.hub import *
>   File "/opt/sudonim/sudonim/utils/hub.py", line 7, in <module>
>     env, log = getenv()
>                ^^^^^^^^
>   File "/opt/sudonim/sudonim/env.py", line 99, in getenv
>     vars.append(global_env())
>                 ^^^^^^^^^^
>   File "/opt/sudonim/sudonim/env.py", line 32, in global_env
>     CUDA_DEVICES = nim.cudaDeviceQuery()
>                    ^^^^^^^^^^^^^^^^^^^^^
>   File "/opt/sudonim/sudonim/utils/cuda.py", line 14, in cudaDeviceQuery
>     raise error
>   File "/opt/sudonim/sudonim/utils/cuda.py", line 11, in cudaDeviceQuery
>     return _cudaDeviceQuery()
>            ^^^^^^^^^^^^^^^^^^^^^
>   File "/opt/sudonim/sudonim/utils/cuda.py", line 49, in _cudaDeviceQuery
>     raise RuntimeError("cuInit failed with error code %d: %s" % (result, error_str.value.decode()))
>                                                                          ^^^^^^^^^^^^^^^^^^
> AttributeError: 'NoneType' object has no attribute 'decode'
> ```
> 
> ## Root Cause Analysis
> 
> The error occurs in `/opt/sudonim/sudonim/utils/cuda.py` at line 49 when `cuInit()` fails and `error_str.value` is `None`. This suggests that CUDA initialization is failing at a fundamental level.
> 
> ## Affected Commits
> 
> * ✅ Working: [798d3ce](https://github.com/dusty-nv/jetson-containers/commit/798d3ce9a4439367b44c610526be2ff786fcb852), [975e6a9](https://github.com/dusty-nv/jetson-containers/commit/975e6a9a882bd4a420441bc8c1bcb8d09e0dbad1), [5c3436b](https://github.com/dusty-nv/jetson-containers/commit/5c3436bcd606bd5f8fa2b6a38d428395c4cae1a4)
> * ❌ Failing: [39e1316](https://github.com/dusty-nv/jetson-containers/commit/39e1316ea5ce280809470a3af47fe5da8e4149ea) (latest dev)
> 
> ## Suspicious Changes in Commit `39e1316e`
> 
> BuildKit Version Changes: Conditional BuildKit usage based on JetPack version CUDA Library Path Changes: Prioritizing stub libraries over actual CUDA runtime libraries Major Build System Refactoring: "KISS pattern... working again" suggests significant changes
> 
> ## Impact
> 
> * Sudonim package: Cannot initialize CUDA, rendering it non-functional
> * Other packages: CUDA package itself works, suggesting the issue is in dependency resolution or linking
> * Build system: May affect other CUDA-dependent packages
> 
> ## Reproduction Steps
> 
> 1. Checkout commit `39e1316e`
> 2. Build sudonim package: `./build.sh --build-flags="--no-cache" --no-github-api sudonim`
> 3. Run sudonim test: The test will fail with the CUDA error above
> 
> ## Expected Behavior
> 
> Sudonim should successfully initialize CUDA and display device information, as it does in earlier commits.
> 
> ### Environment
> 
> * Jetson Thor (JetPack 7.0)
> * Jetson AGX Orin (JetPack 6.2.1)
> 
> ### Additional
> 
> * The issue persists even after reverting the CUDA library path changes
> * The problem appears to be systemic rather than package-specific
> * BuildKit version changes may be affecting library linking behavior
> 
> ### Are you willing to submit a PR?
> 
> * [x]  Yes I'd like to help by submitting a PR!

Which is the command to reproduce this?
I only added more paths that not should affect to this

[image]

This was the major problems of all frameworks.

https://developer.nvidia.com/blog/whats-new-and-important-in-cuda-toolkit-13-0/

johnnynunez, 2025-08-30:
I don't have problems with 0 or 1

[image]

johnnynunez, 2025-08-30:
catched:
The problem is LD_LIBRARY_PATH: Adding /usr/local/cuda/lib64/stubs to LD_LIBRARY_PATH in your final image causes the operating system to find the “fake” (stub) library instead of the “real” driver library provided by the host.
Solution:deep

OriNachum, 2025-08-30:
@tokk-nv are we sure it was broken because of the commit and not because of another issue? (One of Sudonim dependencies, maybe?)

johnnynunez, 2025-08-30:
> hi [@johnnynunez](https://github.com/johnnynunez) [@OriNachum](https://github.com/OriNachum) Thank you for getting on this.
> 
> There may be one other small issue being played here. I found that the `sudonim` package's Dockerfile missing `cuda` as a dependency. And I was applying this when I got them passed. Let me correct this in my original post.
> 
> Johnny, can you try this?
> 
> ```
> git checkout upstream/dev
> git checkout 5c3436bc
> sed -i '/^# depends: \[python\]$/s/\[python\]/[cuda, python]/' packages/llm/sudonim/Dockerfile
> ./build.sh --build-flags="--no-cache" --no-github-api sudonim
> ```

Thanks💚

I will try it @tokk-nv

johnnynunez, 2025-08-30:
It is working @OriNachum @tokk-nv 

[image]

tokk-nv, 2025-08-31:
Thank you, Johnny.

Can you also try this on the latest commit `39e1316e`?
```
git checkout 39e1316e
sed -i '/^# depends: \[python\]$/s/\[python\]/[cuda, python]/' packages/llm/sudonim/Dockerfile
./build.sh --build-flags="--no-cache" --no-github-api sudonim
```

This will likely fail, which will reveal two things.

- `sudonim` package's Dockerfile has been needing `cuda` as a dependency.
- the workaround above no longer works on the latest. CUDA library path change affecting?

johnnynunez, 2025-08-31:
> Thank you, Johnny.
>
> Can you also try this on the latest commit `39e1316e`?
>
> ```
> git checkout 39e1316e
> sed -i '/^# depends: \[python\]$/s/\[python\]/[cuda, python]/' packages/llm/sudonim/Dockerfile [...]

I deleted stubs path that well some libraries are starting to use because they were independent from real gpu driver.

On las commit from from dev i deleted those but i added extra cccl path, changing first to the end. So if it not take the old path from cuda 12, finally cmake it will find on the end of the ching

sudonim is working from me