dusty-nv/jetson-containers #1360: Cannot build SGLang for Spark

Lyken17, 2025-09-01:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Bug

Thanks for sharing the great tool! I am following the tutorial  https://github.com/dusty-nv/jetson-containers/tree/master/packages/llm/sglang to run `jetson-containers build sglang` but facing 

```
FATAL: kernel `fmha_cutlassF_f32_aligned_64x64_rf_sm80` is for sm80-sm100, but was built for sm121
```

when building. Seems previous dockerfile uses some pre-compiled binary which is not suitable for spark cuda-arch. Is there any method to show the building dockerfile and diagnose the issue?

### Environment

latest main

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

Lyken17, 2025-09-01:
That's cool! Can you share the dockerfile file / instructions? @johnnynunez

johnnynunez, 2025-09-02:
@Lyken17 https://github.com/sgl-project/sglang/pull/9911

johnnynunez, 2025-09-02:
@Lyken17 im working on it... i think that i fixed vllm

Lyken17, 2025-09-03:
I see. Is this related with SGLang? Current SGLang no longer relies on the vllm.

johnnynunez, 2025-09-03:
> I see. Is this related with SGLang? Current SGLang no longer relies on the vllm.

I fixed vllm for cuda 13! I found the error:
https://github.com/Dao-AILab/flash-attention/pull/1860

whl:
https://pypi.jetson-ai-lab.io/sbsa/cu130/sgl-kernel/0.3.8
https://pypi.jetson-ai-lab.io/sbsa/cu130/sglang/0.5.1.post3

Lyken17, 2025-09-08:
I cannot access the two repos.

johnnynunez, 2025-09-08:
> I cannot access the two repos.

I will check

wwj-2017-1117, 2025-09-16:
cat /home/laiye/jetson-containers/logs/20250916_093928/test/40-1_sglang_r38.2.aarch64-cu130-24.04-sglang_test.py.txt
testing SGLang...
Traceback (most recent call last):
  File "/test/test.py", line 3, in <module>
    import sglang as sgl
ModuleNotFoundError: No module named 'sglang'

env
1. cuda13.0  
2. sglang:r38.2.aarch64-cu130-24.04-sglang  
3. https://docs.sglang.ai/platforms/nvidia_jetson.html