dusty-nv/jetson-containers #1373: build pytorch:2.8 on agx thor failed

m986883511, 2025-09-02:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### Bug

command is 
```shell
LSB_RELEASE=24.04 jetson-containers build pytorch:2.8
     jetson-containers run dustynv/pytorch:2.8-r36.4-cu128-24.04
```

the end of the buiild log
```shell
USE_GLOO=1
USE_MPI=0
USE_NCCL=1
USE_NNPACK=1
USE_PRIORITIZED_TEXT_FOR_LD=1
USE_PYTORCH_QNNPACK=1
USE_XNNPACK=0
++ date +%s
++ uptime
++ awk '-Fload average:' '{print $2}'
+ BUILD_START=1756856300
+ export USE_CUDNN=1
+ USE_CUDNN=1
+ export USE_CUSPARSELT=1
+ USE_CUSPARSELT=1
+ export USE_CUDSS=1
+ USE_CUDSS=1
+ export USE_CUFILE=1
+ USE_CUFILE=1
+ export USE_NATIVE_ARCH=0
+ USE_NATIVE_ARCH=0
+ export USE_DISTRIBUTED=1
+ USE_DISTRIBUTED=1
+ export USE_FLASH_ATTENTION=1
+ USE_FLASH_ATTENTION=1
+ export USE_MEM_EFF_ATTENTION=1
+ USE_MEM_EFF_ATTENTION=1
+ export USE_TENSORRT=0
+ USE_TENSORRT=0
CUDA 13 detected, turn off build with CUDSS (as 0.6 supported on CUDA 12).
=== Starting PyTorch build ===
+ export USE_BLAS=1
+ USE_BLAS=1
+ export BLAS=NVPL
+ BLAS=NVPL
+ [[ cu130 == cu13* ]]
+ echo 'CUDA 13 detected, turn off build with CUDSS (as 0.6 supported on CUDA 12).'
+ export USE_CUDSS=0
+ USE_CUDSS=0
+ echo '=== Starting PyTorch build ==='
Build command: python3 setup.py bdist_wheel --dist-dir /opt
+ echo 'Build command: python3 setup.py bdist_wheel --dist-dir /opt'
+ python3 setup.py bdist_wheel --dist-dir /opt
+ tee /tmp/pytorch_build.log
++ df -m /opt
++ tail -1
++ awk '{print $3"/"$2"MB ("$3*100/$2"%)"}'
2025-09-02 23:38:19 - Memory: 9248/125772MB (7.35299%) - Load:  2.13, 1.65, 1.33 - Disk: 114567/958865MB (11.9482%)
+ echo '2025-09-02 23:38:19 - Memory: 9248/125772MB (7.35299%) - Load:  2.13, 1.65, 1.33 - Disk: 114567/958865MB (11.9482%)'
+ sleep 60
Building wheel torch-2.8.0
-- Building version 2.8.0
Could not find any of CMakeLists.txt, Makefile, setup.py, LICENSE, LICENSE.md, LICENSE.txt in /opt/pytorch/third_party/pybind11
Did you run 'git submodule update --init --recursive'?
+ BUILD_EXIT_CODE=1
++ date +%s
[... 99 lines cut ...]
pytorch                   2.8-r38.2.tegra-aarch64-cu130-24.04-gdrcopy           4377424bb6ed   14 hours ago   8.74GB
pytorch                   2.8-r38.2.tegra-aarch64-cu130-24.04-nccl              af28fee531f3   14 hours ago   8.74GB
pytorch                   2.8-r38.2.tegra-aarch64-cu130-24.04-cudnn             301c6841fcba   14 hours ago   8.14GB
pytorch                   2.8-r38.2.tegra-aarch64-cu130-24.04-cuda_13.0         0b1c77fba0c0   14 hours ago   7.64GB
pytorch                   2.8-r38.2.tegra-aarch64-cu130-24.04-pip_cache_cu130   8ba52ac61100   15 hours ago   850MB
pytorch                   2.8-r38.2.tegra-aarch64-cu130-24.04-build-essential   724599994157   15 hours ago   850MB
ultralytics/ultralytics   latest-jetson-jetpack6                                4409f0f94667   18 hours ago   14.5GB
ultralytics/ultralytics   latest-jetson-jetpack5                                edb75ff5755f   18 hours ago   13.9GB
ubuntu                    24.04                                                 b60cb74dfa1b   2 weeks ago    101MB
redis                     alpine                                                b3891f782dea   2 weeks ago    69.3MB
postgres                  alpine                                                4e21f75150d1   2 weeks ago    270MB
nvidia/cuda               13.0.0-tensorrt-devel-ubuntu24.04                     ede3dfeba7fd   3 weeks ago    14.1GB
nvcr.io/nvidia/pytorch    25.08-py3                                             eeefcd24d725   3 weeks ago    21.9GB
adminer                   standalone                                            df4baae7ed3f   3 months ago   124MB
jetson@jetson:~/code/github$ 
```

### Environment

agx thor
VERSION="24.04.3 LTS (Noble Numbat)"

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

johnnynunez, 2025-09-03:
you should know that pytorch 2.8.0 is not compatible with cuda 13

m986883511, 2025-09-03:
change to 2.9 is ok

closed 2025-09-03 by m986883511