dusty-nv/jetson-containers #1395: Failed to build ollama image

kentsuiGitHub, 2025-09-06:
### Search before asking

- [x] I have searched the jetson-containers [issues](https://github.com/dusty-nv/jetson-containers/issues) and found no similar feature requests.

### jetson-containers Component

Build

### Bug

[buildOllama.txt](https://github.com/user-attachments/files/22184682/buildOllama.txt)

┌───────────────────────────────────────────────────────────┐
│ [1m> TESTING  ollama:r36.4.tegra-aarch64-cu126-22.04-sudonim[22m │
└───────────────────────────────────────────────────────────┘

[1mdocker run -t --rm --network=host --privileged --runtime=nvidia \
  --volume /opt/jetson-containers/packages/llm/sudonim:/test \
  --volume /opt/jetson-containers/data:/data \
  ollama:r36.4.tegra-aarch64-cu126-22.04-sudonim \
    /bin/bash -c '/bin/bash /test/test.sh[22m

[12:48:07] ===================================================================================== 
[12:48:07] ===================================================================================== 
[12:48:07] 💣 `jetson-containers build` failed after 677.6 seconds (11.3 minutes) 
[12:48:07] Error: Command 'docker run -t --rm --network=host --privileged --runtime=nvidia   --volume /opt/jetson-containers/packages/llm/sudonim:/test   --volume /opt/jetson-containers/data:/data   ollama:r36.4.tegra-aarch64-cu126-22.04-sudonim     /bin/bash -c '/bin/bash /test/test.sh' 2>&1 | tee /opt/jetson-containers/logs/20250906_123649/test/09-1_ollama_r36.4.tegra-aarch64-cu126-22.04-sudonim_test.sh.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1. 
[12:48:07] ===================================================================================== 
[12:48:07] ===================================================================================== 
[12:48:07] Failed building:  ollama

Traceback (most recent call last):
  File "/opt/jetson-containers/jetson_containers/build.py", line 129, in <module>
    build_container(**vars(args))
  File "/opt/jetson-containers/jetson_containers/container.py", line 246, in build_container
    test_container(container_name, pkg, simulate, build_idx=idx)
  File "/opt/jetson-containers/jetson_containers/container.py", line 456, in test_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'docker run -t --rm --network=host --privileged --runtime=nvidia   --volume /opt/jetson-containers/packages/llm/sudonim:/test   --volume /opt/jetson-containers/data:/data   ollama:r36.4.tegra-aarch64-cu126-22.04-sudonim     /bin/bash -c '/bin/bash /test/test.sh' 2>&1 | tee /opt/jetson-containers/logs/20250906_123649/test/09-1_ollama_r36.4.tegra-aarch64-cu126-22.04-sudonim_test.sh.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.

Failed to build ollama image

### Environment

Jetson Orin AGX

### Are you willing to submit a PR?

- [ ] Yes I'd like to help by submitting a PR!

motote0302, 2025-09-08:
this is  same case?  https://github.com/dusty-nv/jetson-containers/issues/1351

johnnynunez, 2025-09-08:
> this is same case? [#1351](https://github.com/dusty-nv/jetson-containers/issues/1351)

yes, and it was fixed in dev

kentsuiGitHub, 2025-09-08:
Tried to build with dev but same error

fcollonval, 2025-09-27:
Trying directly the Python within the sudonim docker, I get:

```
Python 3.12.3 (main, Aug 14 2025, 17:47:21) [GCC 13.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ctypes
>>> c = ctypes.CDLL('libcuda.so')
>>> result = ctypes.c_int()
>>> result = c.cuInit(0)
>>> result
34
>>> error_str = ctypes.c_char_p()
>>> c.cuGetErrorString(result, ctypes.byref(error_str))
34
>>> error_str
c_char_p(None)
```

Googling to get the error of value 34, I found out in [that doc](https://docs.nvidia.com/cuda/cuda-driver-api/group__CUDA__TYPES.html#group__CUDA__TYPES_1gc6c391505e117393cc2558fff6bfc2e9):

CUDA_ERROR_STUB_LIBRARY = 34
    This indicates that the CUDA driver that the application has loaded is a stub library. Applications that run with the stub rather than a real driver loaded will result in CUDA API returning this error. 

---

As I don't know anything about CUDA, I don't have much idea what can be wrong.

fcollonval, 2025-09-27:
Diving a bit more, I saw there are 2 `libcuda.so` in the docker image:

```
root@thunder-ai:/# find / -iname libcuda.so
/usr/lib/aarch64-linux-gnu/nvidia/libcuda.so
/usr/lib/aarch64-linux-gnu/libcuda.so
/usr/local/cuda-13.0/targets/sbsa-linux/lib/stubs/libcuda.so
/usr/local/cuda-13.0/targets/sbsa-linux/lib/libcuda.so
root@thunder-ai:/# echo $LIBRARY_PATH
/usr/local/cuda/lib64/stubs:
```

Strangely only the `stubs` folder is listed in `LIBRARY_PATH`.

If I force loading the non-stubs, initialization succeeds:

```
>>> d = ctypes.CDLL('/usr/lib/aarch64-linux-gnu/libcuda.so')
>>> result = d.cuInit(0)
>>> result
0
>>> nGpus = ctypes.c_int()
>>> result = d.cuDeviceGetCount(ctypes.byref(nGpus))
>>> nGpus
c_int(1)
>>> result
0
```

I hope this is useful to fix the issue.

kentsuiGitHub, 2025-09-27:
I follow #1351 to run the following lines and can build the image successfully. This is not the permanent fix but at least I can build and run ollama 0.12.3 now.
```
git checkout origin/dev
git checkout 5c3436bc
sed -i '/^# depends: \[python\]$/s/\[python\]/[cuda, python]/' packages/llm/sudonim/Dockerfile
./build.sh --build-flags="--no-cache" --no-github-api sudonim
```

D-G-Dimitrov, 2025-09-30:
As a "fallback" solution you can check the official `ollama/ollama:latest` [docker image](https://hub.docker.com/r/ollama/ollama/tags), it has support for `linux/arm64` OS and last time I've checked it was working for the Jetson Orin AGX.
Check the [info here about Jetson](https://github.com/ollama/ollama/blob/main/docs/docker.md#nvidia-gpu)